MarlinFirmware/Marlin #17823: [BUG] Bugfix-2.0.x Commit "Allow STM32 pins to specify timers (#17805)" Breaks compiling for TMC's on UART on STM32 platform

GhostlyCrowd, 2020-05-01:
This merge breaks compiling for TMC 2209's on the STM32 platform and I'm assuming anything else that uses SW serial, reverting this commit solves this issue 

`Marlin\src\HAL\STM32\HAL.cpp: In function 'void HAL_init()':
Marlin\src\HAL\STM32\HAL.cpp:86:42: error: 'SWSERIAL_TIMER_IRQ_PRIO' was not declared in this scope
     SoftwareSerial::setInterruptPriority(SWSERIAL_TIMER_IRQ_PRIO, 0);
                                          ^~~~~~~~~~~~~~~~~~~~~~~
Marlin\src\HAL\STM32\HAL.cpp:86:42: note: suggested alternative: 'EXTI_IRQ_PRIO'
     SoftwareSerial::setInterruptPriority(SWSERIAL_TIMER_IRQ_PRIO, 0);
                                          ^~~~~~~~~~~~~~~~~~~~~~~
                                          EXTI_IRQ_PRIO
*** [.pio\build\BIGTREE_SKR_PRO\src\src\HAL\STM32\HAL.cpp.o] Error 1`

GhostlyCrowd, 2020-05-01:
Moving the defines out of the class file and back into the header fixes this obvious, I'm not a coder so I'm not sure why it was done this way, was this in error?

viper93458, 2020-05-01:
Same issue with 2208s on SKR Pro 1.1 for me. :(

-William

thisiskeithb, 2020-05-01:
It's probably across all STM32F4 since I can't compile for a BigTreeTech BTT002 after b4aebbe78d67ec16e63bcc5240aa50c351519012 / https://github.com/MarlinFirmware/Marlin/pull/17805 either:

```prolog
Marlin/src/HAL/STM32/HAL.cpp: In function 'void HAL_init()':
Marlin/src/HAL/STM32/HAL.cpp:86:42: error: 'SWSERIAL_TIMER_IRQ_PRIO' was not declared in this scope
     SoftwareSerial::setInterruptPriority(SWSERIAL_TIMER_IRQ_PRIO, 0);
                                          ^~~~~~~~~~~~~~~~~~~~~~~
Compiling .pio/build/BIGTREE_BTT002/src/src/core/utility.cpp.o
Marlin/src/HAL/STM32/HAL.cpp:86:42: note: suggested alternative: 'EXTI_IRQ_PRIO'
     SoftwareSerial::setInterruptPriority(SWSERIAL_TIMER_IRQ_PRIO, 0);
                                          ^~~~~~~~~~~~~~~~~~~~~~~
                                          EXTI_IRQ_PRIO
Compiling .pio/build/BIGTREE_BTT002/src/src/feature/babystep.cpp.o
Compiling .pio/build/BIGTREE_BTT002/src/src/feature/backlash.cpp.o
*** [.pio/build/BIGTREE_BTT002/src/src/HAL/STM32/HAL.cpp.o] Error 1
```

sjasonsmith, 2020-05-02:
I posted a PR which I believe resolves this. I don't have an STM32F4 board set up right now, so it would be great if one of you that has encountered this would be willing to try the fix in PR #17831.

viper93458, 2020-05-02:
I'll test right now. :)

-William

viper93458, 2020-05-02:
It did compile successfully! I will try a print as soon as my current one is done in about 20-30 minutes.

-William

thisiskeithb, 2020-05-02:
@sjasonsmith: I rebased my BTT002 config on top of that pr, but I'm now getting the following error:

```prolog
In file included from Marlin/src/HAL/STM32/../../inc/../core/boards.h:24:0,
                 from Marlin/src/HAL/STM32/../../inc/MarlinConfigPre.h:35,
                 from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:28,
                 from Marlin/src/HAL/STM32/timers.h:25,
                 from Marlin/src/HAL/STM32/timers.cpp:24:
Marlin/src/HAL/STM32/timers.cpp: In function 'void SetSoftwareSerialTimerInterruptPriority()':
Marlin/src/HAL/STM32/timers.cpp:48:44: error: 'setInterruptPriority' is not a member of 'SoftwareSerial'
   TERN_(HAS_TMC_SW_SERIAL, SoftwareSerial::setInterruptPriority(SWSERIAL_TIMER_IRQ_PRIO, 0));
                                            ^
Marlin/src/HAL/STM32/../../inc/../core/macros.h:406:26: note: in definition of macro 'THIRD'
 #define THIRD(a,b,c,...) c
                          ^
Compiling .pio/build/BIGTREE_BTT002_BETA/src/src/feature/bedlevel/ubl/ubl_G29.cpp.o
Marlin/src/HAL/STM32/../../inc/../core/macros.h:201:29: note: in expansion of macro '___TERN'
 #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
Compiling .pio/build/BIGTREE_BTT002_BETA/src/src/feature/bedlevel/ubl/ubl_motion.cpp.o
                             ^~~~~~~
Marlin/src/HAL/STM32/../../inc/../core/macros.h:200:29: note: in expansion of macro '__TERN'
 #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
                             ^~~~~~
Marlin/src/HAL/STM32/../../inc/../core/macros.h:199:29: note: in expansion of macro '_TERN'
 #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION converted to A or '<nul>'
                             ^~~~~
Marlin/src/HAL/STM32/timers.cpp:48:3: note: in expansion of macro 'TERN_'
   TERN_(HAS_TMC_SW_SERIAL, SoftwareSerial::setInterruptPriority(SWSERIAL_TIMER_IRQ_PRIO, 0));
   ^~~~~
*** [.pio/build/BIGTREE_BTT002_BETA/src/src/HAL/STM32/timers.cpp.o] Error 1
```

[Link to rebased BTT002 beta config](https://github.com/thisiskeithb/Marlin/tree/testing/17495-btt002-beta)

viper93458, 2020-05-02:
Printing some parts now. Again, I have 2208s, not 2209s. I simply manually merged your PR into the copy I downloaded this morning that threw the error in this issue.

-William

sjasonsmith, 2020-05-02:
> @sjasonsmith: I rebased my BTT002 config on top of that pr, but I'm now getting the following errors

You need to ignore the built-in SoftwareSerial library, and add an include path to the HAL/STM32 folder so TMCStepper will find Marlin's version. The version in the STM32 framework we currently use is broken, so we have our own copy for now.

[env:BIGTREE_BTT002_BETA]
platform          = ststm32@5.6.0
board             = BigTree_Btt002_beta
platform_packages = framework-arduinoststm32@>=3.10700,<4
build_flags       = ${common.build_flags}
  -DUSBCON -DUSBD_USE_CDC -DUSBD_VID=0x0483 -DUSB_PRODUCT=\"STM32F407VE\"
  -DTARGET_STM32F4 -DSTM32F407_5VX -DVECT_TAB_OFFSET=0x8000
  -DHAVE_HWSERIAL2
  -DHAVE_HWSERIAL3
  -DPIN_SERIAL2_RX=PD_6
  -DPIN_SERIAL2_TX=PD_5
  **-IMarlin/src/HAL/STM32**
build_unflags     = -std=gnu++11
extra_scripts     = pre:buildroot/share/PlatformIO/scripts/generic_create_variant.py
lib_ignore        = Adafruit NeoPixel, SailfishLCD, SlowSoftI2CMaster, **SoftwareSerial**
src_filter        = ${common.default_src_filter} +<src/HAL/STM32>

thisiskeithb, 2020-05-02:
Looks like `platformio.ini` needs to be updated for at least the `BIGTREE_BTT002` environment (I know I'm using a special beta version since it has a smaller MCU) and maybe others.

What should the include path be?

```prolog
Marlin/src/HAL/STM32/timers.cpp:45:12: fatal error: SoftwareSerial.h: No such file or directory

************************************************************************
* Looking for SoftwareSerial.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SoftwareSerial.h"
* Web  > https://platformio.org/lib/search?query=header:SoftwareSerial.h
*
************************************************************************

   #include <SoftwareSerial.h>
            ^~~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pio/build/BIGTREE_BTT002_BETA/src/src/MarlinCore.cpp.o
Compiling .pio/build/BIGTREE_BTT002_BETA/src/src/core/multi_language.cpp.o
Compiling .pio/build/BIGTREE_BTT002_BETA/src/src/core/serial.cpp.o
*** [.pio/build/BIGTREE_BTT002_BETA/src/src/HAL/STM32/timers.cpp.o] Error 1
In file included from Marlin/src/module/stepper/trinamic.h:29:0,
                 from Marlin/src/module/stepper/indirection.h:44,
                 from Marlin/src/module/stepper.h:47,
                 from Marlin/src/MarlinCore.cpp:45:
.pio/libdeps/BIGTREE_BTT002_BETA/TMCStepper_ID5513/src/TMCStepper.h:23:11: fatal error: SoftwareSerial.h: No such file or directory
```

sjasonsmith, 2020-05-02:
@thisiskeithb I just pushed a new commit to my PR to fix it!

thisiskeithb, 2020-05-02:
I didn't see the `-IMarlin/src/HAL/STM32` line 🙂

GhostlyCrowd, 2020-05-02:
I'll give this a go on the SKR Pro 1.1 in about an hour when i get to my office.

GhostlyCrowd, 2020-05-02:
> @thisiskeithb I just pushed a new commit to my PR to fix it!

Youre PR does indeed fix it and im running a test print now, now if only i can figure out why my Y is layer shifting ive replaced all mechanics and the shift changes direction with an axis invert and the new stepper.  Now time to roll back 7-8 days and see whats up.

GhostlyCrowd, 2020-05-02:
I will close this issue as the PR solves the issue and looks like it will be merged.

closed 2020-05-02 by GhostlyCrowd