MarlinFirmware/Marlin #17828: [BUG] Z2_USE_ENDSTOP can't be a custom pin and allow Z2_MIN_ENDSTOP_INVERTING to be defined

jgwinner, 2020-05-01:
### Bug Description

I have a fairly custom layout, and all my `_*max_` and `_*min_` pins are being used. So I used an expansion pin for a REPRAP style limit switch on Z2. This all works fine. However, I cannot #define Z2_MIN_ENDSTOP_INVERTING as in conditionals_post.h, line 942 has the code:

```
    #if Z2_USE_ENDSTOP == _XMIN_
      #define Z2_MIN_ENDSTOP_INVERTING X_MIN_ENDSTOP_INVERTING
    #elif 
      ...
    #else
      #define Z2_MIN_ENDSTOP_INVERTING false
    #endif
```

As I already have Z2_MIN_ENDSTOP_INVERTING set true, this causes copious warnings with a redefined symbol.

### Suggested fix:
Change the code to:
```
    #else
      #ifndef Z2_MIN_ENDSTOP_INVERTING
        #define Z2_MIN_ENDSTOP_INVERTING false
      #endif
    #endif
```
### My Configurations

The configuration that's germane is

```
#define Z2_MIN_PIN PC9
#define Z2_MIN_ENDSTOP_INVERTING true
```

Let me know if you want all the rest of it.

### Steps to Reproduce

1. Have a control board with all `_*MAX_` and `_*MIN_` connectors already used.
Mine is an SKR PRO 1.1
2. Set a #define for Z2_USE_ENDSTOP in your configuration.h
3. Purchase REPRAP style limit switches, available on Nile, er, some other large river.
4. Set a #define for Z2_MIN_ENDSTOP_INVERTING true in configuration.h
5. Compile.

**Expected behavior:** [What you expect to happen]
Compiles without warning
**Actual behavior:** [What actually happens]
Warnings bleed yellow all over the screen, obfusticating other errors.
#### Additional Information

sjasonsmith, 2020-05-02:
Please post your configuration files as requested in the issue template.

jgwinner, 2020-05-02:
I did see that, not sure you read what I posted ... 

Here they are anyway, and now we're going down a rabbit hole. Note I had to change the pin assignments as well, for other reasons. 

[MyConfig.zip](https://github.com/MarlinFirmware/Marlin/files/4567054/MyConfig.zip)

sjasonsmith, 2020-05-02:
I'm not sure what this rabbit hole you are referring to. All I did was ask for your configuration files. For anybody to reproduce the issue you are discussing, they have to go out of their way to create a configuration and _hope_ it matches yours closely enough to expose the same issue. By posting your configurations, you ensure that anybody hoping to help encounters exactly the same issues you do.

Also, if you already have a solution for it, you can go ahead and post a pull request to fix the problem. I admittedly hasn't read the second code snippet with the proposed fix. It looks like the right thing to do, although I haven't gone and looked at it in context.

jgwinner, 2020-05-02:
I was worried anyone looking at the configuration would go off about "Well, no wonder he's got a problem, he changed the pins file!!" which has little to do with this specific problem. 

> you can go ahead and post a pull request to fix the problem.

Ok, will do. I wasn't sure how this project looks on pull requests. Thanks!

sjasonsmith, 2020-05-02:
I think the goal is that a typical user doesn't have to modify the pins files, but changing them is pretty common. Your scenario doesn't sound very typical, so it's probably hard to avoid.

jgwinner, 2020-05-02:
Thanks! I appreciate that.

boelle, 2020-06-20:
@jgwinner is the issue still there?

jgwinner, 2020-06-21:
I haven't done the pull request :(  but I fixed it a while ago, with the exact code in my first post.

You want me to open a pull request?

boelle, 2020-06-21:
yes please and remember to mention the PR here and mention the issue in the PR

boelle, 2020-06-29:
@jgwinner did you open a pull request ?

jgwinner, 2020-06-30:
Sorry, I'm totally jammed trying to find a job, I have not. I'll try to get to it in a couple of days.

I have to admit - I've never done a pull request on a major project like this, so I have to walk through a tutorial too :)

sjasonsmith, 2020-07-04:
I suspect this was been fixed by a recent pull request a few days ago:
https://github.com/MarlinFirmware/Marlin/pull/18481

When you are able, please try that out (it is in the `bugfix-2.0.x branch), and report back whether it has resolved the issue.

closed 2020-08-09 by github-actions[bot]