MarlinFirmware/Marlin #17829: Hi guys i bought skr1.3 and tried to flash latest marlin , I have successfully updated it but it seems to have an issue : when i try to print over onboard SD for 30-40 min and after that board start rebooting it self i dont know weather its the marlin problem because i have 5 skr boards with older marlin working in my print farm but i deleted old marlin and downloaded new one since then i am getting this issue[BUG] (short description)

fnhhnxfn, 2020-05-02:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

chepo92, 2020-05-02:
Now that is a title

Powerprobot, 2020-05-02:
I think it is a hardware problem. My SKR1.3 and latest bugfix version (1. May) runs without problems for 15 hours. Printing from SD-Card from the 12864 LCD Display. Check the 5Volt Line of your Board, it is a litte unstable. You can fix it with a 100µF Capicitor.
You added no configuration files and no more Informations of Printer or Laser cutter. 
Check your Lineup savely and step to step and close this issue.
PS: Title means title and not an novel...

fnhhnxfn, 2020-05-03:
i am sorry i dindnt realised that it was title ,  actually i am pretty new to this field , you said to check the 5V line can you please send me the pic of what i actually have to check?
and where should i put the 100µF capacitor for it to work

Powerprobot, 2020-05-03:
@fnhhnxfn , at first watch the printer during printing and recognize all abnormal temperatures (right Temperature sensor?), movings and acoustic differences to normal. 
Have you use another SD-Card to check?
Have you use another Slicer to correct GCODE?
Take a fresh copy of Marlin-Bugfix and compile fresh.
Check all Cables of broke or Short Circuits esp. from HeadBed to Aluminium-Plate. Take a Multimeter and the SKR1.3 PINS Diagramm to connect and check the Voltage during Print. Check the 3Volts and 5Volts and 12/24 Volts parallel. Is this not possible make this one by one.
[Anleitung Bigtree 1.3 mit Kondensator.pdf](https://github.com/MarlinFirmware/Marlin/files/4571517/Anleitung.Bigtree.1.3.mit.Kondensator.pdf)
I had problems a long time ago with my Display. So i found a solution and i hope this can help you. Read the PDF i attached. 
Close this Issue, I can neverless work on your Problem in this Thread.

bill-orange, 2020-05-04:
I will try to reproduce this problem on my SKR1.3 running 2.0.5.3.

closed 2020-05-04 by thinkyhead