MarlinFirmware/Marlin #17835: [BUG] Homing Bump Configuration Error For Building

Gerengnick, 2020-05-02:
### Bug Description

When I compile using the configurations for an SKR Mini E3 V1.2 board using the CR-10 config there is an error due to the homing bump configurations. When compared to the former versions of marlin, the formatting for setting bump configuration was with 3 separate lines for X, Y, and Z, but with the newest version it has been condensed to one line. When I changed it to the old formatting it compiled with no issues.

### My Configurations

This is the one that created errors (located in configuration_adv.h)
[bugfix_Submission.zip](https://github.com/MarlinFirmware/Marlin/files/4568581/bugfix_Submission.zip)

This is correctly fixed and compiles properly (Again, within configuration_adv.h)
[bugfix_Submission_FIXED.zip](https://github.com/MarlinFirmware/Marlin/files/4568588/bugfix_Submission_FIXED.zip)

### Steps to Reproduce

1. Configure Marlin for SKR Mini E3 V1.2 and Cr-10
2. Compile for flashing
3. Error given for incorrect formatting for homing bump

**Expected behavior:** Builds properly

**Actual behavior:** Fails building and requires formatting for XYZ bump to be reformatted to the old style.

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

sjasonsmith, 2020-05-02:
I have had to make exactly the opposite change when using configs with the current bugfix-2.0.x branch.

It seems like you are probably mixing different versions of examples and Marlin source.

If you think there is actually an issue to resolve here, please make sure you are using clean Marlin source and examples (from matching branch names, such as `bugfix-2.0.x`), then post _exactly_ where you obtained each file from.

boelle, 2020-06-20:
@Gerengnick still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle