MarlinFirmware/Marlin #17836: [BUG] Unable to compile with #define Z_DUAL_STEPPER_DRIVERS selected

bill-orange, 2020-05-02:
Bug Description:
Compilation results in the error :
 `#error "Z_DUAL_STEPPER_DRIVERS is now NUM_Z_STEPPER_DRIVERS with a value of 2. Please update Configuration_adv.h.` from sanity check

My Configurations: included below

Steps to Reproduce:

1. revise configuration.h to include   `#define Z2_DRIVER_TYPE TMC2130`
2. revise configuration_adv.h to include    `#define Z_DUAL_STEPPER_DRIVERS`
3. note that current options in  configuration_adv.h  remain shaded out for E1
4. attempt to compile and get the error noted above

Expected behavior: Sucessful compilation

Actual behavior: get' #error "Z_DUAL_STEPPER_DRIVERS is now NUM_Z_STEPPER_DRIVERS with a value of 2. Please update Configuration_adv.h.` from sanity check

Additional Information: am I missing something obvious?

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4568801/Marlin.zip)

sjasonsmith, 2020-05-02:
Isn't that error message explaining exactly what you need to do to solve your problem?

Azrael321, 2020-05-02:
You need to use the actual Configuration_adv.h

bill-orange, 2020-05-02:
@Azrael321 I did not understand the error message.   I commented out the old block of code and put in this one and I am compiling.

```
//
// For Z set the number of stepper drivers
//
#define NUM_Z_STEPPER_DRIVERS 1   // (1-4) Z options change based on how many

#if NUM_Z_STEPPER_DRIVERS > 1
  //#define Z_MULTI_ENDSTOPS
  #if ENABLED(Z_MULTI_ENDSTOPS)
    #define Z2_USE_ENDSTOP          _XMAX_
    #define Z2_ENDSTOP_ADJUSTMENT   0
    #if NUM_Z_STEPPER_DRIVERS >= 3
      #define Z3_USE_ENDSTOP        _YMAX_
      #define Z3_ENDSTOP_ADJUSTMENT 0
    #endif
    #if NUM_Z_STEPPER_DRIVERS >= 4
      #define Z4_USE_ENDSTOP        _ZMAX_
      #define Z4_ENDSTOP_ADJUSTMENT 0
    #endif
  #endif
#endif

```

bill-orange, 2020-05-02:
Success.  Thanks.

closed 2020-05-02 by bill-orange