MarlinFirmware/Marlin #17838: [BUG] CoreXY Sensorless Homing only Monitors one Motor

xalpha, 2020-05-03:
### Bug Description

CoreXY setups use both motors (motors A and B) to achieve motion, even when only one axis is involved. However, when doing sensorless homing, **it appears that only a single motor is monitored for changes in load: motor A for X-axis or motor B for Y-axis**.

In my case, when homing the Y-axis, the carriage reaches the end of the Y-axis and only motor A skips steps. Because of that, motor B, which is the one being monitored, can continue to turn unimpeded. Homing the Y-axis thus fails

### My Configurations

DIY Hypercube (Tech2C), 24V power supply, with BTT SKR v1.3 and TMC2130 (watterott)

Stall Sensitivity has been set for both axies to the lowest possible value, i.e. first one where the homing doesn't trigger immediately. 

### Steps to Reproduce

1. Start Y-axis homing: OK
2. Clamp hard on motor A (carefully not to hurt fingers): motor A skips steps but homing continues
3. Clamp hard on motor B (carefully not to hurt fingers): homing cycle detects the load and stops homing

**Expected behavior:** homing detects load in any of the two motors

**Actual behavior:** homing only detects load in one of the motors

### Additional Information
Might just be an issue in motion.cpp:do_homing_move(), but I'm not familiar with this code base.

If someone could assist with the intricacies of the code and architecture, I could maybe contribute some time towards a fix.

deltaford, 2020-05-04:
I have observed the same issue on my CoreYX running sensorless with BTT SKR 1.4 Turbo TMC2209.

From what I have been able to gather from digging around in the code with my very limited experience, it seems to me that the problem is that endstops are only checked for the axis that is moving. But for CoreXY with sensorless/stallguard both X and Y endstop (motor stall triggered) needs to checked when homing X or Y.

In an attempt to try fix the issue I modified Endstops::update() in endstops.ccp, adding this highlighted code. It may be completly wrong but so far seem to work better.

 // Now, we must signal, after validation, if an endstop limit is pressed or not
  if (stepper.axis_is_moving(X_AXIS)) {
    if (stepper.motor_direction(X_AXIS_HEAD)) { // -direction
      #if HAS_X_MIN || (X_SPI_SENSORLESS && X_HOME_DIR < 0)
        PROCESS_ENDSTOP_X(MIN);
        **_#if CORE_IS_XY && X_SENSORLESS && Y_SENSORLESS
          PROCESS_ENDSTOP_Y(MIN);_**
        #endif
      #endif
    }
    else { // +direction
      #if HAS_X_MAX || (X_SPI_SENSORLESS && X_HOME_DIR > 0)
        PROCESS_ENDSTOP_X(MAX);
      #endif
    }
  }

  if (stepper.axis_is_moving(Y_AXIS)) {
    if (stepper.motor_direction(Y_AXIS_HEAD)) { // -direction
      #if HAS_Y_MIN || (Y_SPI_SENSORLESS && Y_HOME_DIR < 0)
        PROCESS_ENDSTOP_Y(MIN);
        **_#if CORE_IS_XY && X_SENSORLESS && Y_SENSORLESS
          PROCESS_ENDSTOP_X(MIN);_**
        #endif
      #endif
    }
    else { // +direction
      #if HAS_Y_MAX || (Y_SPI_SENSORLESS && Y_HOME_DIR > 0)
        PROCESS_ENDSTOP_Y(MAX);
      #endif
    }
  }

XDA-Bam, 2020-05-05:
To clarify: What do you mean by "seem to work better"? Is it running totally fine for you with this fix, or are there still problems?

deltaford, 2020-05-05:
@XDA-Bam What I mean is that homing now seem to work as it should. It triggers without any motor grinding. 
I can't tell if it is "running totally fine" as I'm running the code on a new custom built machine that's not finished yet and still has too many unknowns.

XDA-Bam, 2020-05-05:
OK, thanks. I think checking for `CORE_IS_XY` here should not be necessary, because regular CoreXY printers are homing correctly. At least I haven't seen any complaints recently.

My guess would be, that the check for `X_SENSORLESS & Y_SENSORLESS` is what's missing. Your solution includes those, so at least one endstop triggers. That would also mean, that your fix actually triggers the endstops in a switched manner: It triggers the Y endstop when checking for X movements and vice versa. 

Could you try the following code and see if this also runs fine?
```
  // Now, we must signal, after validation, if an endstop limit is pressed or not
  if (stepper.axis_is_moving(X_AXIS)) {
    if (stepper.motor_direction(X_AXIS_HEAD)) { // -direction
      #if HAS_X_MIN || (X_SPI_SENSORLESS && X_HOME_DIR < 0) || (X_SENSORLESS && X_HOME_DIR < 0)
        PROCESS_ENDSTOP_X(MIN);
      #endif
    }
    else { // +direction
      #if HAS_X_MAX || (X_SPI_SENSORLESS && X_HOME_DIR > 0) || (X_SENSORLESS && X_HOME_DIR > 0)
        PROCESS_ENDSTOP_X(MAX);
      #endif
    }
  }

  if (stepper.axis_is_moving(Y_AXIS)) {
    if (stepper.motor_direction(Y_AXIS_HEAD)) { // -direction
      #if HAS_Y_MIN || (Y_SPI_SENSORLESS && Y_HOME_DIR < 0) || (Y_SENSORLESS && Y_HOME_DIR < 0)
        PROCESS_ENDSTOP_Y(MIN);
      #endif
    }
    else { // +direction
      #if HAS_Y_MAX || (Y_SPI_SENSORLESS && Y_HOME_DIR > 0) || (Y_SENSORLESS && Y_HOME_DIR > 0)
        PROCESS_ENDSTOP_Y(MAX);
      #endif
    }
  }
```

I would check it myself, but I've only got a cartesian machine.

deltaford, 2020-05-05:
What I did in my fix was just to add another check for the other axis endstop If the current axis endstop is being checked. If moving in X axis and X endstop (X motor stall) is checked, then check also Y endstop (Y motor stall) if `CORE_IS_XY ` and sensorless.

If I understand correctly, if not `CORE_IS_XY ` (and assuming cartesian) then the extra check should not be needed since moving in X axis using only X motors shouldnt need checking Y endstop (Y motor stall).

From what I can tell from the code if moving in X axis, only  `if (stepper.axis_is_moving(X_AXIS))` would be true? If so, then I don't see how your code would fix the problem with not catching Y endstop (Y motor stall) being triggered?

XDA-Bam, 2020-05-05:
Maybe you're right. But "regular" switch-endstop CoreXY printers seem to home normally, so either `if (stepper.axis_is_moving(Y_AXIS))` is true for them and not for your sensorless setup, or `HAS_Y_MIN` is false in your case. It could theoretically also break, if `Y_AXIS_HEAD` is not defined at all and the code stalls, but that seems less likely.

I'm also not familiar with the code, but looking into it, the steppers seem to be [mapped A->X and B->Y](https://github.com/MarlinFirmware/Marlin/blob/7a2cc782b406f7e89a0a61d168cc7a5d90835c06/Marlin/src/module/stepper.cpp#L1944) in case of CoreXY. `axis_did_move` is then checked by `axis_is_moving()`. Therefore, I would expect `stepper.axis_is_moving()` to be true for both, X and Y (or A and B) when homing on a CoreXY, which means we would also correctly check both stepper endstops.

So maybe try my code nonetheless - if it works, we know that `HAS_Y_MIN` is likely to be the problem. If it doesn't work, we know that `if (stepper.axis_is_moving(Y_AXIS))` is the problem.

thinkyhead, 2020-05-06:
@XDA-Bam — SPI endstops need to be talked to via SPI. But non-SPI endstops just trigger the regular endstop pins. So your code change above is inappropriate. Those checks are only needed to get the states from SPI.

deltaford, 2020-05-12:
I have updated my fix and it should now account for both +/- directions as well as that X and Y endstops might be configured differently. It has been running without issues on my now completed machine and homing works as expected.

`// Now, we must signal, after validation, if an endstop limit is pressed or not
  if (stepper.axis_is_moving(X_AXIS)) {
    if (stepper.motor_direction(X_AXIS_HEAD)) { // -direction
      #if HAS_X_MIN || (X_SPI_SENSORLESS && X_HOME_DIR < 0)
        PROCESS_ENDSTOP_X(MIN);
        **#if CORE_IS_XY && X_SENSORLESS && Y_SENSORLESS  
          #if HAS_Y_MIN
            PROCESS_ENDSTOP_Y(MIN);
          #elif HAS_Y_MAX
            PROCESS_ENDSTOP_Y(MAX);
          #endif
        #endif**
      #endif
    }
    else { // +direction
      #if HAS_X_MAX || (X_SPI_SENSORLESS && X_HOME_DIR > 0)
        PROCESS_ENDSTOP_X(MAX);
        **#if CORE_IS_XY && X_SENSORLESS && Y_SENSORLESS
          #if HAS_Y_MIN
            PROCESS_ENDSTOP_Y(MIN);
          #elif HAS_Y_MAX
            PROCESS_ENDSTOP_Y(MAX);
          #endif
        #endif**
      #endif
    }
  }

  if (stepper.axis_is_moving(Y_AXIS)) {
    if (stepper.motor_direction(Y_AXIS_HEAD)) { // -direction
      #if HAS_Y_MIN || (Y_SPI_SENSORLESS && Y_HOME_DIR < 0)
        PROCESS_ENDSTOP_Y(MIN);
        **#if CORE_IS_XY && X_SENSORLESS && Y_SENSORLESS
          #if HAS_X_MIN
            PROCESS_ENDSTOP_X(MIN);
          #elif HAS_X_MAX
            PROCESS_ENDSTOP_X(MAX);
          #endif
        #endif**
      #endif
    }
    else { // +direction
      #if HAS_Y_MAX || (Y_SPI_SENSORLESS && Y_HOME_DIR > 0)
        PROCESS_ENDSTOP_Y(MAX);
        **#if CORE_IS_XY && X_SENSORLESS && Y_SENSORLESS
          #if HAS_X_MIN
            PROCESS_ENDSTOP_X(MIN);
          #elif HAS_X_MAX
            PROCESS_ENDSTOP_X(MAX);
          #endif
        #endif**
      #endif
    }
  }`

thinkyhead, 2020-05-12:
While your solution does cause some endstop bits to get triggered so that the homing is interrupted, the bits that get set don't necessarily match the axis that was homing. So a little more processing will need to be added so the endstop hit state reflects the axis or axes which were moving. With CoreXY quick homing, we don't know whether X or Y endstop was hit, so we might need to trigger both when both axes are moving. And so on….

thinkyhead, 2020-05-12:
Please check to see if #17972 does the trick. It seems like it ought to.

deltaford, 2020-05-12:
It looks to me as if #17972 would only fix the issue if using SPI sensorless, not if running non-SPI (DIAG1) sensorless?

thinkyhead, 2020-05-16:
Keep us posted if any other features need fixing for Core kinematics. We don't have one on hand here to test with, but one of these days we'll port Marlin to the E3D Toolchanger.

closed 2020-05-16 by thinkyhead