MarlinFirmware/Marlin #17847: [BUG] Fysetc S6 does not start with inserted SD Card in the Display

P-C-R, 2020-05-03:
### Bug Description

When SD Card is inserted in the SD Slot of the Display everything works till i restart the board. After restart the Board is completly dead and i have to flash the firmware new.

But when i insert the SD, print, and remove the card the printer restarts fine.

Tried different displays too!

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4569903/Marlin.zip)
### Steps to Reproduce

Start Board
Insert SD Card into Slot of Display
Restart Board

**Expected behavior:**
Restart of the board

**Actual behavior:** [What actually happens]
Board is dead! Bootloop?

trouch, 2020-05-10:
This has nothing related with Marlin. S6 has SD bootloader for firmware upgrade. 
1. Make sure you don't have any `firmware.bin` file on your SD card.
Maybe you broke it somehow, and "flashing" allows you to actually reset the MCU at the proper firmware address `0x08010000`.

trouch, 2020-05-10:
S6 has an SD bootloader for firmware upgrade, so it might be broken.

1. Make sure you don't have any firmware.bin file on your SD card.

2. Maybe you erased the bootloader, and "flashing" allows you to actually reset the MCU at the proper firmware address.
Flash the [bootloader hex file](https://github.com/FYSETC/Bootloader-S6) at `0x08000000` using STM32CubeProgrammer*.
Then flash Marlin** at `0x08010000`.

If you still have an issue, please report to Fysetc directly as it has nothing to do with Marlin here.

(*) STM32CubeProgrammer is mandatory to flash the bootloader **hex** file, then you can use dfu-util to flash **bin** file such as Marlin.
(**) I've just submitted a PR to allow flashing Marlin under DFU directly from PlatformIO https://github.com/MarlinFirmware/Marlin/pull/17943

@thinkyhead you can probably close this one.

P-C-R, 2020-05-10:
I will Check it. It could be a old firmware.bin

Best

P-C-R, 2020-05-12:
It was a firmware.bin on the SD Card!

closed 2020-05-12 by P-C-R