MarlinFirmware/Marlin #17850: [BUG] Re-arm 32 and Ramps1.6 hot bed still on after M112

KrX3D, 2020-05-03:
### Description

I got an re-arm 32 board:
https://www.aliexpress.com/item/33007283020.html?ws_ab_test=searchweb0_0%2Csearchweb201602_10_10065_10068_10547_319_317_10548_10696_10084_453_10083_454_10618_10304_10307_10820_10821_537_10302_536_10843_10059_10884_10887_321_322_10103%2Csearchweb201603_53%2CppcSwitch_0&algo_expid=0eeedca5-5fbc-41cf-9b42-d849ba235c2e-0&algo_pvid=0eeedca5-5fbc-41cf-9b42-d849ba235c2e

and an ramps 1.6 board:
https://www.aliexpress.com/item/32919533439.html?spm=2114.search0104.3.1.78ee12a28zqul9&ws_ab_test=searchweb0_0%2Csearchweb201602_10_10065_10068_10547_319_317_10548_10696_10084_453_10083_454_10618_10304_10307_10820_10821_537_10302_536_10843_10059_10884_10887_321_322_10103%2Csearchweb201603_53%2CppcSwitch_0&algo_expid=739bf55a-89db-44a9-9096-aeec618caf62-0&algo_pvid=739bf55a-89db-44a9-9096-aeec618caf62

i use on D8 (HOT BED) an Spindle and i power it on with "M106 P1 S255", to power it off i use "M106 P1 S0" that works.

but when i use "M112" the  Spindle does not power off. the LCD shows printer halted, and all movements are stopped only the spindle does not power off.
hitting the reset button on the re-arm resets eveything and the spindle goes off

### Steps to Reproduce

1. "M106 P1 S255"
2. "M112"
3. all movement stops only spindle is still ON

**Expected behavior:** [What you expect to happen]
after M112 the spindle (or hot bed) should also power off

**Actual behavior:** [What actually happens]
after the emergency M112 spindle (hot bed) is still powered on

i reported it a while back:

https://github.com/MarlinFirmware/Marlin/issues/14599

but the thread was closed and nothing happened since more than 4 months

my last answer there is still valid: 
https://github.com/MarlinFirmware/Marlin/issues/14599#issuecomment-578278619

AnHardt, 2020-05-03:
M106 is switching the fan not the bed.
The fan is by purpose not switched off with M112.

You should play with the EBF, EFF, .... settings. They are switching the purpose of the MOSFETs.

KrX3D, 2020-05-03:
> 
> 
> M106 is switching the fan not the bed.
> The fan is by purpose not switched off with M112.
> 
> You should play with the EBF, EFF, .... settings. They are switching the purpose of the MOSFETs.

yeah you are right but i got the spindle pluged in into D8 which is with my configuration "BOARD_RAMPS_14_RE_ARM_EFF" the Extruder "port" and just setting with M106 the spindle speed

so M112 should disabled D8 (Extruder) so im my case the spindle

boelle, 2020-06-20:
@KrX3D still an issue?

KrX3D, 2020-06-21:
hello, yes it is still happening

boelle, 2020-06-23:
Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

KrX3D, 2020-06-27:
hello, i tested it with the newest marlin version which i could get today. i made my changes to the config files but the problem still exists.
i was looking a bit more into this since the printer halts after M112 but i always need to hit the reset button manually.
is it possible that the re arm32 does not have a auto reset mode? 

since auto resetting it would solve the problem (like i said when i hit the rest button manually the heatbed/spinde/fan goes off)

sjasonsmith, 2020-07-04:
@AnHardt makes a very good point that fans are intentionally leaves fans enabled. M112 should work if you use  `SPINDLE_FEATURE` rather than controlling the spindle like a fan.

KrX3D, 2020-07-08:
hi, ok i will try the spindle feature in a couple of days (BOARD_RAMPS_14_RE_ARM_SF)

> makes a very good point that fans are intentionally left enabled.

not really how i am useing it, the spindle is pluged into D8 which should be with my configuration "BOARD_RAMPS_14_RE_ARM_EFF" (Hotend, Fan0, Fan1)
and with M112 the hotend should be definitely turned to OFF

KrX3D, 2020-07-12:
Hi, i tested it with BOARD_RAMPS_14_RE_ARM_SF as board, 

first test without the spindle feature. i couldnt start my spindle with M106 P1 S100 but needed M106 S100, M3 didnt work.
than i tested it with the spindle feature, also only M106 S100 worked but not M3 (S100)

with both test when the spindle was running i entered M112, but the spindle wasnt powered off

ellensp, 2020-07-16:
I was going to test this, but alas no config files... here is hoping https://github.com/MarlinFirmware/Marlin/files/4110421/Marlin.zip is still valid.

ellensp, 2020-07-16:
motherboard BOARD_RAMPS_14_RE_ARM_EFF
So your controlling the fan, and as already stated fan is not parts of m112
when you set EFF you have no hotbed, so hotbed is not turned off.

ellensp, 2020-07-16:
to add this feature in Marlincore.cpp find function void kill(PGM_P and replace it with the following.
```
void kill(PGM_P const lcd_error/*=nullptr*/, PGM_P const lcd_component/*=nullptr*/, const bool steppers_off/*=false*/) {
  thermalManager.disable_all_heaters();
  #if PIN_EXISTS(FAN)
    TERN_(USING_FAN_PIN, WRITE(FAN_PIN, 0));    // Kill fan_pin if requested.  
  #endif
  #if PIN_EXISTS(FAN1)
    TERN_(USING_FAN1_PIN, WRITE(FAN1_PIN, 0));  // Kill fan1_pin if requested.  
  #endif
  TERN_(HAS_CUTTER, cutter.kill()); // Full cutter shutdown including ISR control
  SERIAL_ERROR_MSG(STR_ERR_KILLED);
  #if HAS_DISPLAY
    ui.kill_screen(lcd_error ?: GET_TEXT(MSG_KILLED), lcd_component ?: NUL_STR);
  #else
    UNUSED(lcd_error);
    UNUSED(lcd_component);
  #endif
  #ifdef ACTION_ON_KILL
    host_action_kill();
  #endif
  minkill(steppers_off);
}
```

also replace the function just below it minkill(const  with the following.

```
void minkill(const bool steppers_off/*=false*/) {
  // Wait a short time (allows messages to get out before shutting down.
  for (int i = 1000; i--;) DELAY_US(600);
  cli(); // Stop interrupts
  // Wait to ensure all interrupts stopped
  for (int i = 1000; i--;) DELAY_US(250);
  // Reiterate heaters off
  thermalManager.disable_all_heaters();
  #if PIN_EXISTS(FAN)
    TERN_(USING_FAN_PIN, WRITE(FAN_PIN, 0));    // Reiterate Kill fan_pin if requested.  
  #endif
  #if PIN_EXISTS(FAN1)
    TERN_(USING_FAN1_PIN, WRITE(FAN1_PIN, 0));  // Reiterate Kill fan1_pin if requested.  
  #endif
  TERN_(HAS_CUTTER, cutter.kill());  // Reiterate cutter shutdown
  // Power off all steppers (for M112) or just the E steppers
  steppers_off ? disable_all_steppers() : disable_e_steppers();
  TERN_(PSU_CONTROL, PSU_OFF());
  TERN_(HAS_SUICIDE, suicide());
  #if HAS_KILL
    // Wait for kill to be released
    while (kill_state()) watchdog_refresh();
    // Wait for kill to be pressed
    while (!kill_state()) watchdog_refresh();
    void (*resetFunc)() = 0;      // Declare resetFunc() at address 0
    resetFunc();                  // Jump to address 0
  #else
    for (;;) watchdog_refresh();  // Wait for reset
  #endif
}

```

then just add either or both of these to Configuration_adv.h
```
  #define USING_FAN_PIN                true   // set if your actually using the fan_pin for spindle or laser control 
  #define USING_FAN1_PIN               true   // set if your actually using the fan_pin for spindle or laser control 

```

ellensp, 2020-07-16:
I have added this as a PR, Its now up to thinkyhead to decide if this is a good idea for all of not.

KrX3D, 2020-07-16:
> no hotbed, so hotbed is not

my mistake, i meant extruder, since 
E(xtruder)F(an)F(an) -> EFF im thinking 
E-> D8; F -> D9 and F -> D10. 

i got my spindle pluged in into D8 whick is Extruder, so when hitting M112 it should definitly be turned off.

i will try your code above later when i get home, also i will upload the newest config files here

ellensp, 2020-07-16:
no that is not correct.. with on a real ramps on a mega EFF is
```
PIN:   8   Port: H5        FAN1_PIN       ie D8
PIN:   9   Port: H6        FAN_PIN        ie D9
PIN:  10   Port: B4        HEATER_0_PIN   ie D10

```
and on a real re-arm

```
PIN: 2.07        FAN1_PIN       ie D8
PIN: 2.04        FAN_PIN        ie D9
PIN: 2.05        HEATER_0_PIN   ie D10

```
your controlling it with m106, its a fan. as that gcode can only control fan pins.

KrX3D, 2020-07-16:
big thanks, i connected my spindle to D10 and now it works like it should ;)

i just dont understand why its switched the oppsite like in ramps. also D8 got that chunky connector like for a heatbed, and D10 (which works now) are just some tiny connectors.

it would be nice if the pinout would be switched if thats possible, so it would be like on a ramps

ellensp, 2020-07-17:
You went from EFB  on pins 10,9,8 respectively to EFF still on pins 10,9,8  in that order. Nothing got switched anywhere. 

Now If you want a custom setup where pins on  8 and 10 are deliberately switched around
Add this to your Configuration.h 
```
  #define RAMPS_D8_PIN                     P2_05  // (10)
  #define RAMPS_D10_PIN                   P2_07  // (8)
```
then you will have EFF  on pins 8,9,10

KrX3D, 2020-07-18:
Thanks that worked nicely. so i will go that way. D8 works now with my spindle and also M112 is shutting it down like i wanted

still its a bit odd that the pins are the other way around on the rearm as on the ramps

closed 2020-07-18 by ellensp