MarlinFirmware/Marlin #17851: Compile fails with missing type STEALHCHOP_XY

ibsmith04, 2020-05-03:
Bug Description

I have a new SKR 1.4 turbo, with 2209's and BLTouch. I have it working in klipper, but want to move back to Marlin. Today downloaded latest version of Marlin 2.0.x bugfix, loaded into VSCODE with latest platformio. Updated platformio to LPC1769.
Initially I just updated config.h to set the following
#define SERIAL_PORT -1
#define SERIAL_PORT_2 0
#define BAUDRATE 250000
#define MOTHERBOARD BOARD_BTT_SKR_V1_4_TURBO
I compiled it, and it compiled cleanly.

I then updated config and conf_adv.h to reflect (hopefully) my setup of BLTOUCH, 2209's with senssorless homing.
When I compile I get several errors, all similar
[image]
I eventually found that if I comment out
STEALTHCHOP_XY Z & E and also #define HYBRID_THRESHOLD
then it will compile cleanly.

I am pretty new to all of this, wouldn't be surprised if I have done something stupid.
I did manage to see that I have included/used the latest TMC stepper lib 6.2 , but interestingly when I look at installed library updates, it shows the lib, but it does not appear to include 2209's
[image]

### My Configurations
[IansConfigs.zip](https://github.com/MarlinFirmware/Marlin/files/4570811/IansConfigs.zip)

### Steps to Reproduce

See above]

**Expected behavior:** [What you expect to happen]
Compile cleanly

**Actual behavior:** [What actually happens]
Fails to comple

#### Additional Information

Thank you
Ian

teemuatlut, 2020-05-03:
Configuration_adv.h, lines 2303-2305.
You're missing the hash symbols from defines.

```diff
-  define STEALTHCHOP_XY
-  define STEALTHCHOP_Z
-  define STEALTHCHOP_E
+  #define STEALTHCHOP_XY
+  #define STEALTHCHOP_Z
+  #define STEALTHCHOP_E
```

ibsmith04, 2020-05-03:
Doh !!

Thank you!! I now have a clean compile :-)

sjasonsmith, 2020-05-29:
@ibsmith04, looks like your issue is fixed! Would you mind closing this issue?

closed 2020-05-29 by ibsmith04