MarlinFirmware/Marlin #17852: [BUG] (Z Probe Past Bed no matter coordinates)

Torandir, 2020-05-03:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Torandir, 2020-05-03:
With a z probe enabled, compiling marlin 2.x on windows 10, no matter which coordinates are used the printer cannot home and throws the (Z Probe Past Bed Error). In my case, the probe can never go outisde he print area and I'd like to ideally home at X0, Y0.

Steps To Reproduce
Home Printer (G28)

Should Home to 0 , 0 , 0
Home X and Y, Z probe past Z
[CFG.zip](https://github.com/MarlinFirmware/Marlin/files/4570859/CFG.zip)

grantpeterwilliamson, 2020-05-06:
not 100% sure im follwong how ever looking at your config, have you tried playing around with the section below?

"
// @section homing

// The center of the bed is at (X=0, Y=0)
//#define BED_CENTER_AT_0_0

// Manually set the home position. Leave these undefined for automatic settings.
// For DELTA this is the top-center of the Cartesian print volume.
//#define MANUAL_X_HOME_POS 0
//#define MANUAL_Y_HOME_POS 0
//#define MANUAL_Z_HOME_POS 0

"

the manual z home i recon is what you wanna be looking at.
 good luck!

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle