MarlinFirmware/Marlin #17859: [BUG] Cannot home Z axis with BLTouch probe alone

logiclrd, 2020-05-03:
### Bug Description

With bugfix-2.0.x, I am trying to configure a printer that doesn't have a hard Z stop pin, but uses a BLTouch probe exclusively for locating the bed. As such, I commented out `USE_ZMIN_PLUG`. After doing this, G28 will not home Z at all, which I have tracked down to the definition of `CAN_HOME_Z` in `homeaxis` in `module/motion.cpp`. That macro requires `Z_MIN_PIN` to be set. even though, presumably, with `USE_PROBE_FOR_Z_HOMING` set, it's going to be listening to the probe not the Z min pin for homing purposes.

It occurs to be that it is _possible_ this behaviour is intentional, that a dumb endstop switch is seen as mandatory as a backup for if the sensor fails to trip, to avoid damage to the printer. In the event that that is the case, I believe there should be a build error when `USE_PROBE_FOR_Z_HOMING` is set and `USE_ZMIN_PLUG` is not set to inform the user that you must still have a Z min sensor installed and properly configured, even if you don't plan to use it to home Z.;

### My Configurations

[MarlinConfiguration.zip](https://github.com/MarlinFirmware/Marlin/files/4571505/MarlinConfiguration.zip)

### Steps to Reproduce

Compile with `USE_PROBE_FOR_Z_HOMING` and no `USE_ZMIN_PLUG`, then issue `G28`. After moving the probe to the probing position, the homing operation just stops. I believe it is because of this line in `src/module/motion.cpp`:
```
if (!CAN_HOME_X && !CAN_HOME_Y && !CAN_HOME_Z) return;
```
This aborts the function because `CAN_HOME_Z` evaluates to false, which in turn is because the macro requires `Z_MIN_PIN` to be set.

**Expected behavior:**
Marlin uses the probe to detect when the bed has reached Z home.

**Actual behavior:**
Marlin aborts the homing process without lowering Z at all.

logiclrd, 2020-05-03:
In my local build, I changed the definition of `CAN_HOME_Z` in `homeaxis` in `src/module/motion.cpp` to this:
```
      #define CAN_HOME_Z_WITH_ENDSTOP _CAN_HOME(Z)
      #define CAN_HOME_Z_WITH_PROBE \
        ((axis == Z_AXIS) && ENABLED(USE_PROBE_FOR_Z_HOMING) && Z_HOME_DIR < 0 && Z_MIN_PROBE_PIN > -1)

      #define CAN_HOME_Z (CAN_HOME_Z_WITH_ENDSTOP || CAN_HOME_Z_WITH_PROBE)
```
With this change, `G28` now homes Z using my BLTouch sensor. Is this an appropriate change?

sjasonsmith, 2020-05-29:
I think most of us just have our probes connected to the Z_MIN connector. This has been a recurring topic, I'm not sure whether changes have been made to make that no longer necessary.

logiclrd, 2020-05-29:
From my investigating, no such changes have yet been made, but I proposed changes that allowed it to work with the dedicated SKR 1.4 pins on my local build. Maybe I'll just open up a PR with that change to get more eyes on it, and then it can be determined whether it is the right change or not.

logiclrd, 2020-05-29:
https://github.com/MarlinFirmware/Marlin/pull/18150

closed 2020-05-31 by thinkyhead