MarlinFirmware/Marlin #17860: [FR]avr-gcc: error: CreateProcess: No such file or directory

Tchicken95, 2020-05-03:
Hello,

Could you help me, I don't understand the following error (Compilation verbose) :
Linking everything together...
"D:\\Da Vinci\\0-DocumentationLogiciels\\arduino-nightly\\hardware\\tools\\avr/bin/avr-gcc" -w -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections,--relax -mmcu=atmega2560 -o "C:\\Users\\Mario\\AppData\\Local\\Temp\\arduino_build_696616/Marlin.ino.elf" "sketch\\Marlin.ino.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\ctrl_access.c.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\sysclk.c.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\udc.c.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\udi_cdc.c.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\udi_cdc_desc.c.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\udi_composite_desc.c.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\udi_msc.c.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\uotghs_device_due.c.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\usb_task.c.o" "sketch\\src\\HAL\\HAL_LPC1768\\include\\digipot_mcp4451_I2C_routines.c.o" "sketch\\src\\HAL\\HAL_LPC1768\\include\\i2c_util.c.o" "sketch\\src\\HAL\\HAL_LPC1768\\u8g\\LCD_pin_routines.c.o" "sketch\\src\\HAL\\HAL_AVR\\HAL.cpp.o" "sketch\\src\\HAL\\HAL_AVR\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\HAL_AVR\\MarlinSerial.cpp.o" "sketch\\src\\HAL\\HAL_AVR\\Servo.cpp.o" "sketch\\src\\HAL\\HAL_AVR\\fast_pwm.cpp.o" "sketch\\src\\HAL\\HAL_AVR\\fastio.cpp.o" "sketch\\src\\HAL\\HAL_AVR\\persistent_store_eeprom.cpp.o" "sketch\\src\\HAL\\HAL_AVR\\u8g_com_HAL_AVR_sw_spi.cpp.o" "sketch\\src\\HAL\\HAL_AVR\\watchdog.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\DebugMonitor.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\EepromEmulation.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\HAL.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\InterruptVectors.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\MarlinSerial.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\MarlinSerialUSB.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\Servo.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\Tone.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\dogm\\u8g_com_HAL_DUE_shared_hw_spi.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\dogm\\u8g_com_HAL_DUE_st7920_sw_spi.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\dogm\\u8g_com_HAL_DUE_sw_spi.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\dogm\\u8g_com_HAL_DUE_sw_spi_shared.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\fastio\\G2_PWM.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\persistent_store_eeprom.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\timers.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\sd_mmc_spi_mem.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\watchdog.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\FlushableHardwareSerial.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\HAL.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\Servo.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\WebSocketSerial.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\i2s.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\ota.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\persistent_store_impl.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\spiffs.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\timers.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\watchdog.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\web.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\wifi.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\HAL.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\arduino.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\hardware\\Clock.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\hardware\\Gpio.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\hardware\\Heater.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\hardware\\IOLoggerCSV.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\hardware\\LinearAxis.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\hardware\\Timer.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\include\\pinmapping.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\main.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\persistent_store_impl.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\timers.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\watchdog.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\DebugMonitor.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\HAL.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\MarlinSerial.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\fast_pwm.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\main.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\persistent_store_flash.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\persistent_store_sdcard.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\timers.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\u8g\\LCD_I2C_routines.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\u8g\\u8g_com_HAL_LPC1768_hw_spi.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\u8g\\u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\u8g\\u8g_com_HAL_LPC1768_st7920_hw_spi.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\u8g\\u8g_com_HAL_LPC1768_st7920_sw_spi.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\u8g\\u8g_com_HAL_LPC1768_sw_spi.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\usb_serial.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\watchdog.cpp.o" "sketch\\src\\HAL\\HAL_SAMD51\\HAL.cpp.o" "sketch\\src\\HAL\\HAL_SAMD51\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\HAL_SAMD51\\MarlinSerial_AGCM4.cpp.o" "sketch\\src\\HAL\\HAL_SAMD51\\Servo.cpp.o" "sketch\\src\\HAL\\HAL_SAMD51\\persistent_store_eeprom.cpp.o" "sketch\\src\\HAL\\HAL_SAMD51\\timers.cpp.o" "sketch\\src\\HAL\\HAL_SAMD51\\watchdog.cpp.o" "sketch\\src\\HAL\\HAL_STM32F1\\HAL.cpp.o" "sketch\\src\\HAL\\HAL_STM32F1\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\HAL_STM32F1\\SPI.cpp.o" "sketch\\src\\HAL\\HAL_STM32F1\\Servo.cpp.o" "sketch\\src\\HAL\\HAL_STM32F1\\SoftwareSerial.cpp.o" "sketch\\src\\HAL\\HAL_STM32F1\\dogm\\u8g_com_stm32duino_fsmc.cpp.o" "sketch\\src\\HAL\\HAL_STM32F1\\dogm\\u8g_com_stm32duino_swspi.cpp.o" "sketch\\src\\HAL\\HAL_STM32F1\\msc_sd.cpp.o" "sketch\\src\\HAL\\HAL_STM32F1\\onboard_sd.cpp.o" "sketch\\src\\HAL\\HAL_STM32F1\\persistent_store_eeprom.cpp.o" "sketch\\src\\HAL\\HAL_STM32F1\\persistent_store_flash.cpp.o" "sketch\\src\\HAL\\HAL_STM32F1\\persistent_store_sdcard.cpp.o" "sketch\\src\\HAL\\HAL_STM32F1\\sdio.cpp.o" "sketch\\src\\HAL\\HAL_STM32F1\\timers.cpp.o" "sketch\\src\\HAL\\HAL_STM32F1\\watchdog.cpp.o" "sketch\\src\\HAL\\HAL_STM32\\HAL.cpp.o" "sketch\\src\\HAL\\HAL_STM32\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\HAL_STM32\\Sd2Card_sdio_stm32duino.cpp.o" "sketch\\src\\HAL\\HAL_STM32\\Servo.cpp.o" "sketch\\src\\HAL\\HAL_STM32\\SoftwareSerial.cpp.o" "sketch\\src\\HAL\\HAL_STM32\\fastio.cpp.o" "sketch\\src\\HAL\\HAL_STM32\\persistent_store_flash.cpp.o" "sketch\\src\\HAL\\HAL_STM32\\persistent_store_impl.cpp.o" "sketch\\src\\HAL\\HAL_STM32\\persistent_store_sdcard.cpp.o" "sketch\\src\\HAL\\HAL_STM32\\timers.cpp.o" "sketch\\src\\HAL\\HAL_STM32\\watchdog.cpp.o" "sketch\\src\\HAL\\HAL_STM32_F4_F7\\EmulatedEeprom.cpp.o" "sketch\\src\\HAL\\HAL_STM32_F4_F7\\HAL.cpp.o" "sketch\\src\\HAL\\HAL_STM32_F4_F7\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\HAL_STM32_F4_F7\\STM32F4\\timers.cpp.o" "sketch\\src\\HAL\\HAL_STM32_F4_F7\\STM32F7\\TMC2660.cpp.o" "sketch\\src\\HAL\\HAL_STM32_F4_F7\\STM32F7\\timers.cpp.o" "sketch\\src\\HAL\\HAL_STM32_F4_F7\\Servo.cpp.o" "sketch\\src\\HAL\\HAL_STM32_F4_F7\\eeprom_emul.cpp.o" "sketch\\src\\HAL\\HAL_STM32_F4_F7\\persistent_store_eeprom.cpp.o" "sketch\\src\\HAL\\HAL_STM32_F4_F7\\watchdog.cpp.o" "sketch\\src\\HAL\\HAL_TEENSY31_32\\HAL.cpp.o" "sketch\\src\\HAL\\HAL_TEENSY31_32\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\HAL_TEENSY31_32\\Servo.cpp.o" "sketch\\src\\HAL\\HAL_TEENSY31_32\\persistent_store_impl.cpp.o" "sketch\\src\\HAL\\HAL_TEENSY31_32\\timers.cpp.o" "sketch\\src\\HAL\\HAL_TEENSY31_32\\watchdog.cpp.o" "sketch\\src\\HAL\\HAL_TEENSY35_36\\HAL.cpp.o" "sketch\\src\\HAL\\HAL_TEENSY35_36\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\HAL_TEENSY35_36\\Servo.cpp.o" "sketch\\src\\HAL\\HAL_TEENSY35_36\\persistent_store_eeprom.cpp.o" "sketch\\src\\HAL\\HAL_TEENSY35_36\\timers.cpp.o" "sketch\\src\\HAL\\HAL_TEENSY35_36\\watchdog.cpp.o" "sketch\\src\\HAL\\shared\\HAL_spi_L6470.cpp.o" "sketch\\src\\HAL\\shared\\backtrace\\backtrace.cpp.o" "sketch\\src\\HAL\\shared\\backtrace\\unwarm.cpp.o" "sketch\\src\\HAL\\shared\\backtrace\\unwarm_arm.cpp.o" "sketch\\src\\HAL\\shared\\backtrace\\unwarm_thumb.cpp.o" "sketch\\src\\HAL\\shared\\backtrace\\unwarmbytab.cpp.o" "sketch\\src\\HAL\\shared\\backtrace\\unwarmmem.cpp.o" "sketch\\src\\HAL\\shared\\backtrace\\unwinder.cpp.o" "sketch\\src\\HAL\\shared\\backtrace\\unwmemaccess.cpp.o" "sketch\\src\\HAL\\shared\\eeprom_i2c.cpp.o" "sketch\\src\\HAL\\shared\\eeprom_spi.cpp.o" "sketch\\src\\HAL\\shared\\persistent_store_api.cpp.o" "sketch\\src\\HAL\\shared\\servo.cpp.o" "sketch\\src\\MarlinCore.cpp.o" "sketch\\src\\core\\multi_language.cpp.o" "sketch\\src\\core\\serial.cpp.o" "sketch\\src\\core\\utility.cpp.o" "sketch\\src\\feature\\I2CPositionEncoder.cpp.o" "sketch\\src\\feature\\Max7219_Debug_LEDs.cpp.o" "sketch\\src\\feature\\babystep.cpp.o" "sketch\\src\\feature\\ [... cut ...] ies\\TMCStepper\\source\\TMC2209Stepper.cpp.o" "libraries\\TMCStepper\\source\\TMC2660Stepper.cpp.o" "libraries\\TMCStepper\\source\\TMC5130Stepper.cpp.o" "libraries\\TMCStepper\\source\\TMC5160Stepper.cpp.o" "libraries\\TMCStepper\\source\\TMCStepper.cpp.o" "libraries\\SPI\\SPI.cpp.o" "libraries\\SoftwareSerial\\SoftwareSerial.cpp.o" "core\\core.a" "-LC:\\Users\\Mario\\AppData\\Local\\Temp\\arduino_build_696616" -lm
avr-gcc: error: CreateProcess: No such file or directory

Thanks for your help.

ellensp, 2020-05-04:
This is a windows Process limit. Windows will not launch a process that has a command line over 32k long.  You need to reduce the size of the final link command or use vscode + platformio which gets around this issue in other ways.

You can reduce the length a number of ways.
1) move marlin to the root of a drive and rename it M
2) delete unused HALS (don' t removed shared!) eg is your building for a DUE you can safely delete directories HAL_TEENSY31_32, HAL_STM32F1 etc
3) if you not using the extensible_ui delete that directory.

In newer version of marlin, some of these long names like extensible_ui have been shorted to try and stop this issue.

Tchicken95, 2020-05-04:
Thank you, that was my problem.

Tchicken95, 2020-05-04:
Big thanks

closed 2020-05-04 by Tchicken95