MarlinFirmware/Marlin #17862: SKR Mini v1.2 upload

byxox, 2020-05-03:
Hello
I have a skr mini e3 v1.2 card. I downloaded the compatible marlin on the site. I have built seamlessly with Platformio. But when I upload through the platform, I get this error.

https://prnt.sc/sacei7
https://prnt.sc/sacep8
https://prnt.sc/sacexl

what could I be missing?

I can see the device as a serial port on the platform.
https://prnt.sc/sacfal

sjasonsmith, 2020-05-03:
The SKR Mini E3 typically expects you to copy the firmware.bin file to the SD card, and the bootloader will flash the firmware from it on the next power-cycle.

Using the upload commands wants to write firmware using a separate debug adapter called an ST-Link.

byxox, 2020-05-04:
Yeah. As you said, it was updated when firmware.bin was thrown on the sd card. Thank you so much.

closed 2020-05-04 by byxox