MarlinFirmware/Marlin #17863: [FR] (E3 E3D is set separately in the platfromio.ini file and support Neopiexl lights)

makerbase-mks, 2020-05-04:
### Description
- MKS Robin E3 E3D are different motherboards
- They can be set individually in the platformio.ini file
- Support Neopiexl lights

### Feature Workflow

1. Add env: mks_robin_e3d in platformio.ini file
2. MKS Robin E3 and E3D board support Neopiexl lights, need remove Adafruit NeoPixel in lib_ignore and add "Adafruit NeoPixel=https://github.com/makerbase-mks/Adafruit_NeoPixel/archive/master.zip" to lib_deps path
[image]

#### Additional Information
MKS Robin E3 and E3D platfromio.ini set link: https://github.com/makerbase-mks/MKS-Robin-E3-E3D/blob/master/firmware/marlin2.0%20for%20Ender3/platformio.ini

thisiskeithb, 2020-05-04:
It looks like both use the same MCU and build scripts, so what benefit would splitting up build environments bring?

You could also rename the environment to `STM32F103RC_mks` similar to what was done for the BigTreeTech SKR Mini E3 boards that share the same MCU to make it more generalized.

https://github.com/MarlinFirmware/Marlin/blob/5e23fd6ef1bfda86fb79ede937b62e45085ea590/platformio.ini#L304-L371

Updating the NeoPixel exclude would be easy enough to do.

sjasonsmith, 2020-09-11:
@makerbase-mks is this still an issue? I assume this is work you could contribute yourself in a pull request?
I agree that they should continue sharing the same environment, but if there are changes needed it seems like you may be the best person to do so?

sjasonsmith, 2020-11-12:
These boards are present in the platformio file. I'm going to call this a duplicate of another NeoPixel feature request for STM32F1 boards.

sjasonsmith, 2020-11-12:
Duplicate of #19800

closed 2020-11-12 by sjasonsmith