MarlinFirmware/Marlin #17866: [FR] KILL_PIN invertion

AbdullahGheith, 2020-05-04:
### Description

kill() is called when KILL_PIN is LOW. Add a config to invert this functionality

#### Additional Information

I have a momentary switch connected to a relay . Theres a pin going from the relay that becomes HIGH when button is pressed. I can turn off the printer (Desired functionality) if i reverse this if statement in MarlinCore.cpp:
if (!READ(KILL_PIN)) //Ie. if i remove the !

As i do not wish to edit MarlinCore, i want this to be configurable .

thinkyhead, 2020-05-05:
852a8d6685ec1137eb65e78fa748cae41fbd36b6

closed 2020-05-05 by thinkyhead

AbdullahGheith, 2020-05-05:
Is there a better solution to my problem?
It worked for a while, but the printer will turn off mid print as if the switch has been pressed

thinkyhead, 2020-05-05:
To change the KILL pin triggered state from LOW to HIGH add to `Configuration.h`:
```cpp
#define KILL_PIN_STATE HIGH
```

thinkyhead, 2020-05-05:
You may need to add a proper pull-down resistor between your switch and ground. On AVR there are no pull-downs only pull-ups.

AbdullahGheith, 2020-05-05:
> [852a8d6](https://github.com/MarlinFirmware/Marlin/commit/852a8d6685ec1137eb65e78fa748cae41fbd36b6)

Thx for this btw. I tried to contribute to the project, but I think your changes are better: https://github.com/MarlinFirmware/Marlin/pull/17869