MarlinFirmware/Marlin #17868: [FR] Alternative UART for GCode file handling

tiwanacote, 2020-05-04:
**The project:
--------------------------------------------------------------------------------------------------------------------**
Hi, some friends and me are planning to **use the low cost ESP32-CAM** board in an open source project for **GCode streaming but not as TCP-UART bridge**, as many projects developed. We would like to use a server in order to send a batch from web browser interface, allocate it in a ESP32 ring stack buffer and finally get a remote control printer from outside the local WLAN  

**The project will allow you do remotely:**
**- Print GCode file remotely**
**- Send simple GCodes instruction for control the printer as many PC host does**
**- Send Pictures or short streaming captured with cam (Nothing to do with Marling)**
**- Allow to put printers in farm (Nothing to do with Marling)**

Also you can take advantage of **ESP32-CAM (7 USD)** possibilities and controll a for example the ILI9488 (12 USD) touch screen at the same time, using the ESP32 as controller/bridge between TFT and board

**The Question:
--------------------------------------------------------------------------------------------------------------------**
We know many motherboard has 2 UARTS, generally the first one used for USB connection and firmware update.
**We would like to use the second UART in order to send and use the same protocol as Repetier, Pronterface and others host use, but througth the ESP32**

**It is implemented into Marling the setup of two UARTs for GCode sending?** 

If not, what could be the best way to do it?
(May be detectinng USB_STATUS and changing the UART to hear if not computer’s USB conected? … or using  LCD_SERIAL or BLUETOOTH_SERIAL)
Any kick start could be usefull for us… 

Thank you

AnHardt, 2020-05-04:
At first learn spelling **Marlin**.

> It is implemented into Marling the setup of two UARTs for GCode sending?

Yes. Did you recently have a look into Marlins Configuration.h file? Unlikely!

You are not ready for that project.

GMagician, 2020-05-04:
> Hi, some friends and me are planning to use the low cost ESP32-CAM board in an open source project for GCode streaming but not as TCP-UART bridge, as many projects developed

do you want to develop something like https://github.com/probonopd/WirelessPrinting?

sjasonsmith, 2020-09-11:
This feature already exists. Marlin supports processing GCode from two serial ports. This is how every serial-connected display or ESP Wifi module works today.

sjasonsmith, 2020-09-11:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-09-11 by sjasonsmith