MarlinFirmware/Marlin #17870: AUTO_BED_LEVELING_3POINT manual probe positions ignored

nick9one1, 2020-05-04:
### Bug Description

When using AUTO_BED_LEVELING_3POINT and setting the XY in

```
/**
 * Points to probe for all 3-point Leveling procedures.
 * Override if the automatically selected points are inadequate.
 */

```
manual probe positions are ignored. 

the standard 3 point probe positions cause the furthest Y coordinate to probe off the back of my bed. 

G1 X15 Y125 is as far back as I can go. 

when the manaual probe position is set to 

```
/**
 * Points to probe for all 3-point Leveling procedures.
 * Override if the automatically selected points are inadequate.
 */
#if EITHER(AUTO_BED_LEVELING_3POINT, AUTO_BED_LEVELING_UBL)
  //#define PROBE_PT_1_X 15
  #define PROBE_PT_1_Y 125
  //#define PROBE_PT_2_X 15
  //#define PROBE_PT_2_Y 20
  //#define PROBE_PT_3_X 170
  //#define PROBE_PT_3_Y 20
  #endif
```

G29 ignores #define PROBE_PT_1_Y 125 and still probes at Y180

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

SwiftingSpeed, 2020-05-08:
The conditional check needs all three points to be defined fully (both X and Y), you can't define part of them.
https://github.com/MarlinFirmware/Marlin/blob/36bab2b30e11817377bde66650a1404d574b16a4/Marlin/src/inc/Conditionals_post.h#L1995

thinkyhead, 2020-05-10:
Now patched with a better sanity check.

commit 86c11253 (2020-05-10): Sanity check 3-point defines

Fixes #17870
closed 2020-05-10 by thinkyhead