MarlinFirmware/Marlin #17871: [FR] Generate STRING_DISTRIBUTION_DATE at compile time

Taomyn, 2020-05-04:
### Description

Would there be a benefit in changing the setting STRING_DISTRIBUTION_DATE in version.h to be generated at compile time? After all the description does state "represents when the binary file was built" and the default at the moment will never be that.

### Feature Workflow

Change
//#define STRING_DISTRIBUTION_DATE "2019-07-10"

to
//#define STRING_DISTRIBUTION_DATE __DATE__ " " __TIME__

#### Additional Information

I'm using VSCode for my builds so I don't know if this is specific to that, but I find this handy to confirm on the printer info screen that the firmware I uploaded really has been flashed, hence adding the time value as well.

Evg33, 2020-05-04:
IMHO not good idea.
May be better add compilation time ?

Powerprobot, 2020-05-04:
I add for my personal own in the file MENU_INFO.CPP
at Row 245
`    STATIC_ITEM_P(PSTR(STRING_CONFIG_H_AUTHOR));                // Author of config
    STATIC_ITEM_P(PSTR(__TIME__));                              // Time of compile
`
this works fine for me.

randellhodges, 2020-05-04:
I added an extra script to do something similar and to include the branch (since I have a branch for every config)

https://github.com/randellhodges/Marlin/blob/skr-1.4/buildroot/share/PlatformIO/scripts/common-version.py

https://github.com/randellhodges/Marlin/blob/f31919ceee21212364603c9cb7352de885adf60b/platformio.ini#L262

thisiskeithb, 2020-05-05:
> IMHO not good idea.
> May be better add compilation time ?

I agree that `STRING_DISTRIBUTION_DATE` should remain the same since that’s an easy way to tell how old a copy of Marlin is, or at least when it was forked.

Adding a 
New define like `STRING_COMPILED_DATE __DATE__ " " __TIME__` (and making it optional) would be a better approach.

Taomyn, 2020-05-05:
@thisiskeithb I agree if the description of the current define is wrong i.e. "represents when the binary file was built", but can this also be made part of the info display as well?

Also, as you state "since that’s an easy way to tell how old a copy of Marlin is, or at least when it was forked" what is "2019-07-10" referring to?

thisiskeithb, 2020-05-05:
> Also, as you state "since that’s an easy way to tell how old a copy of Marlin is, or at least when it was forked" what is "2019-07-10" referring to?

There’s a cron job that bumps `STRING_DISTRIBUTION_DATE` in this repo every day. See https://github.com/MarlinFirmware/Marlin/commit/7655620fd0ba3c2f3d38eb71d5c0d7fa7a56c181 for example.

Taomyn, 2020-05-05:
@thisiskeithb doesn't seem to change in the bugfix zip files I download, the one I did 3rd May at the weekend, version.h looks like this

[image]

I downloaded it again and it's still the same, I'm I downloading the zip incorrectly?

[image]

thisiskeithb, 2020-05-05:
The one that is updated is [Marlin/src/inc/Version.h](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/inc/Version.h), not [Marlin/Version.h](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/Version.h).

The latter is used by vendors to override the defaults with a `CUSTOM_VERSION_FILE` define in `Configuration.h`.

Taomyn, 2020-05-05:
@thisiskeithb ah sorry, my bad. Thanks for the explanation.

Evg33, 2020-05-21:
```
>M115
<FIRMWARE_NAME:Marlin bugfix-2.0.x (May 21 2020 03:46:30) ....
```

closed 2021-09-24 by thisiskeithb