MarlinFirmware/Marlin #17872: Marlin - ABL - BLTOUCH 3.1

thierryzoller, 2020-05-04:
As promised, following the many ABL & BLTouch issues, I proceeded to debug mine.

- Fresh bugfix2.0 build
- Config attached
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/4574687/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/4574688/Configuration_adv.txt)

thierryzoller, 2020-05-04:
Issued a G28 : 

`2020-05-04 14:38:01,832 - RECV: >>> G28
2020-05-04 14:38:01,832 - RECV: Machine Type: Cartesian
2020-05-04 14:38:01,842 - RECV: Probe: BLTOUCH
2020-05-04 14:38:01,848 - RECV: Probe Offset X28.00 Y-33.00 Z-3.05 (Right-Front & Below Nozzle)
2020-05-04 14:38:01,848 - RECV: Auto Bed Leveling: BILINEAR (disabled)
2020-05-04 14:38:01,858 - RECV: Raise Z (before homing) to 4.00
2020-05-04 14:38:01,858 - RECV: >>> do_blocking_move_to X0.00 Y0.00 Z4.00
2020-05-04 14:38:01,858 - RECV: <<< do_blocking_move_to
2020-05-04 14:38:02,336 - RECV: >>> homeaxis(X)
2020-05-04 14:38:02,336 - RECV: Home 1 Fast:
2020-05-04 14:38:02,345 - RECV: >>> do_homing_move(X, -465.00, [40.00])
2020-05-04 14:38:03,398 - SENT: M105
2020-05-04 14:38:03,835 - RECV: echo:busy: processing
2020-05-04 14:38:05,634 - RECV: <<< do_homing_move(X)
2020-05-04 14:38:05,634 - RECV: Move Away:
2020-05-04 14:38:05,644 - RECV: >>> do_homing_move(X, 5.00, [40.00])
2020-05-04 14:38:05,831 - RECV: echo:busy: processing
2020-05-04 14:38:05,881 - RECV: <<< do_homing_move(X)
2020-05-04 14:38:05,881 - RECV: Home 2 Slow:
2020-05-04 14:38:05,891 - RECV: >>> do_homing_move(X, -10.00, 20.00)
2020-05-04 14:38:06,273 - RECV: <<< do_homing_move(X)
2020-05-04 14:38:06,315 - RECV: >>> set_axis_is_at_home(X)
2020-05-04 14:38:06,361 - RECV: Axis 88 home_offset = 0.00 position_shift = 0.00
2020-05-04 14:38:06,361 - RECV: > home_offset[X] = 0.00
2020-05-04 14:38:06,361 - RECV:   current_position= X0.00 Y0.00 Z4.00 :
2020-05-04 14:38:06,374 - RECV: <<< set_axis_is_at_home(X)
2020-05-04 14:38:06,375 - RECV:   current_position= X0.00 Y0.00 Z4.00 : sync_plan_position
2020-05-04 14:38:06,375 - RECV:   current_position= X0.00 Y0.00 Z4.00 : > AFTER set_axis_is_at_home
2020-05-04 14:38:06,375 - RECV: <<< homeaxis(X)
2020-05-04 14:38:06,375 - RECV: >>> homeaxis(Y)
2020-05-04 14:38:06,375 - RECV: Home 1 Fast:
2020-05-04 14:38:06,385 - RECV: >>> do_homing_move(Y, -465.00, [40.00])
2020-05-04 14:38:07,835 - RECV: echo:busy: processing
2020-05-04 14:38:09,838 - RECV: echo:busy: processing
2020-05-04 14:38:11,075 - RECV: <<< do_homing_move(Y)
2020-05-04 14:38:11,085 - RECV: Move Away:
2020-05-04 14:38:11,085 - RECV: >>> do_homing_move(Y, 5.00, [40.00])
2020-05-04 14:38:11,323 - RECV: <<< do_homing_move(Y)
2020-05-04 14:38:11,323 - RECV: Home 2 Slow:
2020-05-04 14:38:11,334 - RECV: >>> do_homing_move(Y, -10.00, 20.00)
2020-05-04 14:38:11,687 - RECV: <<< do_homing_move(Y)
2020-05-04 14:38:11,687 - RECV: >>> set_axis_is_at_home(Y)
2020-05-04 14:38:11,687 - RECV: Axis 89 home_offset = 0.00 position_shift = 0.00
2020-05-04 14:38:11,697 - RECV: > home_offset[Y] = 0.00
2020-05-04 14:38:11,697 - RECV:   current_position= X0.00 Y0.00 Z4.00 :
2020-05-04 14:38:11,707 - RECV: <<< set_axis_is_at_home(Y)
2020-05-04 14:38:11,707 - RECV:   current_position= X0.00 Y0.00 Z4.00 : sync_plan_position
2020-05-04 14:38:11,716 - RECV:   current_position= X0.00 Y0.00 Z4.00 : > AFTER set_axis_is_at_home
2020-05-04 14:38:11,716 - RECV: <<< homeaxis(Y)
2020-05-04 14:38:11,716 - RECV: BLTouch Command :160
2020-05-04 14:38:12,278 - RECV: BLTouch Command :90
2020-05-04 14:38:13,048 - RECV: last_written_mode - 1
2020-05-04 14:38:13,058 - RECV: config mode - BLTOUCH_SET_5V_MODE
2020-05-04 14:38:13,058 - RECV: home_z_safely >>>
2020-05-04 14:38:13,058 - RECV:   current_position= X0.00 Y0.00 Z4.00 : sync_plan_position
2020-05-04 14:38:13,073 - RECV:   destination= X127.00 Y188.00 Z7.05 : home_z_safely
2020-05-04 14:38:13,073 - RECV: >>> do_blocking_move_to X127.00 Y188.00 Z4.00
2020-05-04 14:38:13,084 - RECV: <<< do_blocking_move_to
2020-05-04 14:38:13,084 - RECV: echo:busy: processing
2020-05-04 14:38:14,937 - RECV: >>> homeaxis(Z)
2020-05-04 14:38:14,944 - RECV:   current_position= X127.00 Y188.00 Z4.00 : Probe::set_deployed
2020-05-04 14:38:14,944 - RECV: deploy: 1
2020-05-04 14:38:14,944 - RECV: Probe::move_z(5.00)
2020-05-04 14:38:14,953 - RECV: >>> do_blocking_move_to X127.00 Y188.00 Z8.05
2020-05-04 14:38:14,953 - RECV: <<< do_blocking_move_to
2020-05-04 14:38:15,063 - RECV: echo:busy: processing
2020-05-04 14:38:15,444 - RECV: BLTouch DEPLOY requested
2020-05-04 14:38:15,444 - RECV: BLTouch Command :10
[... 23 lines cut ...]
2020-05-04 14:38:20,134 - RECV: > probe.offset.z = -3.05
2020-05-04 14:38:20,140 - RECV: Axis 90 home_offset = 0.00 position_shift = 0.00
2020-05-04 14:38:20,140 - RECV: > home_offset[Z] = 0.00
2020-05-04 14:38:20,150 - RECV:   current_position= X127.00 Y188.00 Z3.05 :
2020-05-04 14:38:20,150 - RECV: <<< set_axis_is_at_home(Z)
2020-05-04 14:38:20,160 - RECV:   current_position= X127.00 Y188.00 Z3.05 : sync_plan_position
2020-05-04 14:38:20,174 - RECV:   current_position= X127.00 Y188.00 Z3.05 : > AFTER set_axis_is_at_home
2020-05-04 14:38:20,174 - RECV:   current_position= X127.00 Y188.00 Z3.05 : Probe::set_deployed
2020-05-04 14:38:20,174 - RECV: deploy: 0
2020-05-04 14:38:20,184 - RECV: Probe::move_z(5.00)
2020-05-04 14:38:20,184 - RECV: >>> do_blocking_move_to X127.00 Y188.00 Z8.05
2020-05-04 14:38:20,184 - RECV: <<< do_blocking_move_to
2020-05-04 14:38:20,690 - RECV: BLTouch STOW requested
2020-05-04 14:38:20,690 - RECV: BLTouch Command :90
2020-05-04 14:38:21,486 - RECV: bltouch.stow_proc() end
2020-05-04 14:38:21,496 - RECV: >>> do_blocking_move_to X127.00 Y188.00 Z8.05
2020-05-04 14:38:21,496 - RECV: <<< do_blocking_move_to
2020-05-04 14:38:21,496 - RECV: <<< homeaxis(Z)
2020-05-04 14:38:21,496 - RECV: <<< home_z_safely
2020-05-04 14:38:21,506 - RECV: >>> do_blocking_move_to X127.00 Y188.00 Z15.00
2020-05-04 14:38:21,506 - RECV: <<< do_blocking_move_to
2020-05-04 14:38:21,516 - RECV: echo:busy: processing
2020-05-04 14:38:22,144 - RECV:   current_position= X127.00 Y188.00 Z15.00 : sync_plan_position
2020-05-04 14:38:22,154 - RECV: X:127.00 Y:188.00 Z:15.00 E:0.00 Count X:10160 Y:15040 Z:6000
2020-05-04 14:38:22,154 - RECV: <<< G28`

thierryzoller, 2020-05-04:
Issued a G29 : 

> 2020-05-04 14:38:34,161 - SENT: G29
2020-05-04 14:38:34,171 - RECV: ok T:20.86 /0.00 B:20.70 /0.00 @:0 B@:0
2020-05-04 14:38:34,171 - RECV: echo:G29
2020-05-04 14:38:34,184 - RECV:   current_position= X127.00 Y188.00 Z15.00 : >>> G29
2020-05-04 14:38:34,187 - RECV: Machine Type: Cartesian
2020-05-04 14:38:34,187 - RECV: Probe: BLTOUCH
2020-05-04 14:38:34,197 - RECV: Probe Offset X28.00 Y-33.00 Z-3.05 (Right-Front & Below Nozzle)
2020-05-04 14:38:34,197 - RECV: Auto Bed Leveling: BILINEAR (disabled)
2020-05-04 14:38:34,207 - RECV:   current_position= X127.00 Y188.00 Z15.00 : Probe::set_deployed
2020-05-04 14:38:34,207 - RECV: deploy: 1
2020-05-04 14:38:34,207 - RECV: Probe::move_z(5.00)
2020-05-04 14:38:34,220 - RECV: BLTouch DEPLOY requested
2020-05-04 14:38:34,226 - RECV: BLTouch Command :10
2020-05-04 14:38:34,236 - SENT: M105
2020-05-04 14:38:34,980 - RECV: bltouch.deploy_proc() end
2020-05-04 14:38:34,990 - RECV: >>> do_blocking_move_to X127.00 Y188.00 Z15.00
2020-05-04 14:38:34,990 - RECV: <<< do_blocking_move_to
2020-05-04 14:38:35,000 - RECV: >>> Probe::probe_at_point(28.00, 25.00, raise, 0, probe_relative)
2020-05-04 14:38:35,010 - RECV:   current_position= X127.00 Y188.00 Z15.00 :
2020-05-04 14:38:35,016 - RECV: >>> do_blocking_move_to X0.00 Y58.00 Z15.00
2020-05-04 14:38:35,016 - RECV: <<< do_blocking_move_to
2020-05-04 14:38:36,167 - RECV: echo:busy: processing
2020-05-04 14:38:36,538 - RECV:   current_position= X0.00 Y58.00 Z15.00 : Probe::set_deployed
2020-05-04 14:38:36,538 - RECV: deploy: 1
2020-05-04 14:38:36,548 - RECV:   current_position= X0.00 Y58.00 Z15.00 : >>> Probe::run_z_probe
2020-05-04 14:38:36,548 - RECV:   current_position= X0.00 Y58.00 Z15.00 : >>> Probe::probe_down_to_z
2020-05-04 14:38:36,558 - RECV: >>> do_blocking_move_to X0.00 Y58.00 Z1.05
2020-05-04 14:38:36,558 - RECV: <<< do_blocking_move_to
2020-05-04 14:38:37,441 - RECV:   current_position= X0.00 Y58.00 Z2.99 : sync_plan_position
2020-05-04 14:38:37,503 - RECV:   current_position= X0.00 Y58.00 Z2.99 : <<< Probe::probe_down_to_z
2020-05-04 14:38:37,503 - RECV: 1st Probe Z:2.99
2020-05-04 14:38:37,513 - RECV: >>> do_blocking_move_to X0.00 Y58.00 Z7.99
2020-05-04 14:38:37,515 - RECV: <<< do_blocking_move_to
2020-05-04 14:38:37,904 - RECV:   current_position= X0.00 Y58.00 Z7.99 : >>> Probe::probe_down_to_z
2020-05-04 14:38:37,917 - RECV: >>> do_blocking_move_to X0.00 Y58.00 Z1.05
2020-05-04 14:38:37,920 - RECV: <<< do_blocking_move_to
2020-05-04 14:38:38,170 - RECV: echo:busy: processing
2020-05-04 14:38:38,561 - RECV:   current_position= X0.00 Y58.00 Z2.99 : sync_plan_position
2020-05-04 14:38:38,571 - RECV:   current_position= X0.00 Y58.00 Z2.99 : <<< Probe::probe_down_to_z
2020-05-04 14:38:38,571 - RECV: 2nd Probe Z:2.99 Discrepancy:0.00
2020-05-04 14:38:38,585 - RECV:   current_position= X0.00 Y58.00 Z2.99 : <<< run_z_probe
2020-05-04 14:38:38,585 - RECV: >>> do_blocking_move_to X0.00 Y58.00 Z6.99
2020-05-04 14:38:38,595 - RECV: <<< do_blocking_move_to
2020-05-04 14:38:39,062 - RECV: <<< Probe::probe_at_point
2020-05-04 14:38:39,072 - RECV: >>> Probe::probe_at_point(70.83, 25.00, raise, 0, probe_relative)
2020-05-04 14:38:39,072 - RECV:   current_position= X0.00 Y58.00 Z6.99 :
2020-05-04 14:38:39,088 - RECV: >>> do_blocking_move_to X42.83 Y58.00 Z6.99
2020-05-04 14:38:39,088 - RECV: <<< do_blocking_move_to
2020-05-04 14:38:39,573 - RECV:   current_position= X42.83 Y58.00 Z6.99 : Probe::set_deployed
2020-05-04 14:38:39,573 - RECV: deploy: 1
2020-05-04 14:38:39,573 - RECV:   current_position= X42.83 Y58.00 Z6.99 : >>> Probe::run_z_probe
2020-05-04 14:38:39,584 - RECV:   current_position= X42.83 Y58.00 Z6.99 : >>> Probe::probe_down_to_z
2020-05-04 14:38:39,584 - RECV: >>> do_blocking_move_to X42.83 Y58.00 Z1.05
2020-05-04 14:38:39,594 - RECV: <<< do_blocking_move_to
2020-05-04 14:38:39,990 - RECV:   current_position= X42.83 Y58.00 Z2.93 : sync_plan_position
2020-05-04 14:38:40,000 - RECV:   current_position= X42.83 Y58.00 Z2.93 : <<< Probe::probe_down_to_z
2020-05-04 14:38:40,000 - RECV: 1st Probe Z:2.93
2020-05-04 14:38:40,000 - RECV: >>> do_blocking_move_to X42.83 Y58.00 Z7.93
2020-05-04 14:38:40,013 - RECV: <<< do_blocking_move_to
2020-05-04 14:38:40,165 - RECV: echo:busy: processing
2020-05-04 14:38:40,539 - RECV:   current_position= X42.83 Y58.00 Z7.93 : >>> Probe::probe_down_to_z
2020-05-04 14:38:40,551 - RECV: >>> do_blocking_move_to X42.83 Y58.00 Z1.05
2020-05-04 14:38:40,553 - RECV: <<< do_blocking_move_to
2020-05-04 14:38:41,203 - RECV:   current_position= X42.83 Y58.00 Z2.92 : sync_plan_position
2020-05-04 14:38:41,217 - RECV:   current_position= X42.83 Y58.00 Z2.92 : <<< Probe::probe_down_to_z
2020-05-04 14:38:41,217 - RECV: 2nd Probe Z:2.92 Discrepancy:0.01
2020-05-04 14:38:41,229 - RECV:   current_position= X42.83 Y58.00 Z2.92 : <<< run_z_probe
2020-05-04 14:38:41,229 - RECV: >>> do_blocking_move_to X42.83 Y58.00 Z6.92
[... 1247 lines cut ...]
2020-05-04 14:40:45,434 - RECV:   current_position= X257.00 Y310.00 Z7.11 : Probe::set_deployed
2020-05-04 14:40:45,434 - RECV: deploy: 0
2020-05-04 14:40:45,437 - RECV: Probe::move_z(5.00)
2020-05-04 14:40:45,447 - RECV: >>> do_blocking_move_to X257.00 Y310.00 Z8.05
2020-05-04 14:40:45,450 - RECV: <<< do_blocking_move_to
2020-05-04 14:40:45,714 - RECV: BLTouch STOW requested
2020-05-04 14:40:45,717 - RECV: BLTouch Command :90
2020-05-04 14:40:46,509 - RECV: bltouch.stow_proc() end
[... cut ...]
2020-05-04 14:40:46,581 - RECV:  5 +0.050 -0.002 +0.044 +0.014 +0.026 +0.105 +0.059
2020-05-04 14:40:46,581 - RECV:  6 +0.069 +0.005 +0.050 +0.021 +0.035 +0.093 +0.056
2020-05-04 14:40:46,594 - RECV: G29 uncorrected Z:8.05
2020-05-04 14:40:46,595 - RECV:  corrected Z:7.97
2020-05-04 14:40:46,595 - RECV:   current_position= X257.00 Y310.00 Z7.97 : sync_plan_position
2020-05-04 14:40:46,595 - RECV: >>> do_blocking_move_to X257.00 Y310.00 Z15.00
2020-05-04 14:40:46,605 - RECV: <<< do_blocking_move_to

thierryzoller, 2020-05-04:
- Issued a M420 V
- Issued a M500
- Issued a M501

> 2020-05-04 14:52:23,920 - RECV: echo:M420 V
2020-05-04 14:52:23,920 - RECV: Bilinear Leveling Grid:
2020-05-04 14:52:23,934 - RECV:       0      1      2      3      4      5      6
2020-05-04 14:52:23,934 - RECV:  0 -0.066 -0.126 -0.048 -0.045 -0.008 +0.108 +0.073
2020-05-04 14:52:23,944 - RECV:  1 -0.045 -0.116 -0.035 -0.039 -0.001 +0.113 +0.082
2020-05-04 14:52:23,944 - RECV:  2 -0.024 -0.097 -0.017 -0.018 +0.029 +0.122 +0.090
2020-05-04 14:52:23,954 - RECV:  3 -0.030 -0.075 +0.001 +0.003 +0.043 +0.138 +0.126
2020-05-04 14:52:23,954 - RECV:  4 +0.011 -0.048 +0.024 +0.013 +0.038 +0.137 +0.106
2020-05-04 14:52:23,967 - RECV:  5 +0.057 -0.003 +0.046 +0.022 +0.027 +0.107 +0.060
2020-05-04 14:52:23,970 - RECV:  6 +0.069 -0.005 +0.046 +0.022 +0.024 +0.086 +0.042
2020-05-04 14:52:23,970 - RECV: echo:Bed Leveling OFF
2020-05-04 14:52:23,980 - RECV: echo:Fade Height OFF
2020-05-04 14:52:23,980 - RECV: ok
2020-05-04 14:52:24,051 - RECV: echo:M105
2020-05-04 14:52:24,063 - RECV: ok T:20.86 /0.00 B:20.62 /0.00 @:0 B@:0
2020-05-04 14:52:27,052 - RECV: echo:M105
2020-05-04 14:52:27,052 - RECV: ok T:20.70 /0.00 B:20.62 /0.00 @:0 B@:0
2020-05-04 14:52:30,052 - RECV: echo:M105
2020-05-04 14:52:30,055 - RECV: ok T:20.86 /0.00 B:20.74 /0.00 @:0 B@:0
2020-05-04 14:52:31,592 - SENT: M500
2020-05-04 14:52:31,592 - RECV: echo:M500
2020-05-04 14:52:31,628 - RECV: echo:Settings Stored (826 bytes; crc 22621)
2020-05-04 14:52:31,628 - RECV: ok
2020-05-04 14:52:33,066 - RECV: echo:M105
2020-05-04 14:52:33,086 - RECV: ok T:20.78 /0.00 B:20.66 /0.00 @:0 B@:0
2020-05-04 14:52:34,532 - SENT: M501
2020-05-04 14:52:34,536 - RECV: echo:M501
2020-05-04 14:52:34,561 - RECV: echo:V78 stored settings retrieved (826 bytes; crc 22621)
2020-05-04 14:52:34,565 - RECV: Axis 88 home_offset = 0.00 position_shift = 0.00
2020-05-04 14:52:34,565 - RECV: Axis 88 min:0.00 max:310.00
2020-05-04 14:52:34,565 - RECV: Axis 89 home_offset = 0.00 position_shift = 0.00
2020-05-04 14:52:34,575 - RECV: Axis 89 min:0.00 max:310.00
2020-05-04 14:52:34,575 - RECV: Axis 90 home_offset = 0.00 position_shift = 0.00
2020-05-04 14:52:34,585 - RECV: Axis 90 min:0.00 max:380.00
2020-05-04 14:52:34,585 - RECV: echo:  G21    ; Units in mm (mm)
2020-05-04 14:52:34,595 - RECV: echo:; Filament settings: Disabled
2020-05-04 14:52:34,599 - RECV: echo:  M200 D1.75
2020-05-04 14:52:34,601 - RECV: echo:  M200 D0
2020-05-04 14:52:34,601 - RECV: echo:; Steps per unit:
2020-05-04 14:52:34,601 - RECV: echo: M92 X80.00 Y80.00 Z400.00 E444.00
2020-05-04 14:52:34,611 - RECV: echo:; Maximum feedrates (units/s):
2020-05-04 14:52:34,611 - RECV: echo:  M203 X300.00 Y300.00 Z50.00 E40.00
2020-05-04 14:52:34,621 - RECV: echo:; Maximum Acceleration (units/s2):
2020-05-04 14:52:34,621 - RECV: echo:  M201 X2000.00 Y2000.00 Z100.00 E5000.00
2020-05-04 14:52:34,621 - RECV: echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
2020-05-04 14:52:34,634 - RECV: echo:  M204 P800.00 R10000.00 T2000.00
2020-05-04 14:52:34,635 - RECV: echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
2020-05-04 14:52:34,641 - RECV: echo:  M205 B20000.00 S0.00 T0.00 J0.01
2020-05-04 14:52:34,641 - RECV: echo:; Home offset:
2020-05-04 14:52:34,641 - RECV: echo:  M206 X0.00 Y0.00 Z0.00
2020-05-04 14:52:34,641 - RECV: echo:; Auto Bed Leveling:
2020-05-04 14:52:34,641 - RECV: echo:  M420 S0 Z0.00
2020-05-04 14:52:34,641 - RECV: echo:  G29 W I0 J0 Z-0.06650
2020-05-04 14:52:34,641 - RECV: echo:  G29 W I1 J0 Z-0.12650
2020-05-04 14:52:34,651 - RECV: echo:  G29 W I2 J0 Z-0.04850
2020-05-04 14:52:34,651 - RECV: echo:  G29 W I3 J0 Z-0.04500
2020-05-04 14:52:34,651 - RECV: echo:  G29 W I4 J0 Z-0.00800
2020-05-04 14:52:34,651 - RECV: echo:  G29 W I5 J0 Z0.10800
2020-05-04 14:52:34,651 - RECV: echo:  G29 W I6 J0 Z0.07300
2020-05-04 14:52:34,651 - RECV: echo:  G29 W I0 J1 Z-0.04500
2020-05-04 14:52:34,661 - RECV: echo:  G29 W I1 J1 Z-0.11650
2020-05-04 14:52:34,663 - RECV: echo:  G29 W I2 J1 Z-0.03500
2020-05-04 14:52:34,663 - RECV: echo:  G29 W I3 J1 Z-0.03950
2020-05-04 14:52:34,664 - RECV: echo:  G29 W I4 J1 Z-0.00100
2020-05-04 14:52:34,664 - RECV: echo:  G29 W I5 J1 Z0.11250
2020-05-04 14:52:34,664 - RECV: echo:  G29 W I6 J1 Z0.08200
[... 17 lines cut ...]
2020-05-04 14:52:34,698 - RECV: echo:  G29 W I3 J4 Z0.01250
2020-05-04 14:52:34,698 - RECV: echo:  G29 W I4 J4 Z0.03750
2020-05-04 14:52:34,698 - RECV: echo:  G29 W I5 J4 Z0.13650
2020-05-04 14:52:34,698 - RECV: echo:  G29 W I6 J4 Z0.10550
2020-05-04 14:52:34,698 - RECV: echo:  G29 W I0 J5 Z0.05650
2020-05-04 14:52:34,698 - RECV: echo:  G29 W I1 J5 Z-0.00300
2020-05-04 14:52:34,709 - RECV: echo:  G29 W I2 J5 Z0.04600
2020-05-04 14:52:34,709 - RECV: echo:  G29 W I3 J5 Z0.02150
2020-05-04 14:52:34,709 - RECV: echo:  G29 W I4 J5 Z0.02700
2020-05-04 14:52:34,709 - RECV: echo:  G29 W I5 J5 Z0.10700
2020-05-04 14:52:34,709 - RECV: echo:  G29 W I6 J5 Z0.06000
2020-05-04 14:52:34,709 - RECV: echo:  G29 W I0 J6 Z0.06850
2020-05-04 14:52:34,719 - RECV: echo:  G29 W I1 J6 Z-0.00500
2020-05-04 14:52:34,719 - RECV: echo:  G29 W I2 J6 Z0.04550
2020-05-04 14:52:34,719 - RECV: echo:  G29 W I3 J6 Z0.02150
2020-05-04 14:52:34,719 - RECV: echo:  G29 W I4 J6 Z0.02350
2020-05-04 14:52:34,719 - RECV: echo:  G29 W I5 J6 Z0.08600
2020-05-04 14:52:34,719 - RECV: echo:  G29 W I6 J6 Z0.04200
2020-05-04 14:52:34,719 - RECV: echo:; PID settings:
2020-05-04 14:52:34,730 - RECV: echo:  M301 P12.80 I0.91 D45.09
2020-05-04 14:52:34,733 - RECV: echo:  M304 P33.00 I48.00 D29.00
2020-05-04 14:52:34,733 - RECV: echo:; Z-Probe Offset (mm):
2020-05-04 14:52:34,733 - RECV: echo:  M851 X28.00 Y-33.00 Z-3.05
2020-05-04 14:52:34,733 - RECV: echo:; Linear Advance:
2020-05-04 14:52:34,733 - RECV: echo:  M900 K0.18

thierryzoller, 2020-05-04:
Issued a G29Q

> 2020-05-04 14:54:27,382 - SENT: G29 Q

2020-05-04 14:54:27,385 - RECV: echo:G29 Q
2020-05-04 14:54:27,388 - RECV:   current_position= X257.00 Y310.00 Z15.07 : >>> G29
2020-05-04 14:54:27,388 - RECV: Machine Type: Cartesian
2020-05-04 14:54:27,398 - RECV: Probe: BLTOUCH
2020-05-04 14:54:27,398 - RECV: Probe Offset X28.00 Y-33.00 Z-3.05 (Right-Front & Below Nozzle)
2020-05-04 14:54:27,408 - RECV: Auto Bed Leveling: BILINEAR (disabled)

thierryzoller, 2020-05-04:
Issued an M420 S1

> 2020-05-04 14:56:20,423 - RECV: echo:M420 S1
2020-05-04 14:56:20,433 - RECV:   current_position= X257.00 Y310.00 Z15.07 : Leveling OFF
2020-05-04 14:56:20,443 - RECV:   current_position= X257.00 Y310.00 Z15.00 : ...Now ON
2020-05-04 14:56:20,447 - RECV:   current_position= X257.00 Y310.00 Z15.00 : sync_plan_position
2020-05-04 14:56:20,447 - RECV: echo:Bed Leveling ON
2020-05-04 14:56:20,447 - RECV: echo:Fade Height OFF
2020-05-04 14:56:20,457 - RECV: X:257.00 Y:310.00 Z:15.00 E:0.00 Count X:20560 Y:24800 Z:6028

thierryzoller, 2020-05-04:
Issued an G26 R999

- Scratches the bed immediately
- Verification
>> m420 s1
SENDING:M420 S1
echo:Bed Leveling ON
echo:Fade Height OFF

Videos here :
- https://youtu.be/-RpWCo1oagw
- https://youtu.be/sYaAf5wG_LQ

Picture here:

thierryzoller, 2020-05-04:
I dont think that is the way it is supposed to work. The only way I suceed in printing is leveling the bed manually

thierryzoller, 2020-05-04:
Let me know what you need from me please.

qwewer0, 2020-05-04:
Videos are private, can't watch them.

thierryzoller, 2020-05-06:
I turned off the Printer after this post and haven't touched it. Today I power it up, connect to it via Pronterface and the Z-Offset changed to
>   M851 X28.00 Y-33.00 Z-2.35

As you can see in the logs above it was during the tests:
> M851 X28.00 Y-33.00 Z-3.05

I have not changed printed or done anything to the printer. Is there some sort of Safestate error and the wrong values are being used?

thierryzoller, 2020-05-06:
Here is the full log from todays startup :

> Marlin bugfix-2.0.x
echo: Last Updated: 2020-05-03 | Author: (ZOLLER, BLTouch)
echo:Compiled: May  3 2020
echo: Free Memory: 4735  PlannerBufferBytes: 1312
echo:V78 stored settings retrieved (826 bytes; crc 19789)
Printer is now online.
>>> m503
SENDING:M503
echo:  G21    ; Units in mm (mm)
echo:; Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:; Steps per unit:
echo: M92 X80.00 Y80.00 Z400.00 E444.00
echo:; Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z50.00 E40.00
echo:; Maximum Acceleration (units/s2):
echo:  M201 X2000.00 Y2000.00 Z100.00 E5000.00
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P800.00 R10000.00 T2000.00
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.01
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Auto Bed Leveling:
echo:  M420 S0 Z0.00
echo:  G29 W I0 J0 Z-0.06650
echo:  G29 W I1 J0 Z-0.12650
echo:  G29 W I2 J0 Z-0.04850
echo:  G29 W I3 J0 Z-0.04500
echo:  G29 W I4 J0 Z-0.00800
echo:  G29 W I5 J0 Z0.10800
echo:  G29 W I6 J0 Z0.07300
echo:  G29 W I0 J1 Z-0.04500
echo:  G29 W I1 J1 Z-0.11650
echo:  G29 W I2 J1 Z-0.03500
echo:  G29 W I3 J1 Z-0.03950
echo:  G29 W I4 J1 Z-0.00100
echo:  G29 W I5 J1 Z0.11250
echo:  G29 W I6 J1 Z0.08200
echo:  G29 W I0 J2 Z-0.02400
echo:  G29 W I1 J2 Z-0.09750
echo:  G29 W I2 J2 Z-0.01700
echo:  G29 W I3 J2 Z-0.01850
echo:  G29 W I4 J2 Z0.02900
echo:  G29 W I5 J2 Z0.12200
echo:  G29 W I6 J2 Z0.08950
echo:  G29 W I0 J3 Z-0.03050
echo:  G29 W I1 J3 Z-0.07500
echo:  G29 W I2 J3 Z0.00100
echo:  G29 W I3 J3 Z0.00300
echo:  G29 W I4 J3 Z0.04300
echo:  G29 W I5 J3 Z0.13750
echo:  G29 W I6 J3 Z0.12550
echo:  G29 W I0 J4 Z0.01050
echo:  G29 W I1 J4 Z-0.04800
echo:  G29 W I2 J4 Z0.02400
echo:  G29 W I3 J4 Z0.01250
echo:  G29 W I4 J4 Z0.03750
echo:  G29 W I5 J4 Z0.13650
echo:  G29 W I6 J4 Z0.10550
echo:  G29 W I0 J5 Z0.05650
echo:  G29 W I1 J5 Z-0.00300
echo:  G29 W I2 J5 Z0.04600
echo:  G29 W I3 J5 Z0.02150
echo:  G29 W I4 J5 Z0.02700
echo:  G29 W I5 J5 Z0.10700
echo:  G29 W I6 J5 Z0.06000
echo:  G29 W I0 J6 Z0.06850
echo:  G29 W I1 J6 Z-0.00500
echo:  G29 W I2 J6 Z0.04550
echo:  G29 W I3 J6 Z0.02150
echo:  G29 W I4 J6 Z0.02350
echo:  G29 W I5 J6 Z0.08600
echo:  G29 W I6 J6 Z0.04200
echo:; PID settings:
echo:  M301 P12.80 I0.91 D45.09
echo:  M304 P33.00 I48.00 D29.00
echo:; Z-Probe Offset (mm):
echo:  M851 X28.00 Y-33.00 Z-2.35
echo:; Linear Advance:
echo:  M900 K0.18
Probe Offset X28.00 Y-33.00 Z-2.35

thierryzoller, 2020-05-06:
Gents I went through all that hassle just to get to the bottom of this, would apprecivate someone chiming in :)  @thinkyhead

polato94, 2020-05-06:
Hey,
i also have the Bltouch 3.1 (on Skr 1.3) i use the following settings:

`define Z_MIN_PROBE_ENDSTOP_INVERTING true`
`define Z_PROBE_LOW_POINT          -10 ` <- this should be higher then the Z-Probe offset 

also level the corners
br

thierryzoller, 2020-05-07:
Thanks polato but we had ton of ABL report not compensating on Marlin since the latest releases. This post is an attempt to help properly debug this. IFor instance I am not goin to set "Inverting True" when this is not required for my setup. Leveling the corners is not the issue.

ziporah, 2020-05-08:
I experienced issues as wel, for the sake of trying, I inverted my axis and set X to max end-stop instead of min-end stop, altered all other required settings to make X work correctly and re-ran the probe. My conclusion here is that the probing sequence is exactly the same, while it should be reverted. it starts at X 220-probing distance, Y0-probing distance, while it should start at X0-probing distance, Y0-probing distance. 
This seems to verify the issues I'm seeing, the leveling matrix doesn't take the inversion settings into account and seems to map the wrong matrix on the bed. I will try printing again with this settings and update here. Not sure I might have to do the same for my Y axis, I'm still trying to tracking this...

polato94, 2020-05-08:
> Hey,
> i also have the Bltouch 3.1 (on Skr 1.3) i use the following settings:
> 
> `define Z_MIN_PROBE_ENDSTOP_INVERTING true`
> `define Z_PROBE_LOW_POINT -10 ` <- this should be higher then the Z-Probe offset
> 
> also level the corners
> br
> 
> Edit: i know -10 for Z_PROBE_LOW_POINT is a little bit to much but i don´t even know why this setting is there

well, I forgot to say that I also changed my homing. Before my homing point where on rear right now I put the end stops to front left and also changed these settings. And now it´s working fine.

@thierryzoller could you try following:
Calibrate the Z-Offset with a sheet of paper and save the value.
Make bed leveling with: `G29 ` 
Turn of Bed leveling: `M420 S0`
Go to the first leveling Point: `G42 I0 J0`
Print the Bed Leveling Mesh Values: `M420 V`
Then go to zero with the z axis: 'G1 Z0'
Turn Bed leveling on: `M420 S1`

the current Z-position on the Display should change to the (0,0) value of the printed Mesh values. If you use the following command `G1 Z0`, the same space should be available as for the paper calibration. 
When I perform these steps, the distance between the nozzle and the bed is more or less than the point at which I performed the paper calibration.

ziporah, 2020-05-08:
I followed your procedure exactly, the values shown are inverted.
I went to G42 I2 J2, my grid value is +0.43 there. Then go to G1 Z0 with bed leveling disabled.
enable leveling. The display shows - 0.43 and go to G1 Z0, the bed moves up 0.43 instead of going down.

polato94, 2020-05-09:
> I followed your procedure exactly, the values shown are inverted.
> I went to G42 I2 J2, my grid value is +0.43 there. Then go to G1 Z0 with bed leveling disabled.
> enable leveling. The display shows - 0.43 and go to G1 Z0, the bed moves up 0.43 instead of going down.

ok i think there is the bug.

ziporah, 2020-05-09:
I have a simple bl touch. I only defined this in the config:
‘’’
#define BLTOUCH
‘’’

thierryzoller, 2020-05-09:
Upload you config please

thierryzoller, 2020-05-09:
@polato94 : There is no need to as it worked beforehand on previous builds, rather than staying  theoretical one should take a look at the code and determine where the error happens. Signed/unsigned ?  For that I have to leave it to someone else. The bug was demonstrated and reproduced. I do not know it what source file to code for this is located and I would except a Dev to step in.

> You said:
to get a better understanding about the bug someone with another Probe or another Probe system should try the same steps. Then we can determine if this bug is only on BLTouch 3.1 or in the FW.

thierryzoller, 2020-05-09:
Should be somewhere around here :
https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/module/probe.cpp
https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/module/probe.h

marcjwebb, 2020-05-09:
hya all, hope everyone here is safe and well. wondered if this was fixed in the latest bugfix on the firmware page? i am too getting this issue. ( ender 3 pro, bltouch v3.1, marlin 2.0.5.3 )

thinkyhead, 2020-05-10:
Still tracking down the cause. If someone knows a particular version of Marlin where it's still okay, that would be useful data. I'll be able to get into troubleshooting this issue in about 12 hours….

thinkyhead, 2020-05-10:
> I went to G42 I2 J2, my grid value is +0.43 there.

- In this position the bed is 0.43mm higher, so the bed and nozzle end up 0.43mm closer at this position when leveling is disabled. The correct behavior at this point should be:
  - Leveling ON to OFF: Z=0 becomes Z=+0.43 (and shouldn't be constrained by soft endstops, etc.)
  - Leveling OFF to ON: Z=0 becomes Z=-0.43 (and shouldn't be constrained by soft endstops, etc.)

> Then go to G1 Z0 with bed leveling disabled. Enable leveling. The display shows -0.43. Go to `G1 Z0`, the bed moves **up 0.43 instead of going down**.

If the Z position was 0 before leveling was disabled, then the display should should indeed show **`-0.43`**. That accords with my description above. And, at the very least, moving to Z0 is supposed to move the nozzle farther away from the bed. So there is definitely something amiss!

That could be a case where the planner wasn't sync'ed with the updated Z position after it was changed by the `set_bed_leveling_enabled` function… or as alluded to, the soft endstops could be getting in the way.

You can check whether the planner and the logical position are properly synchronized by turning on `M114_DETAIL` and then sending `M114 D` at the point just after you disable leveling, but before you (don't) do the move to Z0. That will probably reveal that the planner and the logical position are out of phase. If only it could tell us where and how…

It would be worth doing some experiments to see if changing the values of the Z min soft endstop has any effect on leveling behavior. I bet there's something there. See if doing `M211 S0` has any effect on the way this bug manifests, or if turning off soft endstops altogether makes some difference.

If handling has been changed recently around `set_bed_leveling_enabled` or around the way soft endstops get applied or reckoned — perhaps in relation to the MIN_POS/MAX_POS settings— in recent months, these would things to check…

I will look into this more in about 10-12 hours, after the whole sleep thing.

ziporah, 2020-05-10:
> Upload you config please
This is my current configuration

[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/4605356/Configuration_adv.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/4605357/Configuration.txt)

ziporah, 2020-05-10:
I reconfigured my X axis to default behaviour, uploaded clean firmware, reset to default settings, ran G28, followed by G29
here's the output:

```
echo:Index: 628 Size: 692
echo:Hardcoded Default Settings Loaded
echo:SD card ok
Bilinear Leveling Grid:
      0      1      2      3      4
 0 +0.843 +0.985 +0.160 +0.646 +1.167
 1 +0.674 +0.645 +1.152 +1.095 +1.210
 2 +0.334 +0.479 +0.880 +0.996 +1.076
 3 +0.581 +0.032 +0.751 +0.955 +1.199
 4 +0.491 +0.103 -0.358 -0.238 +0.372
>>> M114 D
SENDING:M114 D
X:108.00 Y:120.00 Z:9.66 E:0.00 Count X:10680 Y:11752 Z:4129
>>> M420 S0
SENDING:M420 S0
echo:Bed Leveling OFF
echo:Fade Height OFF
>>> G42 I2 J2
SENDING:G42 I2 J2
>>> M114 D
SENDING:M114 D
X:110.00 Y:93.50 Z:10.40 E:0.00 Count X:10878 Y:9156 Z:4129
>>> M114_DETAIL
SENDING:M114_DETAIL
X:110.00 Y:93.50 Z:10.40 E:0.00 Count X:10878 Y:9156 Z:4129
>>> G1 Z0
SENDING:G1 Z0
>>> M114_DETAIL
SENDING:M114_DETAIL
X:110.00 Y:93.50 Z:0.00 E:0.00 Count X:10878 Y:9156 Z:0
>>> M420 S1
SENDING:M420 S1
echo:Bed Leveling ON
echo:Fade Height OFF
>>> M114_DETAIL
SENDING:M114_DETAIL
X:110.00 Y:93.50 Z:-0.88 E:0.00 Count X:10878 Y:9156 Z:0
>>> G1 Z0
SENDING:G1 Z0
```
at Z:4129 => the Z axis was still in the position after homing and bed leveling
then issued the first G1 Z0
display showed -0.88 befor the final G1 Z0 and Z went up on G1 Z0
If you need some more debugging info, let me know, I'll try to collect asap.
Should I try the 0.2.0-bugfix branch ?

thierryzoller, 2020-05-10:
> the current Z-position on the Display should change to the (0,0) value

My display doesn't show Z, what M503 value should I watch?

ziporah, 2020-05-10:
M114 D shows the current value

thierryzoller, 2020-05-10:
@polato94 : 
I did what you said step by step. After the 420 S1 there was 0 movement.
X:28.00 Y:25.00 Z:0.19 E:0.00 Count X:2240 Y:2000 Z:0

ziporah, 2020-05-10:
Clean again, homing G28, bed level G29
Z position after homing on display: 9.55
Z position after disabling bed leveling : 10
Z position after G42 I2 J2 , G1 Z0 : 0
Re-enable leveling :  -0.442
Z position after G1 Z0 : 0 => axis went up
detailed info in this gist: 
https://gist.github.com/ziporah/f82595526618b0a5deaf7a4ba887897e

thierryzoller, 2020-05-10:
Here is the Debug Log:

> 2020-05-10 20:08:13,345 - RECV: echo:M105
> 2020-05-10 20:08:13,361 - RECV: ok T:23.09 /0.00 B:23.05 /0.00 @:0 B@:0
> 2020-05-10 20:08:16,348 - RECV: echo:M105
> 2020-05-10 20:08:16,362 - RECV: ok T:23.09 /0.00 B:23.05 /0.00 @:0 B@:0
> 2020-05-10 20:08:17,457 - SENT: M420 S0
> 2020-05-10 20:08:17,457 - RECV: echo:M420 S0
> 2020-05-10 20:08:17,457 - RECV:   current_position= X257.00 Y310.00 Z15.00 : Leveling ON
> 2020-05-10 20:08:17,457 - RECV:   current_position= X257.00 Y310.00 Z15.22 : ...Now OFF
> 2020-05-10 20:08:17,471 - RECV:   current_position= X257.00 Y310.00 Z15.22 : sync_plan_position
> 2020-05-10 20:08:17,471 - RECV: echo:Bed Leveling OFF
> 2020-05-10 20:08:17,471 - RECV: echo:Fade Height OFF
> 2020-05-10 20:08:17,471 - RECV: ok
> 2020-05-10 20:08:19,345 - RECV: echo:M105
> 2020-05-10 20:08:19,359 - RECV: ok T:23.20 /0.00 B:23.09 /0.00 @:0 B@:0
> 2020-05-10 20:08:22,348 - RECV: echo:M105
> 2020-05-10 20:08:22,362 - RECV: ok T:23.05 /0.00 B:23.20 /0.00 @:0 B@:0
> 2020-05-10 20:08:25,358 - RECV: echo:M105
> 2020-05-10 20:08:25,365 - RECV: ok T:23.20 /0.00 B:23.05 /0.00 @:0 B@:0
> 2020-05-10 20:08:28,359 - RECV: echo:M105
> 2020-05-10 20:08:28,359 - RECV: ok T:22.93 /0.00 B:23.20 /0.00 @:0 B@:0
> 2020-05-10 20:08:31,352 - RECV: echo:M105
> 2020-05-10 20:08:31,352 - RECV: ok T:23.16 /0.00 B:23.05 /0.00 @:0 B@:0
> 2020-05-10 20:08:34,346 - RECV: echo:M105
> 2020-05-10 20:08:34,361 - RECV: ok T:23.09 /0.00 B:23.05 /0.00 @:0 B@:0
> 2020-05-10 20:08:34,783 - SENT: G42 I0 J0
> 2020-05-10 20:08:34,783 - RECV: echo:G42 I0 J0
> 2020-05-10 20:08:34,783 - RECV: ok
> 2020-05-10 20:08:37,346 - RECV: echo:M105
> 2020-05-10 20:08:37,361 - RECV: ok T:23.05 /0.00 B:23.09 /0.00 @:0 B@:0
> 2020-05-10 20:08:40,358 - RECV: echo:M105
> 2020-05-10 20:08:40,358 - RECV: ok T:22.97 /0.00 B:23.01 /0.00 @:0 B@:0
> 2020-05-10 20:08:43,345 - RECV: echo:M105
> 2020-05-10 20:08:43,361 - RECV: ok T:23.24 /0.00 B:23.05 /0.00 @:0 B@:0
> 2020-05-10 20:08:46,358 - RECV: echo:M105
> 2020-05-10 20:08:46,358 - RECV: ok T:23.24 /0.00 B:23.05 /0.00 @:0 B@:0
> 2020-05-10 20:08:49,359 - RECV: echo:M105
> 2020-05-10 20:08:49,359 - RECV: ok T:22.97 /0.00 B:23.20 /0.00 @:0 B@:0
> 2020-05-10 20:08:52,358 - RECV: echo:M105
> 2020-05-10 20:08:52,358 - RECV: ok T:23.32 /0.00 B:22.97 /0.00 @:0 B@:0
> 2020-05-10 20:08:55,030 - SENT: M420 V
> 2020-05-10 20:08:55,030 - RECV: echo:M420 V
> 2020-05-10 20:08:55,030 - RECV: Bilinear Leveling Grid:
> 2020-05-10 20:08:55,046 - RECV:       0      1      2      3      4      5      6
> 2020-05-10 20:08:55,046 - RECV:  0 -0.191 -0.146 -0.147 -0.134 -0.000 +0.046 +0.125
> 2020-05-10 20:08:55,046 - RECV:  1 -0.178 -0.113 -0.119 -0.115 +0.008 +0.066 +0.134
> 2020-05-10 20:08:55,046 - RECV:  2 -0.108 -0.051 -0.062 -0.058 +0.061 +0.096 +0.159
> 2020-05-10 20:08:55,061 - RECV:  3 -0.043 +0.011 -0.003 -0.010 +0.093 +0.111 +0.177
> 2020-05-10 20:08:55,061 - RECV:  4 +0.044 +0.096 +0.060 +0.036 +0.120 +0.142 +0.189
> 2020-05-10 20:08:55,061 - RECV:  5 +0.183 +0.203 +0.146 +0.111 +0.171 +0.148 +0.190
> 2020-05-10 20:08:55,076 - RECV:  6 +0.240 +0.254 +0.194 +0.154 +0.228 +0.204 +0.248
> 2020-05-10 20:08:55,076 - RECV: echo:Bed Leveling OFF
> 2020-05-10 20:08:55,076 - RECV: echo:Fade Height OFF
> 2020-05-10 20:08:55,076 - RECV: ok
> 2020-05-10 20:08:55,348 - RECV: echo:M105
> 2020-05-10 20:08:55,362 - RECV: ok T:23.12 /0.00 B:23.20 /0.00 @:0 B@:0
> 2020-05-10 20:08:58,349 - RECV: echo:M105
> 2020-05-10 20:08:58,365 - RECV: ok T:23.24 /0.00 B:23.12 /0.00 @:0 B@:0
> 2020-05-10 20:08:58,802 - SENT: G1 Z0
> 2020-05-10 20:08:58,802 - RECV: echo:G1 Z0
> 2020-05-10 20:08:58,802 - RECV: ok
> 2020-05-10 20:09:01,345 - RECV: echo:M105
> 2020-05-10 20:09:01,361 - RECV: ok T:23.09 /0.00 B:23.09 /0.00 @:0 B@:0
> 2020-05-10 20:09:03,749 - SENT: M420 S1
> 2020-05-10 20:09:03,749 - RECV: echo:M420 S1
> 2020-05-10 20:09:03,749 - RECV:   current_position= X28.00 Y25.00 Z0.00 : Leveling OFF
> 2020-05-10 20:09:03,749 - RECV:   current_position= X28.00 Y25.00 Z0.19 : ...Now ON
> 2020-05-10 20:09:03,763 - RECV:   current_position= X28.00 Y25.00 Z0.19 : sync_plan_position
> 2020-05-10 20:09:03,763 - RECV: echo:Bed Leveling ON
> 2020-05-10 20:09:03,763 - RECV: echo:Fade Height OFF
> 2020-05-10 20:09:03,763 - RECV: X:28.00 Y:25.00 Z:0.19 E:0.00 Count X:2240 Y:2000 Z:0
> 2020-05-10 20:09:03,763 - RECV: ok
> 2020-05-10 20:09:04,345 - RECV: echo:M105
> 2020-05-10 20:09:04,361 - RECV: ok T:23.09 /0.00 B:23.01 /0.00 @:0 B@:0
> 2020-05-10 20:09:07,355 - RECV: echo:M105
> 2020-05-10 20:09:07,355 - RECV: ok T:23.16 /0.00 B:23.09 /0.00 @:0 B@:0
> 2020-05-10 20:09:08,984 - SENT: M114 D
> 2020-05-10 20:09:08,984 - RECV: echo:M114 D
> 2020-05-10 20:09:08,984 - RECV: X:28.00 Y:25.00 Z:0.19 E:0.00 Count X:2240 Y:2000 Z:0
> 2020-05-10 20:09:09,000 - RECV: Logical: X:28.000 Y:25.000 Z:0.191
> 2020-05-10 20:09:09,000 - RECV: Raw:     X:28.000 Y:25.000 Z:0.191
> 2020-05-10 20:09:09,000 - RECV: Leveled: X:28.000 Y:25.000 Z:0.000
> 2020-05-10 20:09:09,000 - RECV: UnLevel: X:28.000 Y:25.000 Z:0.191
> 2020-05-10 20:09:09,016 - RECV: Stepper: X:2240 Y:2000 Z:0 E:0
> 2020-05-10 20:09:09,016 - RECV: FromStp: X:28.000 Y:25.000 Z:0.000 E:0.000
> 2020-05-10 20:09:09,016 - RECV: Diff:  X:0.000 Y:0.000 Z:0.000 E:0.000
> 2020-05-10 20:09:09,016 - RECV: ok
> 2020-05-10 20:09:10,354 - RECV: echo:M105
> 2020-05-10 20:09:10,354 - RECV: ok T:23.01 /0.00 B:23.01 /0.00 @:0 B@:0
> 2020-05-10 20:09:13,354 - RECV: echo:M105
> 2020-05-10 20:09:13,354 - RECV: ok T:23.05 /0.00 B:23.05 /0.00 @:0 B@:0

thierryzoller, 2020-05-10:
I can also notice no compensation when doing G1 travel moves. Not sure that is supposed to be compensated.

thierryzoller, 2020-05-10:
Another observation, I turned ABL off, the Z motor slightly adjusted midst printing move (althought it is on the very same layer). I turned ABL on, same thing. Why would the zmotor ever so slighly spin when printing move in the middle of a layer?

hitbyatruck, 2020-05-11:
Hello, everyone.
This bug has been driving me nuts as well. It's been over a week of trying to figure out the problem and I still can't figure out why the ABL is not compensating for the mesh while printing. 
As anyone been able to figure out what may be causing this bug?
After several tries with different firmware versions, nothing seems to make sense for why this keeps happening, when several people don't have the same problem.
I'm with an Ender 3 with a SKR mini E3 V1.2 and a BLtouch V2.1. Let me know if there's any other information that may be prove usefull on solving this problem and I'll try to debug it along with you guys, if you don't mind.

HooverC91, 2020-05-11:
Following and willing to provide input. I have a Prusa Bear with both a PINDA and a Genuine BLTouch 3.1 and noticed the BLTouch has far worse results than the PINDA (as if its compensating some but not enough). I tested this by greatly skewing the bed and printing first layer tests. The only difference is the mounting location- PINDA (23, 5, 0.68)  vs BLTouch (-24.3, -34.1, 2.13)

hitbyatruck, 2020-05-11:
As of now, the current bugfix 2.0.x seems to be working out ok... so far, at least.

thinkyhead, 2020-05-11:
Good to hear that the current bugfix ABL is behaving. ABL has been pretty solid on our test machines, but I still hear about some anomalies related to …maybe… soft endstops. We're not entirely sure. But, between prints there seems to be an accumulating error in Z. Other than that, the compensation does follow the bed contours when it's supposed to.

For troubleshooting leveling and getting a deeper look at what might be going on, the `DEBUG_LEVELING_FEATURE` adds a lot of extra logging that appears with `M111 S32`. But for checking whether the leveling at a particular spot on the bed is correct, the combination of that with `M114_DETAIL` allows you to use `G29 Q` to get a report of bed leveling compensation at the current point. And also `M114 D` to get deeper details about the steppers and leveling calculations.

thierryzoller, 2020-05-11:
Sorry to rain on the parade hitbyatruck  - latest bugfix didn't fix anything and introduce Y stutter.
Hence why my issue is open, still waiting on way to help solving it.

thinkyhead, 2020-05-11:
Thanks for all the data. At this point you will have to wait for someone with troubleshooting and coding ability to install the latest Marlin on a test machine, investigate the issue, narrow down the cause, and hopefully resolve the issue and submit a patch for testing.

GhostlyCrowd, 2020-05-11:
> Still tracking down the cause. If someone knows a particular version of Marlin where it's still okay, that would be useful data. I'll be able to get into troubleshooting this issue in about 12 hours….

If it helps every person with ABL issues who has rolled back to 2.0.3 is 100% happy that i have personally rolled back. Many of them are too busy printing to move forward again to test, I have one friend who is willing to test as soon as this last batch of PPE is done.  Also UBL does no suffer from this issue, but as stated above it appears to be a mesh inversion or non-iversion, related to no respecting the inversion or lack of in the axis in the firmware.

thinkyhead, 2020-05-11:
> it appears to be a mesh inversion or non-iversion

We'll see. I have closely examined the log output from @thierryzoller related to how turning leveling on and off affects the current Z position and the underlying stepper count. Everything there looks appropriate.

A very simple controlled test that anyone can try is to make a fake mesh that is all zeros to start with, and then just change the center point value. Do small and large moves around the center to see how the nozzle behaves. Of course, keep the nozzle above the bed so that you don't damage any components.

Start out with the Z Fade turned off so that you get full Z adjustment at any height. Later, experiment with how changing Z fade affects movement. We've tried to make changing the Z fade during segmented movement a safe thing to do. However, it's not necessarily recommended, so during testing only change Z fade in-between moves.

By using very controlled numbers and a very controlled set of moves, along with detailed logging using `G29 Q` and `M114 D` (as needed) it should be obvious when Z is moving too much or too little or going in the wrong direction.

The thing to do then is to make some short, fake print job that doesn't heat up or extrude, but just turns on leveling, moves XY around, then raises Z and moves XY some more, and does this for about a centimeter in height. The G-code could contain your usual slicer start/end G-code, along with "`M114 D`" commands to examine any "interesting" codes that your slicer inserts.

At the end of the fake print job, try to run the fake print job again, and look closely to see if Z is getting thrown off, and by how much. Check to see if the `M114 D` report shows a discrepancy at the same time.

It would be worth trying G-code that contains `G28` and `M420 S1` at the top. And then to try again without the `G28` that re-homes XYZ. Does the `G28` fix the XYZ position?

Take careful note of whether your first `G28` was done with a cold bed, and your second `G28` was done on a heated bed. The bed can expand by a large amount, and the contours of the mesh will vary depending on how hot the bed is. For this reason, always heat your bed before you do `G28` with a probe or measure your bed with `G29`.

Generally, should you do `G28` and/or `G29` between prints?
- If homing with a probe, `G28` is good between prints to account for expansion.
- If homing Z to max and/or with an endstop, another `G28` is not strictly needed.
- Another `G29` between prints can't hurt, since heat warp isn't entirely predictable.

To get a sense of how much the bed may be expanding, Marlin contains useful new feature called `PROBE_TEMP_COMPENSATION`. It requires the probe to have a temp sensor, but you can set `TEMP_SENSOR_PROBE` equal to 998 and it will just use the bed temperature. You can then use the `G76` command to measure the deflection of your bed at a range of temperatures, and this will also pick up some of the error that the bed temperature might be introducing to your probe.

thinkyhead, 2020-05-11:
> 2.0.3

I will closely compare 2.0.3 to the code that followed. It sounds like even 2.0.4 exhibits some anomalies.

ziporah, 2020-05-12:
I just tried compiling 2.0.3 and ran a test print on it, I'm not really convinced the leveling is working as expected. I reset my settings, G28, G29, Z offset measurement on probe point I2 J2, but it slams the head in the glass plate when I reload settings and restart with the measured offset. I lowered the offset from -3.50 to -3.20 and then I can start printing, but it shouldn't be off anyway. 
There must be something amiss by disabling the soft endstops with M211 S0.
Second observation is that the leveling on the back of the plane still seems to be wrong. it still pushes the head to deep in the bed, while at front it is slightly to high.
The print sticks to the bed, but I don't believe this is optimal and leveling is still not correct here.
The print covers the inner 3x3 matrix, it's not a full bed coverage test
```
>>> M420 V
SENDING:M420 V
Bilinear Leveling Grid:
      0      1      2      3      4
 0 +0.203 +0.064 -0.025 +0.229 -0.193
 1 -0.628 -0.116 -0.270 +0.195 -0.389
 2 -0.760 -0.368 -0.017 -0.015 -0.154
 3 -1.172 -0.442 -0.079 +0.150 -0.246
 4 -0.963 -0.974 -0.375 +0.246 -0.106
echo:Bed Leveling ON
...
M503
echo:  M851 X2.00 Y-63.00 Z-3.20
```

GhostlyCrowd, 2020-05-12:
> I just tried compiling 2.0.3 and ran a test print on it, I'm not really convinced the leveling is working as expected. I reset my settings, G28, G29, Z offset measurement on probe point I2 J2, but [...]

@ziporah  

Is your bed really this un level (below) or is this part of the ABL error you are seeing? If its really this unlevel you should try to manually address that at least get the corners with in reason of each other for ABL to work, it compensates for bed level issues but it's not a miracle worker, you will still have issues on a very unlevel bed even with ABL so if your bed is indeed as below, wildly unlevel this may be part of the problem. 

[image]

ziporah, 2020-05-12:
I'm totally baffled what's wrong with the machine/firmware. I've leveled the bed manually several times, and I did manage to print without the ABL, before I added the BL Touch. I added it in order to get better prints, but even after running G29 several times in a row, it is showing different meshes on each iteration. Yes, the bed is heated at 60 degrees. I've already ran M48 to validate the BL touch and it shows a correction of 0.026 consistent over several runs. In total madness I disassembled the bed, removed the Y axis, cleaned everything, measured all again to be parallel and sliding smooth, so I can't find nothing wrong there. I would at least expect the mesh that's measured to be the same on each run, within the tolerance of the BL touch, but even that's a dream ...

thierryzoller, 2020-05-12:
@GhostlyCrowd : 
From left bottom to left top there is around 1mm of difference, the graph is grossly misrepresenting. ABL should easily cover 1mm from a 300mm distance. Mine is a bit more level, but still doesn't work. Not the problem.

thierryzoller, 2020-05-12:
I do not have the tolerance issue that you are describing, it also can be seen in the logs I posted above. So we can exclude that as the issue.

GhostlyCrowd, 2020-05-12:
This is an example i just pulled from a friend on 2.0.3, 2.0.3 was the latest build where he didnt have abl issues and the bltouch failing to deploy and driving the tool head into his bed. 

He is using more mesh points and extrapolating beyond the grid. Try a few more mesh points and extrapolation and see if it at leats improves your symptoms on 2.0.3

[image]

thierryzoller, 2020-05-12:
BLTouch failed to deploy only in case I had a wiring issue (loos connection)

thierryzoller, 2020-05-12:
Understand that we should be looking everywhere, but as the person that has opened this issue, can we stay within that frame for a second? I have no deviation of BLTouch measures, it deploys correctly and coherently. What I am seeing is :  
- Z movements on 2.0.5 builds although ABL is disabled ??
- Not seeing compensation on newest builds

ziporah, 2020-05-12:
> BLTouch failed to deploy only in case I had a wiring issue (lose connection) - there is an issue documenting this on here.

The BLTouch always deployed here as well. I do see Z movement on the newest builds, but by the looks of it, it's rather an inversed compensation.

thierryzoller, 2020-05-12:
I'd like to use "DEBUG_LEVELING_FEATURE" and "M111 S32" but I won't understand what I am seeing.

thestephenmarshall, 2020-05-14:
> I just tried compiling 2.0.3 and ran a test print on it, I'm not really convinced the leveling is working as expected. 

Same here. Just ran `2.0.3` and leveling is not working as expected here either even after restart on first print. ¯\_(ツ)_/¯

Guess I'll run all the bugfix branch latest and attach my log out as well...

Mils24, 2020-05-17:
Hi all, I am following this thread closely as I also have issues where my bilinear mesh is not being applied to my prints.

Mesh is being measured ok and showing on EEPROM but is not active during printing. (M420 S1 is in my start script after my G28)

I hope the smarter guys than me on here are working to find a solution. Any update on current progress would be greatly appreciated? 

I am also willing to help in any way I can. I can offer data and configs if you need extra material to help find the source of the problem. 

Printer: Homemade delta
Motherboard: MKS SBASE 1.3
Firmware:  Marlin Bugfix 2.0.x (currently) also tried Marlin 2.0.5 before moving to Bugfix as I had read somewhere that Bugfix worked with ABL. 
Sensor: Triangle Labs 3DTouch (BL Touch clone)
ABL type: Bilinear ABL

Really hope we can find a solution soon as about 1/3 of my bed is currently unusable.

Thanks.

Mils24, 2020-05-18:
Please see my debug file below as per @thinkyhead instructions for commenting on this issue.

[Debug.txt](https://github.com/MarlinFirmware/Marlin/files/4643062/Debug.txt)

Mils24, 2020-05-18:
Ive been doing a bit more thinking and more reading of other threads on here with similar issues.

Seems a few people have linked odd ABL and BLTouch issues to Z_SAFE_HOMING.

Question - With Z_SAFE_HOME ON on a Delta should the endeffector move to the centre of the bed before homing up to the top? (assuming you have the safe homing set to (delta_printable_radius)/2??? 

This is something that has puzzled me also. I have Z_SAFE_HOMING on with the safe point set as per above. However when my prints finish or I'm I'm testing points of the bed and then home, my printer just moves straight up. My expected behaviour was move to centre of the bed and then up but it's not doing that.

If my Z_SAFE_HOMING isn't working properly and some people think its linked to ABL issues could this be my problem?

Is it crazy to tun it off?

thinkyhead, 2020-05-18:
With `DEBUG_LEVELING_FEATURE` enabled, you gain another ability with ABL (Bilinear, Linear, 3-point):
- During a print, send `G29 Q` to get the current leveling information. This command will cause Marlin to report the amount of leveling that is being applied at the current position. And, `M114 D` during a print will convey similar information.

Mils24, 2020-05-18:
Thanks @thinkyhead I will add some of these in and see what it returns on the first layer during some test prints and report back.

thinkyhead, 2020-05-18:
The data collection is appreciated. I have not been able to reproduce the issue.

Can I also assume that no one is using `BABYSTEPPING` at the start of their print? Because with the babystep-probe-offset option, of course this will change your probe offset and result in your nozzle height changing on the next `G28`.

[11 more replies]

closed 2020-08-24 by github-actions[bot]