MarlinFirmware/Marlin #17876: G34 only probing Z1 position on 2 axis Z setup

omnillogix, 2020-05-04:
I have a dual Z gantry setup that homes to ZMAX
I have enabled Z_STEPPER_AUTO_ALIGN and set the coordinates Z_STEPPER_ALIGN_XY { {  100, 100 }, { 1200,  100 } }.  All other options are left as default.  

When I issue a G34, the machine homes, and probes the Z1 {100,100} position 4 times and stops.   The terminal says:
ITERATION: 1
DIFFERENCE Z1-Z2=0.01
Target accuracy achieved.
Did 1 of 2
Accuracy: 0.01

I don't think this is correct behavior.  Any help is appreciated

randellhodges, 2020-05-04:
Attach your config files please

omnillogix, 2020-05-04:
Config files attached

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/4577548/Configs.zip)

omnillogix, 2020-05-04:
FYI: I tried to test with the current bugfix release, but with the latest build, the pin on my BLTouch won't deploy for the G34 probing, so I am unable to say if the movement issue is resolved.

randellhodges, 2020-05-04:
Wow, do you really have a 1.2 meter by 2.4 meter printer?

omnillogix, 2020-05-04:
Actually a CO2 Laser.   I have a gantry that falls when power off, so it homes to Z+ on startup, then needs to probe the bed for actual Z offset

randellhodges, 2020-05-04:
Any difference if you pull in the X positions for `Z_STEPPER_ALIGN_XY`, maybe  { 100, 100 }, { 1120, 100 }? If that doesn't do anything, we'll have to see if one the devs knows.

You might want to jump on Discord and see if someone there knows.

omnillogix, 2020-05-04:
I started with {{10,100} , {100,100}}  with same result.   I also tried swapping my Z1/Z2 motor plugs in case the orientation was wrong, but also same result.  I also tried to change the Y offset on the Z2 to see if the problem was only on the X coordinate, but still no change in position for Z2

omnillogix, 2020-05-04:
@randellhodges, your comment got me thinking...   I swapped the coordinates to: { {  500, 125 }, { 100,  100 } }.   Looks like a bug.   
When I ran G34, it did the following:
Homed
Started Iteration #1
Moved to 500,125
Probed 2 times
Moved to 100,100
Probed 2 times
Started Iteration #2
*did not move
Probed 4 times
Output: Accuracy 0.0

omnillogix, 2020-05-11:
So, it's been a week.  Can a dev maybe comment on this issue?  Is there more information that I can provide?
Thanks

sjasonsmith, 2020-05-29:
@omnillogix, do you have a log from the last case you reported, where it probed both points, but failed to move on the second iteration?

The challenge is that not that many people use this feature, and even fewer have a machine which actually resembles yours. This makes it difficult for anybody to try to reproduce your issue to figure out where the problem is.

omnillogix, 2020-05-29:
@sjasonsmith, I don't have a log.   After a few days, I figured that my use case is fairly unusual.  Rather than fight it, I have changed my machine to home to Z-min using a probe and the G34 works except for the occasional know issue of the "decreasing accuracy" message.   If someone wants to chase this issue down, I am happy to provide whatever information I can.    I am sure the problem lies in homing to Z-Max

boelle, 2020-06-21:
@omnillogix still an issue?

omnillogix, 2020-06-21:
As far as I know it is still an issue.   I have stopped homing to Z Max, in favor of well supported homing to Z-Min.   I don't think this should be closed, but rather made a lower priority since it is an edge case.  I'm sure future Marlin users might appreciate being able to G34 and home to Z Max.

boelle, 2020-06-29:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-29 by boelle