MarlinFirmware/Marlin #17878: [BUG] (VSCode Build Failed Multiple Configs)

otisczech, 2020-05-04:
[image]
[image]

### Bug Description
The VSCode fails to build with default Ender 3 config. I've even tried Ender 2 and 3 V2. Same result.

### My Configurations
[MyConfig.zip](https://github.com/MarlinFirmware/Marlin/files/4577244/MyConfig.zip)

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Use Ender 3 config files
2. Click build in VSCode using mega 2560
3. Build fails

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

I'm no programmer myself, I've discussed this with my friend, we think there may be an error in file updated 2 days ago - regarding XYZ, which the error states is wrong. I've attached screenshot of possible error in declaration... If it's something I do wrong, please let me know, i went by the manual: https://www.youtube.com/watch?v=RbbzsJBpEhc

And then even by: https://marlinfw.org/docs/basics/install_platformio_vscode.html
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

otisczech, 2020-05-04:
Sorry, my first post. Obv Expected: Build successful, Actual: Build failed

ellensp, 2020-05-05:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/17594 
Config files are out of sync with code

All you do is take the following line in your Configuration_adv.h

#define HOMING_BUMP_MM { 5, 5, 2 } // (mm) Backoff from endstops after first bump

and change it to these 3 lines.

#define X_HOME_BUMP_MM 5
#define Y_HOME_BUMP_MM 5
#define Z_HOME_BUMP_MM 2

closed 2020-05-05 by otisczech