MarlinFirmware/Marlin #17881: [BUG] Stealthchop is disabled when printing from USB on the BTT SKR E3 mini V1.2

430984, 2020-05-05:
### Bug Description

Stealthchop is disabled when printing from USB on the BTT SKR E3 mini V1.2 using the TMC2209 drivers.

When starting a print from an USB source (Octoprint in my case), StealthChop gets disabled and can not be enabled again until the next reboot. How do I know this? Loud steppers and M122 reports Stealthchop false for all axes.

The weird thing however is that when printing from SD card, all is fine (if done before trying to print from Octoprint, after that Stealthchop is disabled and can not be enabled again until the next reboot)

Tried the bugfix branch, that's not working either.

Some information:
- Ender 3, default motors, fans, screen etc
- BigTreeTech SKR E3 mini V1.2
- TMC2209 for all axes (X, Y, Z and E0)

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4580384/Configuration.zip)

### Steps to Reproduce

1. Install Octoprint and set that up
2. Upload a .gcode file to Octoprint
3. Upload the same .gcode to a SD card
4. Start a print from Octoprint
5. Listen
6. Reboot after print is done (or cancelled)
7. Start the same print from the SD card
8. Listen

The difference is there. M122 also confirmed my suspicions.
When using Octoprint: Stealthchop false for X, Y, Z and E0
When using SD card: Stealthchop true for all axes.

**Expected behavior:** Quiet printing

**Actual behavior:** Loud printing

swilkens, 2020-05-05:
Could it be that you are running into `#define HYBRID_THRESHOLD` ?

Try to disable that, rebuild and test again

430984, 2020-05-06:
Thank you for the suggestion. I tried the option you suggested just now. But it seems that when I disable hybrid threshold, I can't get StealthChop to work at all. Not when homing and not when printing from either USB or SD.

430984, 2020-05-06:
Running a test print now. Will update in a few minutes!

430984, 2020-05-06:
Yes, it seems like that was the issue. Although travel moves are a bit slower now, but I can live with that. Thanks for your help!

swilkens, 2020-05-06:
I will look a bit deeper into this, it's strange that all your axes move to spreadcycle. What are your slicer settings in terms of speed?

Your HYBRID_THRESHOLD was set at:

> X, Y 100 mm/s
> Z 3 mm/s
> E 30 mm/s

So if your slicer speeds were above that, it would be expected that all steppers switch to spreadcycle.

430984, 2020-05-06:
I usually print at 60mm/s (which was the case in this instance).
But I find it particularly strange that StealthChop works when printing from SD, but not when printing from USB, athough the same G-code is being used in both cases. Is there some mechanism in Marlin that selects when/if StealthChop is being used, other than the hybrid threshold?

Also I found out that when you power up the printer and initiate the homing cycle for all axes (G28) from USB, it doesn't use StealthChop. But when initiated from the SD card or the LCD, it does.

swilkens, 2020-05-06:
I have the same board and I can't replicate that behavior, the printer homes by using stealthchop. Can you try the very latest bugfix code ?

How are you initiating homing from USB? Just sending G28 or are you using a host software's homing feature (that may use other speeds?)

430984, 2020-05-06:
I am initiating homing by just sending G28. I'll try the latest bugfix code and report back to you as soon as possible.

430984, 2020-05-06:
I tried the latest bugfix which worked like it was supposed to. I came to the conclusion that this was probably my own mistake for some reason. 

I found it particularly strange that I was the only one with this issue so to rule out some stuff I also tried to go back to the firmware I had the error with in the first place and I came to the conclusion that after resetting the EEPROM (due to the version mismatch) and restarting the printer, Stealthchop works every time (even with hybrid threshold), no matter what I use to home the printer with. 

So that proves that 99% of the time the issue lies with the user ;)
Thanks anyway for your help.

closed 2020-05-06 by 430984

swilkens, 2020-05-06:
Thank you for digging a little deeper so we know this isn't a bug

430984, 2020-05-13:
Hey, I don't know if you are still receiving messages from this issue, but I found the cause of the problem. I've had it happen to me a couple more times, but I didn't want to bother you any further.

On the BTT SKR Mini E3 V1.2 is a jumper (SPREAD). That was in the wrong position. It needs to be away from the stepper connections, but it was the other way around. Found the solution on another github issue (https://github.com/MarlinFirmware/Marlin/issues/17915). Would it be an option to put that in the documentation somewhere? Jumper position in the image given in the other Github issue is correct.

swilkens, 2020-05-13:
Hey, it's good to see you found the actual problem.  

Yes, I believe we can add this to the documentation - perhaps also the configuration files. It's strange these jumpers are in the wrong position though. We should check back with BTT / SKR themselves - perhaps a manufacturing error or some other good reason.

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/248