MarlinFirmware/Marlin #17883: [BUG] LEVEL_BED_CORNERS UI quits early without input

Knifa, 2020-05-05:
### Bug Description

When using `LEVEL_BED_CORNERS` on `bugfix-2.0.x` (852a8d6685ec1137eb65e78fa748cae41fbd36b), the UI quits out early without pressing `Done`.

### My Configurations

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4581653/config.zip)

### Steps to Reproduce

1. Go to `Motion > Level Bed Corners`
2. Wait 5-10s.

**Expected behavior:** [What you expect to happen]

I should be able to proceed through the corner steps indefinitely, as on `2.0.5.3`.

**Actual behavior:** [What actually happens]

The UI exits suddenly without pressing `Done`.

#### Additional Information

This is on an Ender 3 with an SKR Mini E3 v1.2. It works OK on `2.0.5.3`!

swilkens, 2020-05-05:
I can confirm this, I see the same behaviour.

Knifa, 2020-05-05:
A new macro was added around deferring the status screen ten days ago --- maybe it's not working as expected? Will try the commit before this one. https://github.com/MarlinFirmware/Marlin/commit/c536b8de629807b489f054051bb120457f112a11 is the commit that added it. https://github.com/MarlinFirmware/Marlin/blame/bugfix-2.0.x/Marlin/src/lcd/ultralcd.h#L511

```    
FORCE_INLINE static void defer_status_screen(const bool defer=true) {
    TERN(LCD_TIMEOUT_TO_STATUS, defer_return_to_status = defer, UNUSED(defer));
}```

previously just used `#if`.

This macros are so hard to follow as well! What's going on here!? 😂

```
#define TERN(O,A,B)         _TERN(_ENA_1(O),B,A)    // OPTION converted to '0' or '1'
#define _ENA_1(O)           _ISENA(CAT(_IS,CAT(ENA_, O)))
#define _ISENA(V...)        IS_PROBE(V)
#define IS_PROBE(V...) SECOND(V, 0)     // Get the second item passed, or 0
#define SECOND(a,b,...)  b
```

Knifa, 2020-05-05:
Yes, reverting this bit of code to how it was in the prior commit i.e.:

```
    FORCE_INLINE static void defer_status_screen(const bool defer=true) {
      #if LCD_TIMEOUT_TO_STATUS
        defer_return_to_status = defer;
      #else
        UNUSED(defer);
      #endif
    }
```

solves the issue.

Something is not right with `TERN` --- perhaps it is because `LCD_TIMEOUT_TO_STATUS` has a value rather than just being empty?

Knifa, 2020-05-05:
I thought it was perhaps some kind of include ordering issue as I don't have `LCD_TIMEOUT_TO_STATUS` set in my `Configuration_adv.h` so it falls through and gets set in `Conditionals_post.h` however explicitly setting it didn't make a difference.

Knifa, 2020-05-05:
OK --- with more fanangling in Compiler Explorer](https://godbolt.org/z/Hjk-qv), it seems that the macro only works if the define has no value, which `LCD_TIMEOUT_TO_STATUS` does.

Does this even need checked? `LCD_TIMEOUT_TO_STATUS` is always set in `Conditionals_post.h` so it should always be around.

swilkens, 2020-05-06:
> previously just used `#if`. The `TERN` macro is so hard to follow! I can't make heads or tails of it.

I would agree, the code readability didn't improve with this change. Additionally, it seems some expansion is causing unexpected consequences like this one.

Knifa, 2020-05-06:
Fixed in https://github.com/MarlinFirmware/Marlin/commit/7c26a54d3f2434c4d578f58af011a76dd298d4e3.

swilkens, 2020-05-06:
I can confirm the fix works, bed corner menu now properly remains

closed 2020-05-06 by Knifa