MarlinFirmware/Marlin #17885: [BUG] "Change filament" LCD menu disapears after "resume" or "purge more", can't resume

Moyster, 2020-05-05:
### Bug Description

Using latest bugfix-2.x, with a LCD, while printing, selecting "change filament" will first :  
1. retract and park the hotend  
2. unload filament  
3. LCD menu displays something like "wait for filament unload" and "insert a new filament then press Ok to resume" <== this screen disapears but clicking the encoder makes it continue  
4. LCD menu displays "wait for hotend to heat up to temp"  
5. printer will then purge some  
6. BLOCKING STEP: LCD displays the "purge more / continue" for a split second and returns to the main menu  

After step 6, nothing can be done to resume the print, lcd menu still works (turning on/off Leds), entering menu (no idea if the encoder press does "beep" as I turned it off, but they do registers)  
(Not sure if related, I couldn't use usb and prntrface to resume / try sending Gcode, but I might have some missing drivers on this pc)

### My Configurations

[cfg.zip](https://github.com/MarlinFirmware/Marlin/files/4582235/cfg.zip)

### Steps to Reproduce

see description

**Expected behavior:** "Change filament" should allow swapping the filament and resuming the print

**Actual behavior:** resume screen disapears and blocks the print

#### Additional Information

Might actually be related to the changes following the previous bug related to the "end of print" click to resume with LCD and LEDS enabled setups.

Evg33, 2020-05-05:
#17837

Moyster, 2020-05-05:
good catch @Evg33 , couldn't find an issue that looked related to my issue ;)

Evg33, 2020-05-05:
@Moyster 
Same problem.
The menu disappears before the process is complete.

Evg33, 2020-05-05:
#17883

Moyster, 2020-05-06:
should be fixed by : https://github.com/MarlinFirmware/Marlin/commit/7c26a54d3f2434c4d578f58af011a76dd298d4e3

closed 2020-05-06 by Moyster