MarlinFirmware/Marlin #17889: [BUG] v.2.0.5.3 print timer stop count after 1 minute and doesn't update anymore

samuelcarreira, 2020-05-05:
### v.2.0.5.3 print timer stop count after 1 minute

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4583792/Marlin.zip)

**Expected behavior:**
Print time on LCD counts correctly (the last field on the 20x4 LCD screen - line bellow coordinates) 

**Actual behavior:**
It starts to count but stucks around 1 minute. After the print was done it shows the total print time on the last line of the LCD screen but the timer doesn't update the 1 minute mark

#### Additional Information
I didn't have any issues with previous Marlin versions (only on 2.0.5.3)
I am using an Anet A8 and compile with PlatformIO - default bootloader and Mezi platform

thinkyhead, 2020-05-06:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

samuelcarreira, 2020-05-13:
I couldn't compile it. It outputs a lot of errors

```
Compiling .pio\build\melzi\src\src\HAL\AVR\Servo.cpp.o
In file included from Marlin\src\HAL\AVR\../../inc/../pins/pins.h:71:0,
                 from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:32,
                 from Marlin\src\HAL\AVR\HAL_SPI.cpp:34:
Marlin\src\HAL\AVR\../../inc/../pins/ramps/pins_RAMPS.h:56:6: error: #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
     #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
      ^
In file included from Marlin\src\HAL\AVR\../../inc/../pins/pins.h:71:0,
                 from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:32,
                 from Marlin\src\HAL\AVR\HAL.cpp:24:
Marlin\src\HAL\AVR\../../inc/../pins/ramps/pins_RAMPS.h:56:6: error: #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
     #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
      ^
In file included from Marlin\src\HAL\AVR\../../inc/../pins/pins.h:71:0,
                 from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:32,
                 from Marlin\src\HAL\AVR\MarlinSerial.cpp:39:
Marlin\src\HAL\AVR\../../inc/../pins/ramps/pins_RAMPS.h:56:6: error: #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
     #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
      ^
*** [.pio\build\melzi\src\src\HAL\AVR\HAL_SPI.cpp.o] Error 1
*** [.pio\build\melzi\src\src\HAL\AVR\HAL.cpp.o] Error 1
*** [.pio\build\melzi\src\src\HAL\AVR\MarlinSerial.cpp.o] Error 1
In file included from Marlin\src\HAL\AVR\../../inc/../pins/pins.h:71:0,
                 from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:32,
                 from Marlin\src\HAL\AVR\Servo.cpp:56:
Marlin\src\HAL\AVR\../../inc/../pins/ramps/pins_RAMPS.h:56:6: error: #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
     #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
      ^
*** [.pio\build\melzi\src\src\HAL\AVR\Servo.cpp.o] Error 1
```

samuelcarreira, 2020-05-13:
Output log after print with USB cable (Pronterface)

> ```
Loading file: ...Desktop\tampa.gcode
Loaded ...\Desktop\tampa.gcode, 36782 lines
2316.36mm of filament used in this print
The print goes:
- from 0.00 mm to 128.26 mm in X and is 128.26 mm wide
- from 0.00 mm to 124.41 mm in Y and is 124.41 mm deep
- from 0.00 mm to 34.06 mm in Z and is 34.06 mm high
Estimated duration: 158 layers, 0:43:10
Print started at: 21:42:30
echo:Print time: 1m 3s
Done printing file```

As you can see, the print time is incorrect, the print continues after that output

GCODE file slicer: Ideamaker 3.6.0
[tampa.zip](https://github.com/MarlinFirmware/Marlin/files/4624720/tampa.zip)

boelle, 2020-06-20:
@samuelcarreira still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle