MarlinFirmware/Marlin #17890: Can't compile Marlin on Ender 3

OscarKro, 2020-05-05:
### Bug Description

I can't get the Ender 3 Marlin Config to compile with the Arduino IDE. I Tried using the bugfix branch of both the Marlin Repo and the Config repo.  I just take the normal Ender 3 (not the V2) config files, copy paste it to the Marlin folder and leave it as is. I don't even upload it yet. I just verify it.

### My Configurations

I'm not sure how to add a zipfile here? never done this. But it's just the stock Ender 3 Configuration files from the config branch, nothing altered. I tried both the ones from the master branch, and the ones from the bugfix branch

### Steps to Reproduce

1. download the latest Arduino IDE, the Sanguino stuff, etc.
2. cloned the Marlin repo and the config repo
3. copied the files from the config to the marlin folder
4. open the marlin.ino
5. hit the verify button
6. stopped compilation at the end
7. do the same stuff but switch to the bugfix branch in the config repo, to no different result
8. do the stame stuff but switch to the bugfix branch in the Marlin repo, to no different result

**Expected behavior:** 
To be able to compile Marlin... Or get help in doing the right steps, maybe i'm missing something?
**Actual behavior:** 

#### Additional Information

Error message:
Arduino:1.8.12 (Windows 10), Board:"Sanguino, ATmega1284 or ATmega1284P (16 MHz)"

sketch\src\module\motion.cpp:77:71: error: 'X_HOME_BUMP_MM' was not declared in this scope

 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }

                                                                       ^

sketch\src\module\motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'

 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);

 ^~~~~~~~~~

sketch\src\module\motion.cpp:77:71: note: suggested alternative: 'HOMING_BUMP_MM'

 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }

                                                                       ^

sketch\src\module\motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'

 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);

 ^~~~~~~~~~

sketch\src\module\motion.cpp:77:80: error: 'Y_HOME_BUMP_MM' was not declared in this scope

 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }

                                                                                ^

sketch\src\module\motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'

 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);

 ^~~~~~~~~~

sketch\src\module\motion.cpp:77:80: note: suggested alternative: 'HOMING_BUMP_MM'

 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }

                                                                                ^

sketch\src\module\motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'

 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);

 ^~~~~~~~~~

sketch\src\module\motion.cpp:77:89: error: 'Z_HOME_BUMP_MM' was not declared in this scope

 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }

                                                                                         ^

sketch\src\module\motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'

 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);

 ^~~~~~~~~~

sketch\src\module\motion.cpp:77:89: note: suggested alternative: 'HOMING_BUMP_MM'

 #define XYZ_CONSTS(T, NAME, OPT) const PROGMEM XYZval<T> NAME##_P = { X_##OPT, Y_##OPT, Z_##OPT }

                                                                                         ^

sketch\src\module\motion.cpp:83:1: note: in expansion of macro 'XYZ_CONSTS'

 XYZ_CONSTS(float, home_bump_mm,   HOME_BUMP_MM);

 ^~~~~~~~~~

exit status 1
Fout bij het compileren voor board Sanguino

Dit rapport zou meer informatie bevatten met
"Uitgebreide uitvoer weergeven tijden compilatie"
optie aan in Bestand -> Voorkeuren.

OscarKro, 2020-05-05:
##FIXED##
Allright, I jsut fixed it and went for help 1 minute to soon :) sorry for losing anyone its time. 
The only combo I haven't tried was having both the config and marlin repo on the bugfix branch, I always had one on there. 
So for anyone wondering: 
Both the Marlin and the Config repository need to be on the bugfix branch. Than do the usual and it will verify (compile). 
Again sorry if i already wasted anyones time.

closed 2020-05-05 by OscarKro

swilkens, 2020-05-08:
For reference, the firmware does have a readme which gives you the link to the correct configuration files.

In `Marlin-bugfix-2.0.x\config` you will find a README with this link:

https://github.com/MarlinFirmware/Configurations/archive/bugfix-2.0.x.zip