MarlinFirmware/Marlin #17891: [BUG] (LCD "flow"attribute is saving something wrong in EEPROM)

betocabus, 2020-05-06:
I'm used to play with flow atribute during the print. Sometimes change some settings in LCD and then save the modifications in EEPROM. Every modification I made in EEPROM, I take note to certify never lost the corrects values. New day, I turn on the print, no modifications just putting to print, realize that the flow is very very less then normal(+- 60%), checked everything, noise, steps of the motor, even measured if the amount of filament extruded is ok - extrude 100mm in LCD, extruded 100mm, everything was OK.

Solution: 
1-Checked M503
2-writed every value in my notes and erase EEPROM by M502.
3-putted the same values again and everything back to normal.

In the last time, in marlin 2.0.5.3 realize something:
the problem was happening, puted my flow to 160% printing well..
-Stopped the print to fix it
-M503 to reset EEPROM
-Putted the values again
-DIDN'T TURN OFF THE PRINT and tested it again, PRINTING GOOD, checked the flow, it was still in 160%, even with m502, puted in 100%, BAD flow again. Reseted the printer, checked the flow, was 100%, tested, Printing good.

I think the "flow" in LCD is modifying something that can be saved in EEPROM and the way it's happening is wrong because the steps/mm of the motor is the same in every situation. For me it's a BUG.

Every single test I checked my steps/mm of the extruder and it was the same, all the time.

Happened with me 3 times, 2 different printers, both coreXY

one in marlin 1.1.9 and today in marlin 2.0.5.3 (two different printers) 

didn't remember if happened one more time in 1.1.8 or again in 1.1.9.

Thank you guys for all! Expect this information can help to solve this.
[Configuration2.0.5.3.zip](https://github.com/MarlinFirmware/Marlin/files/4584346/Configuration2.0.5.3.zip)
[Configuration1.1.9.zip](https://github.com/MarlinFirmware/Marlin/files/4584350/Configuration1.1.9.zip)

thinkyhead, 2020-05-06:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

thinkyhead, 2020-05-06:
Note that flow is not a value that is saved to EEPROM. It is reset to 100% on boot, and it will typically be set to some value near 100% by your slicer's starting G-code on every print job.

betocabus, 2020-05-06:
Sure! Will try to reproduce and test in bugfix branch as well

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle