MarlinFirmware/Marlin #17892: [BUG] Marlin 2.0.5.3 - M205 jerk settings do nothing

jackw01, 2020-05-06:
### Bug Description

I just installed marlin 2.0.5.3 on a new CoreXY printer I built and enabled classic jerk settings in configuration due to issues with junction deviation (already reported on someone else's Github issue), but changing the jerk settings via M205 or the LCD has absolutely no effect on motion. Setting the X and Y jerk values to 0 should prevent the printer from moving entirely, but it still moves the same as it does when the X and Y jerk are set to the default value of 10. Setting them excessively high should cause the motors to stall but this does not happen either.

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4584804/Marlin.zip)

### Steps to Reproduce

1. Enable classic jerk in configuration
2. Set jerk values using M205 or LCD

**Expected behavior:** 
Newly set jerk values should affect how the printer moves

**Actual behavior:** 
Printer moves the same as it did with the default jerk values

thinkyhead, 2020-05-10:
> Setting the X and Y jerk values to 0 should prevent the printer from moving entirely

Where did you get that idea?

Setting Jerk to 0 only means that there is no "instantaneous" change in feedrate permitted. Instead, every move is accelerated from zero / decelerated to zero. Jerk describes the largest "instantaneous" change in feedrate that is allowed between segments when there is a change in direction on an axis.

jackw01, 2020-05-10:
I remember it doing this in older versions. Still, it doesn't seem like setting the jerk to 0 or very high numbers affects the motion at all.

boelle, 2020-06-20:
@jackw01 is this still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle