MarlinFirmware/Marlin #17894: [BUG] (multiple issues by compling skr 1.4 Turbo with Ender4)

thorstensd, 2020-05-06:
Hello,

I´m using the Marlin Bugfix 2.0.x from today with Ender 4 configuration files to compile a FW for BOARD_BTT_SKR_V1_4_TURBO and got multiple compiling errors:

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1769.html
PLATFORM: NXP Arduino LPC176x 0.1.1 > NXP LPC1769
HARDWARE: LPC1769 120MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES:
 - framework-arduino-lpc176x 0.2.3
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 6 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <LiquidCrystal> 1.0.0
|-- <U8glib-HAL> 0.4.1
|-- <TMCStepper> 0.6.2
|-- <Adafruit NeoPixel> 1.3.0
|-- <SailfishLCD>
Building in release mode

Unable to find destination disk (Autodetect Error)
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or copy the firmware (.pio/build/LPC1769/firmware.bin) manually to the appropriate disk

Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\DebugMonitor.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\HAL.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\HAL_SPI.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\MarlinSerial.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_flash.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_sdcard.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_wired.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\fast_pwm.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\include\digipot_mcp4451_I2C_routines.c.o
**In file included from Marlin\src\HAL\LPC1768\../../core/../inc/MarlinConfig.h:40,
                 from Marlin\src\HAL\LPC1768\../../core/serial.h:24,
                 from Marlin\src\HAL\LPC1768\DebugMonitor.cpp:25:
Marlin\src\HAL\LPC1768\../../core/../inc/SanityCheck.h:499:4: error: #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. Please update Configuration_adv.h."
  499 |   #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. Please update Configuration_adv.h."
      |    ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:40,
                 from Marlin\src\HAL\LPC1768\HAL.cpp:24:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:499:4: error: #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. Please update Configuration_adv.h."
  499 |   #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. Please update Configuration_adv.h."
      |    ^~~~~
Marlin\src\HAL\LPC1768\../../core/../inc/SanityCheck.h:1553:6: error: #error "CASE_LIGHT_ENABLE requires CASE_LIGHT_PIN to be defined."
 1553 |     #error "CASE_LIGHT_ENABLE requires CASE_LIGHT_PIN to be defined."
      |      ^~~~~
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1553:6: error: #error "CASE_LIGHT_ENABLE requires CASE_LIGHT_PIN to be defined."
 1553 |     #error "CASE_LIGHT_ENABLE requires CASE_LIGHT_PIN to be defined."
      |      ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:40,
                 from Marlin\src\HAL\LPC1768\eeprom_flash.cpp:39:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:499:4: error: #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. Please update Configuration_adv.h."
  499 |   #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. Please update Configuration_adv.h."
      |    ^~~~~
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1553:6: error: #error "CASE_LIGHT_ENABLE requires CASE_LIGHT_PIN to be defined."
 1553 |     #error "CASE_LIGHT_ENABLE requires CASE_LIGHT_PIN to be defined."
      |      ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:40,
                 from Marlin\src\HAL\LPC1768\HAL_SPI.cpp:51:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:499:4: error: #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. Please update Configuration_adv.h."
  499 |   #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. Please update Configuration_adv.h."
      |    ^~~~~
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1553:6: error: #error "CASE_LIGHT_ENABLE requires CASE_LIGHT_PIN to be defined."
 1553 |     #error "CASE_LIGHT_ENABLE requires CASE_LIGHT_PIN to be defined."
      |      ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:40,
                 from Marlin\src\HAL\LPC1768\eeprom_sdcard.cpp:25:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:499:4: error: #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. Please update Configuration_adv.h."
  499 |   #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. Please update Configuration_adv.h."
      |    ^~~~~
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1553:6: error: #error "CASE_LIGHT_ENABLE requires CASE_LIGHT_PIN to be defined."
 1553 |     #error "CASE_LIGHT_ENABLE requires CASE_LIGHT_PIN to be defined."
      |      ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:40,
                 from Marlin\src\HAL\LPC1768\eeprom_wired.cpp:24:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:499:4: error: #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. Please update Configuration_adv.h."
  499 |   #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. Please update Configuration_adv.h."
      |    ^~~~~
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1553:6: error: #error "CASE_LIGHT_ENABLE requires CASE_LIGHT_PIN to be defined."
 1553 |     #error "CASE_LIGHT_ENABLE requires CASE_LIGHT_PIN to be defined."
      |      ^~~~~
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\include\i2c_util.c.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\main.cpp.o
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\HAL.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\DebugMonitor.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_flash.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\HAL_SPI.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_sdcard.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_wired.cpp.o] Error 1
In file included from Marlin\src\HAL\LPC1768\../../sd/../inc/MarlinConfig.h:40,
                 from Marlin\src\HAL\LPC1768\../../sd/cardreader.h:24,
                 from Marlin\src\HAL\LPC1768\main.cpp:38:
Marlin\src\HAL\LPC1768\../../sd/../inc/SanityCheck.h:499:4: error: #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. Please update Configuration_adv.h."
  499 |   #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. Please update Configuration_adv.h."
      |    ^~~~~
Marlin\src\HAL\LPC1768\../../sd/../inc/SanityCheck.h:1553:6: error: #error "CASE_LIGHT_ENABLE requires CASE_LIGHT_PIN to be defined."
 1553 |     #error "CASE_LIGHT_ENABLE requires CASE_LIGHT_PIN to be defined."
      |      ^~~~~
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\main.cpp.o] Error 1**

LPC1769                   FAILED    00:00:05.950

swilkens, 2020-05-06:
It looks like you are using old configuration files:

> Marlin\src\HAL\LPC1768../../core/../inc/SanityCheck.h:499:4: error: #error "[XYZ]_HOME_BUMP_MM is now HOMING_BUMP_MM. **Please update Configuration_adv.h.**"

When using the `bugfix-2.0.x` branch of the firmware, please use up to date configuration files from:

https://github.com/MarlinFirmware/Configurations/archive/bugfix-2.0.x.zip

thorstensd, 2020-05-06:
yes, your write - what a mess. the Home Bump has been solved now. But the others a still available.

Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\include\digipot_mcp4451_I2C_routines.c.o
In file included from Marlin\src\HAL\LPC1768\../../core/../inc/MarlinConfig.h:40,
                 from Marlin\src\HAL\LPC1768\../../core/serial.h:24,
                 from Marlin\src\HAL\LPC1768\DebugMonitor.cpp:25:
Marlin\src\HAL\LPC1768\../../core/../inc/SanityCheck.h:1553:6: error: #error "CASE_LIGHT_ENABLE requires CASE_LIGHT_PIN to be defined."
 1553 |     #error "CASE_LIGHT_ENABLE requires CASE_LIGHT_PIN to be defined."
      |      ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:40,
                 from Marlin\src\HAL\LPC1768\HAL.cpp:24:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1553:6: error: #error "CASE_LIGHT_ENABLE requires CASE_LIGHT_PIN to be defined."
 1553 |     #error "CASE_LIGHT_ENABLE requires CASE_LIGHT_PIN to be defined."
      |      ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:40,
                 from Marlin\src\HAL\LPC1768\HAL_SPI.cpp:51:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1553:6: error: #error "CASE_LIGHT_ENABLE requires CASE_LIGHT_PIN to be defined."
 1553 |     #error "CASE_LIGHT_ENABLE requires CASE_LIGHT_PIN to be defined."
      |      ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:40,
                 from Marlin\src\HAL\LPC1768\eeprom_flash.cpp:39:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1553:6: error: #error "CASE_LIGHT_ENABLE requires CASE_LIGHT_PIN to be defined."
 1553 |     #error "CASE_LIGHT_ENABLE requires CASE_LIGHT_PIN to be defined."
      |      ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:40,
                 from Marlin\src\HAL\LPC1768\eeprom_sdcard.cpp:25:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1553:6: error: #error "CASE_LIGHT_ENABLE requires CASE_LIGHT_PIN to be defined."
 1553 |     #error "CASE_LIGHT_ENABLE requires CASE_LIGHT_PIN to be defined."
      |      ^~~~~
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:40,
                 from Marlin\src\HAL\LPC1768\eeprom_wired.cpp:24:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:1553:6: error: #error "CASE_LIGHT_ENABLE requires CASE_LIGHT_PIN to be defined."
 1553 |     #error "CASE_LIGHT_ENABLE requires CASE_LIGHT_PIN to be defined."
      |      ^~~~~
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\include\i2c_util.c.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\main.cpp.o
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\HAL.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\HAL_SPI.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\DebugMonitor.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_flash.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_sdcard.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_wired.cpp.o] Error 1
In file included from Marlin\src\HAL\LPC1768\../../sd/../inc/MarlinConfig.h:40,
                 from Marlin\src\HAL\LPC1768\../../sd/cardreader.h:24,
                 from Marlin\src\HAL\LPC1768\main.cpp:38:
Marlin\src\HAL\LPC1768\../../sd/../inc/SanityCheck.h:1553:6: error: #error "CASE_LIGHT_ENABLE requires CASE_LIGHT_PIN to be defined."
 1553 |     #error "CASE_LIGHT_ENABLE requires CASE_LIGHT_PIN to be defined."
      |      ^~~~~
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\main.cpp.o] Error 1

ellensp, 2020-05-06:
you need a #define  CASE_LIGHT_PIN   {some pin on your board}

swilkens, 2020-05-06:
You are trying to use a case light, but no pin for this has been defined on your board.

To compile the firmware without the case light support, change:

> #define CASE_LIGHT_ENABLE

to 

> //#define CASE_LIGHT_ENABLE

In `Configuration_adv.h`

thorstensd, 2020-05-06:
its working - thank you very much.

closed 2020-05-06 by thorstensd