MarlinFirmware/Marlin #17895: [FR] Validate endstop state during bumping

fabianonline, 2020-05-06:
### Description

Yesterday, the cable connecting my Y endstop came loose, leading to the endstop being "TRIGGERED" all the time. During homing, Marlin didn't notice it and just assumed that the Y axis was already at 0 - leading to the Y axis hitting the mechanical borders of movement and stuff.

I think this problem could be recognized when using bumping: After the carriage has been moved away by Y_HOME_BUMP_MM, the endstop should register "Open". This could be detected by Marlin, maybe as a part of VALIDATE_HOMING_ENDSTOPS.

### Feature Workflow

Pseudo-code of homing:

1. Move in X/Y/Z direction until the corresponding endstop gets triggered.
2. Move HOME_BUMP_MM back.
3. **Check that the endstop is no longer triggered. If it is, stop the print.** (This is the new step.)
4. Move in X/Y/Z direction again until the endstop is triggered.

#### Additional Information

* This check can only be done safely if HOME_BUMP_MM is used.
* You could have set HOME_BUMP_MM to a very small value that doesn't release the endstop. This would trigger an error. But then again, a) the user has to have VALIDATE_HOMING_ENDSTOPS enabled so those errors should be more or less expected anyways and b) your bumping would be completely useless and having your printer alert you to this might noe be the worst idea, either.

Speaka, 2020-06-25:
I had a similar issue with a semi broken endstop cable, which triggered the es right before it shoul. Resulting in several crashes on axis until i figured out what was wrong. I set up this (for me) working code draft. It stops the machine if no change in endstop state is detected. Help is much appreciated!

Speaka, 2020-07-12:
this issue can be closed, solution was provided.
@boelle @fabianonline

closed 2020-07-12 by thisiskeithb