MarlinFirmware/Marlin #17898: [BUG] (short description)

Haschtl, 2020-05-06:
### Bug Description

Unable to save configuration to EEPROM.

### My Configurations

Board: SKR Pro v1.1 (with FLASH_EEPROM_EMULATION
)
Marlin-branch: bugfix-v2.0.x

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4587266/config.zip)

### Steps to Reproduce

```
Send: M502
Recv: echo:Hardcoded Default Settings Loaded
Recv: ok P31 B7
[...]
Send: M500
Recv: echo:Settings Stored (744 bytes; crc 45646)
Recv: //action:notification Einstell. gespei.
Recv: ok P31 B7
[...]
Send: M501
Recv: Error:Field home_offset mismatch.
Recv: Error:EEPROM datasize error.
Recv: echo:Index: 668 Size: 744
Recv: //action:notification Err: EEPROM Index
Recv: echo:Hardcoded Default Settings Loaded
Recv: ok P31 B7
```

**Expected behavior:** [What you expect to happen]
I expect it to save the congifuration to EEPROM without errors.

**Actual behavior:** [What actually happens]
Its not beeing saved to the EEPROM

tpruvot, 2020-05-06:
Same on the STM32F1 HAL... after an eeprom version change, i need to save twice (M500 M500) to get it really saved...

Haschtl, 2020-05-06:
I suppose it's different to your issue... 
I cannot check, if the eeprom is saved correctly, because I cannot read the eeprom. 

Is there any other persistent eeprom emulation i can try? Sram is not persistent, does sdcard emulation work on a stm32f4?

thisiskeithb, 2020-05-06:
> Same on the STM32F1 HAL... after an eeprom version change, i need to save twice (M500 M500) to get it really saved...

I noticed this quirk on a beta BTT002 (also `STM32F4`) and wondered if it was the board or something else upstream.

Haschtl, 2020-05-15:
Issue fixed with #17946 i think.
Also thr 2x M500 should be fixed

closed 2020-05-15 by Haschtl