MarlinFirmware/Marlin #17899: trinamic.app unused variable

thorstensd, 2020-05-06:
Hello,

I detected a problem during compile SKR 1.4 Turbo with TMC2208 and ENDER 4 Conf.

trinamic.app unused variable:

 void tmc_serial_begin() {
    struct {
      const void *ptr[TMCAxis::TOTAL];
      bool began(const TMCAxis a, const void * const p) {
        LOOP_L_N(i, a) if (p == ptr[i]) return true;
        ptr[a] = p; return false;
      };
    } sp_helper;

ellensp, 2020-05-06:
its only a warning in trinamic.cpp
```
Marlin/src/module/stepper/trinamic.cpp: In function 'void tmc_serial_begin()':
Marlin/src/module/stepper/trinamic.cpp:327:7: warning: unused variable 'sp_helper' [-Wunused-variable]
  327 |     } sp_helper;
      |       ^~~~~~~~~
```

thisiskeithb, 2020-05-07:
> its only a warning in trinamic.cpp

In other words, it can be ignored 🙂

closed 2020-05-07 by thorstensd

thinkyhead, 2020-05-07:
To suppress the warning, wrap that section with `HAS_TMC_UART`…
```cpp
#if HAS_TMC_UART
  struct {
    const void *ptr[TMCAxis::TOTAL];
    bool began(const TMCAxis a, const void * const p) {
      LOOP_L_N(i, a) if (p == ptr[i]) return true;
      ptr[a] = p; return false;
    };
  } sp_helper;
  #define HW_SERIAL_BEGIN(A) do{ if (!sp_helper.began(TMCAxis::A, &A##_HARDWARE_SERIAL)) \
                                       A##_HARDWARE_SERIAL.begin(TMC_BAUD_RATE); }while(0)
#endif
```

sjasonsmith, 2020-05-07:
I already posted a PR for it. The solution is a little different than you suggested because that would still warn if you are using software serial.