MarlinFirmware/Marlin #17900: [BUG] (Enabling SPI_ENDSTOPS causes compile error for mixed drivers)

petermccracken, 2020-05-06:
### Bug Description

When using 4 x TMC2209 and 1 x TMC5160 and want to use sensorless homing on both driver types. If SPI_ENDSTOPS is define in the advanced config an error will be thrown during the build task...

Marlin\src\module\endstops.cpp: In static member function 'static bool Endstops::tmc_spi_homing_check()':
Marlin\src\module\endstops.cpp:834:40: error: 'class TMCMarlin<TMC2209Stepper, 'X', '0', X_AXIS>' has no member named 'test_stall_status'
  834 |       if (tmc_spi_homing.x && stepperX.test_stall_status()) {

This looks like the code cannot with a mix of UART/SPI based drivers when sensorless homing is configured, specifically the SPI based sensorless homing on the 5160 as this seems to cause a conflict with the 2209 driver code.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Configure a mix of 2209 and 5160 drivers.
2. Enable sensorless homing.
3. define SPI_ENDSTOPS in the advanced config.

**Expected behavior:**
The build task works and the sensorless homing works for 2209 UART and 5160 SPI.

**Actual behavior:** 
The build task fails with above error citing 2209 stepper error. The SPI_ENDSTOP should be ignored by the 2209 code.

#### Additional Information

Tested using most recent Marlin build available for download v2.0.5.3

K2Robotics, 2020-05-07:
Hi Peter. I just copied the 2328 line in the Configuration_adv.h file: 
//#define SPI_ENDSTOPS              // TMC2130 only
As you can see it seems to work only for the 2130 drivers.  I'm just testing that feature yesterday on may printer and seems to not work well any way.

petermccracken, 2020-05-07:
I think for all practical purposes the 2130 and 5160 are pretty much interchangeable so the bug report still stands.

petermccracken, 2020-05-07:
Basically what I am saying is that if I populate the board with all 5160 drivers the firmware will at least compile, when I mix with 2209 it won't compile.

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

MTK4355, 2020-07-13:
I'm having the very same issue using a mix of 2209 and 2130 drivers exactly as the OP mentioned in this bug report.  Should I provide you with my config here or open a new report?

thinkyhead, 2020-07-13:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

closed 2020-08-18 by github-actions[bot]