MarlinFirmware/Marlin #17905: [BUG] Possible?. 2209's with sensorless homing and Linear advance

ibsmith04, 2020-05-07:
### Bug Description
I recently upgraded my ender 3 to an SKR 1.4 pro with 2209's, sensorless homing and a BLTouch.
Initially I decided to not include Linear Advance, but did have S_Curve included.
I had a lot of trouble fine tuning the 2209s for sensorless, but after reading that the current as well as the bump (M914) affected sensorless operation, I was able to get it all working quite nicely.
No false positives and found the ends every time, and quietly as well.

Today I decided to try out Linear_advance, compiled a new version with Linear Advance included and S_Curve switched off, I set a default for K of 0.0

When I issued a G28, the hotend does not move towards x or y, usually meaning that the M914 is too sensitive.. So I lowered it from 65 to 50. Now issuing a g28 x the head moves towards the endstop, but doesn't stop when it hits it and I have to switch off the printer.

So I tried tuning M914, and eventually discovered that with M914 X58, it is too sensitive and G28 x is triggered without the head moving at all, and if is M914 X57, then a g28 x will cause it keep moving thro the endstop until I switch off the printer. Ie it is impossible to tune.

I started trying different currents, and found that if I reduce the current I then can tune M914 and get it to work properly.

The question is, why does including Linear_advance change the bump sensitivity ?
Why do I have to set a lower current on x&y with Linear_Advance to get sensorless working again.
I would have thought that if Linear_Advance affected anything, it would be the Extruder, which doesn't come into the sensorless equation

### My Configurations
[IansConfigNoLinearAdvamce.zip](https://github.com/MarlinFirmware/Marlin/files/4592417/IansConfigNoLinearAdvamce.zip)

### Steps to Reproduce

Build Marlin with Linear_Advance commented out and S_Curve included
Tune 2209's for sensorless homing

Turn on Linear Advance and disable S_Curve in firmware

**Expected behavior:** [What you expect to happen]
G28 to work

**Actual behavior:** [What actually happens]
G28 only works if I reduce the current to x & y 2209s and retune M914

Thank you
Ian

tomarigr, 2020-05-07:
I can confirm that this is the case as I run into exactly the same problem when trying to enable sensorless homing too. 
Skr e3 mini

jreis1, 2020-05-11:
Do you have QuickHome enabled ? I found sensitivity issues in sensorless homing when QuickHome is enabled. Maybe this is related. 
Check issue:  https://github.com/MarlinFirmware/Marlin/issues/17687

ibsmith04, 2020-05-11:
I just checked, and I do not have QuickHome enabled.

Cheers

SFRSKZ, 2020-05-14:
i might have same problem. la is enabled and sensorless working randomly. updated markin from 2.0.3 to 2.0.5.3

rado79, 2020-05-14:
with LA enabled and QuickHome disabled `X_CURRENT_HOME` (X_CURRENT / 2) does the job in my case, same for `Y_CURRENT`. with this setting i noticed a better tuning for `STALL_SENSITIVITY`

SKR 1.3, TMC2209, `X_CURRENT` `Y_CURRENT` set to 700mA, `STALL_SENSITIVITY` 105, CoreXY

boelle, 2020-06-20:
@ibsmith04 still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle