MarlinFirmware/Marlin #17906: [BUG] BTT SKR PRO  onboard SD Card - detect pin still used from LCD (bugfix-2.0)

unl34shed, 2020-05-07:
### Bug Description

For the Bigtreetech SKR Pro 1.1 the SD_DETECT_PIN is still connected to the LCD although using internal SD card slot.

### My Configurations

Using bugfix-2.0.x from 7th May.
Issue is in pins_BTT_SKR_PRO_V1_1.h
SD_DETECT_PIN is only configured in the LCD routine and not selection is done based on the SD_CARDCONNECTION setting.

### Steps to Reproduce
- Configure SKR PRO to use onboard SD card slot
- Flash
- insert SD Card in onboard slot (Media inserted not detected)
- insert a second SD Card to the LCD slot (media inserted is detected)
- read data -> data from onboard slot is shown

**Expected behavior:** [What you expect to happen]

When inserting SD card in the onboard slot the media inserted should be detected.

**Actual behavior:** [What actually happens]

Media detected and removed is detected from the LCD SD slot but communication is done on the onboard SD slot.

#### Additional Information

Attached my proposal to fix this:
- added onboard SD_DETECT_PIN to #if SD_CONNECTION_IS(ONBOARD) cas
#if SD_CONNECTION_IS(ONBOARD)
  #define SOFTWARE_SPI                            // Use soft SPI for onboard SD
  #define SD_DETECT_PIN                     PB11
...
- moved LCD_SDSS and SD_DETECT_PIN definition to only be used when LCD is selected
  #if SD_CONNECTION_IS(LCD)
    #define SDSS                          PB12  // Uses default hardware SPI for LCD's SD
    #define SD_DETECT_PIN                 PF12
    #define LCD_SDSS                      PB12
  #endif
...

Tested it with SKR PRO and RepRapDiscount Full Graphic Smart Controller. Both SD Card Slots detect when the card is inserted/removed based on the config.

[pins_BTT_SKR_PRO_V1_1.h.txt](https://github.com/MarlinFirmware/Marlin/files/4592887/pins_BTT_SKR_PRO_V1_1.h.txt)

boelle, 2020-06-20:
@unl34shed is this still an issue?

unl34shed, 2020-06-21:
Hi boelle,

I just checked the sources of the latest nightly build and I would say yes it is still the same issue.

SD_DETECT_PIN is still hard coded to PF12 regardless where the SD_CONNECTION define is pointing. The same is true for LCD_SDSS

The code just moved to a new file pins_BTT_SKR_PRO_common due to a new revision of the board.

I haven't flashed it thought, got my board running with my workaround since then.

sjasonsmith, 2020-07-04:
When this is fixed, it _might_ be possible to enable SDIO for the LCD slot, if it is actually using the correct hardware pins.

sjasonsmith, 2020-08-16:
I just tried out the changed proposed in this report. I was able to access contents on the on-board SD card. I'm not entirely sure if the detect pin is working. Is there something I should be able to observe, like a print in the console? I rarely use an SD card so am not sure what to expect from that.

Some care will be needed when applying this change, to be sure it also is correct for the SKR Pro 1.2.

sjasonsmith, 2020-08-16:
Detect wasn't working because I had to uncomment `NO_SD_HOST_DRIVE`. It works now with the detect pin listed in this report.

sjasonsmith, 2020-08-16:
Well, it is a bit silly this took over 3 months to be implemented, but a PR is now up for it.

sjasonsmith, 2020-09-11:
There haven't been any comments since my post 26 days ago. I think we sorted all of this out properly, and several of us tested various combinations of hardware with the change.

I am presuming this is fixed and closing this issue.

closed 2020-09-11 by sjasonsmith