MarlinFirmware/Marlin #17910: SKR Pro 1.1 Fan 1 and stopped working 2.0.5.3 , works in 2.0.5.2

JimGWIn, 2020-05-07:
### Bug Description
SKR Pro 1.1 Fan 1 and stopped working 2.0.5.3 , works in 2.0.5.2
### My Configurations
[MarlinConfics2.0.5.3.zip](https://github.com/MarlinFirmware/Marlin/files/4594868/MarlinConfics2.0.5.3.zip)
### Steps to Reproduce
 I use fan 0 for print cooling , 1 and 2 for board and driver cooling. M106 P2 S255 , M106 P3 S255
is sent at the binning of the G code it works in 2.0.5.2 but in 2..0.5.3 i can,t even turn them on via the menu. I have built 2 almost identical printers and tried this on both, same thing..
thank you.

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

GhostlyCrowd, 2020-05-09:
Not a bug, you want to look into Auto Extruger cooling in the configuration and controller cooling. 

Heatbreak cooking for auto extruder will turn on when configured at a set temp and then shut off when its not hot. and Controller cooling will turn on when the steppers are active and shut off when they are not. M106 no longer controls those fans when configured like this. 

#define USE_CONTROLLER_FAN

**
 * Extruder cooling fans
 *
 * Extruder auto fans automatically turn on when their extruders'
 * temperatures go above EXTRUDER_AUTO_FAN_TEMPERATURE.
 *
 * Your board's pins file specifies the recommended pins. Override those here
 * or set to -1 to disable completely.
 *
 * Multiple extruders can be assigned to the same pin in which case
 * the fan will turn on when any selected extruder is above the threshold.
 */
//#define E0_AUTO_FAN_PIN -1

Configuration_adv.h

JimGWIn, 2020-05-10:
Perfect, thank you for your time.

From: GhostlyCrowd
Sent: Friday, May 8, 2020 8:48 PM
To: MarlinFirmware/Marlin
Cc: JimGWIn; Author
Subject: Re: [MarlinFirmware/Marlin] SKR Pro 1.1 Fan 1 and stopped working2.0.5.3 , works in 2.0.5.2 (#17910)

Not a bug, you want to look into Auto Extruger cooling in the configuration and controller cooling.
Heatbreak cooking for auto extruder will turn on when configured at a set temp and then shut off when its not hot. and Controller cooling will turn on when the steppers are active and shut off when they are not. M106 no longer controls those fans when configured like this.
#define USE_CONTROLLER_FAN
**
• Extruder cooling fans
• 
• Extruder auto fans automatically turn on when their extruders'
• temperatures go above EXTRUDER_AUTO_FAN_TEMPERATURE.
• 
• Your board's pins file specifies the recommended pins. Override those here
• or set to -1 to disable completely.
• 
• Multiple extruders can be assigned to the same pin in which case
• the fan will turn on when any selected extruder is above the threshold.
*/
//#define E0_AUTO_FAN_PIN -1
Configuration_adv.h

sjasonsmith, 2020-05-29:
@JimGWIn if @GhostlyCrowd's answer satisfied your needs, can you go ahead and close this issue? It isn't clear to me whether there is anything actionable in this Issue.

JimGWIn, 2020-05-29:
Thanks guys and gals, my mistake.
I apologize for forgetting to close ,the sun came out and I had to prep the greenhouse for the heat.

closed 2020-05-29 by JimGWIn