MarlinFirmware/Marlin #17912: [BUG] Override CASE_LIGHT_PIN not working

Fixx, 2020-05-07:
Uncommenting and defining a PIN number not working via option //#define CASE_LIGHT_PIN in Configuration_adv. I need to edit board's pin file, same pin number works that way.

randellhodges, 2020-05-07:
For what board? You'll need to add your config files, pin file, what version of Marlin are you using, etc, etc.

I am using it without problems on about a month old pull of bugfix w/o problems.

Fixx, 2020-05-07:
I didn't want to clutter this topic with config files as issue is pretty basic. I'm using RUMBA with latest 2.0.x bugfix, but this was always the case since I added case light, like 6 months. I'm using Heater 2 output as caselight: pin 6.

`/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration_adv.h
 *
 * Advanced settings.
 * Only change these if you know exactly what you're doing.
 * Some of these settings can damage your printer if improperly set!
 *
 * Basic settings can be found in Configuration.h
 *
 */
#define CONFIGURATION_ADV_H_VERSION 020005

// @section temperature

//===========================================================================
//=============================Thermal Settings  ============================
//===========================================================================

//
// Custom Thermistor 1000 parameters
//
#if TEMP_SENSOR_0 == 1000
  #define HOTEND0_PULLUP_RESISTOR_OHMS 4700    // Pullup resistor
  #define HOTEND0_RESISTANCE_25C_OHMS  100000  // Resistance at 25C
  #define HOTEND0_BETA                 3950    // Beta value
#endif

#if TEMP_SENSOR_1 == 1000
  #define HOTEND1_PULLUP_RESISTOR_OHMS 4700    // Pullup resistor
  #define HOTEND1_RESISTANCE_25C_OHMS  100000  // Resistance at 25C
  #define HOTEND1_BETA                 3950    // Beta value
#endif

#if TEMP_SENSOR_2 == 1000
  #define HOTEND2_PULLUP_RESISTOR_OHMS 4700    // Pullup resistor
  #define HOTEND2_RESISTANCE_25C_OHMS  100000  // Resistance at 25C
  #define HOTEND2_BETA                 3950    // Beta value
#endif

#if TEMP_SENSOR_3 == 1000
  #define HOTEND3_PULLUP_RESISTOR_OHMS 4700    // Pullup resistor
  #define HOTEND3_RESISTANCE_25C_OHMS  100000  // Resistance at 25C
  #define HOTEND3_BETA                 3950    // Beta value
#endif

[... 3481 lines cut ...]
  #define LCD_PINS_RS                         19
  #define LCD_PINS_ENABLE                     42
  #define LCD_PINS_D4                         18
  #define LCD_PINS_D5                         38
  #define LCD_PINS_D6                         41
#endif

#define LCD_PINS_D7                           40

//
// Beeper, SD Card, Encoder
//
#define BEEPER_PIN                            44

#if ENABLED(SDSUPPORT)
  #define SDSS                                53
  #define SD_DETECT_PIN                       49
#endif

#if ENABLED(NEWPANEL)
  #define BTN_EN1                             11
  #define BTN_EN2                             12
  #define BTN_ENC                             43
#endif
`

randellhodges, 2020-05-07:
You have `//#define CASE_LIGHT_PIN 6 `. If you have the `//` in front of it, that comments it out so it isn't used. Remove the definition from your pins and then remove the `//` from it inside your Configuration_adv file and try it.

thinkyhead, 2020-05-07:
Fixed in 82bd19ce1e

closed 2020-05-07 by thinkyhead