MarlinFirmware/Marlin #17913: [FR] M851 for LCD_BED_LEVELING

wikolii, 2020-05-07:
### Description

So, as a personal preference i like LCD_BED_LEVELING. This process would benefit infinitely if the mbl.z_offset value was updated with the babystepping process done during the print.
I know this has been implemented for probe offsets (M851), but not for LCD_BED_LEVELING.

At the moment, one has to change this value in the tune menu - BED Z (mbl.z_offset) .

I'd like to ask for a process similar to M851 or even that same process could be extended to LCD_BED_LEVELING

### Feature Workflow

If this gets implemented, the babystepping value would be applied instantly. Just save EEPROM and we are golden for the following prints.

1. Using a printer with LCD_BED_LEVELING and mesh bed leveling.
2. Start a print.
3. Adjust the Z height via the Babystepping menu.
4. When done, the value generated gets added to BED Z (mbl.z_offset), similar to M851 (probe Z offset).
5. Save EEPROM.
6. Enjoy perfectly leveled future prints.

bill-orange, 2020-05-09:
I had not considered routinely doing the leveling from the front panel but what you describe makes sense.  I hope your suggestion is implemented.  A Z-offset test gcode could be more or less permanently left in the SD.

wikolii, 2020-05-17:
I thought this would gt more track since it's really annoying to adjust manually.
I guess automatic probes are more common now a days, but the are really expensive where i live. Upgrading all my machines would be a high budget 

I hope someone gets pumped about doing this 
Thanks!

sjasonsmith, 2020-09-11:
I think your request describes the `BABYSTEP_ZPROBE_OFFSET` feature.

Have you tried that? Is this request for something different?

wikolii, 2020-09-13:
> I think your request describes the `BABYSTEP_ZPROBE_OFFSET` feature.
> 
> 
> 
> Have you tried that? Is this request for something different?

`BABYSTEP_ZPROBE_OFFSET` requires a probe
This is for lcd assisted bed leveling. Using the nozzle and a piece of paper