MarlinFirmware/Marlin #17914: [BUG] TMC2208 UART pin strange signal levels and freeze pulling low

chunter1, 2020-05-07:
I connected the TMC2208 of the extruder on my Creality v2.4 board (CR10S Pro) to the following soft-UART pins:
RxD = 87
TxD = 59
Communication is successful, however the signal levels look very strange and when i print the linear advance test pattern, the TMC2208 immediately pulls the PDN_UART pin to low and outputs a view high-pulses while the motor is moving.
It does not get out of that low-stat anymore and communication is ofcourse impossible until a power-cycle is performed.
Can someone confom that the attached communication levels are normal and maybe also someone has an explanation to the LOW-stuck ouput of the TMC2208.
I tried variouse soft-serial pin combinations but problem stays the same.
I also checked if the LOW-level is coming from the TMC2208 pin by disconnecting everything else.

[image]

sjasonsmith, 2020-05-08:
What firmware are you running? This looks like an issue that was fixed about 6 months ago.

chunter1, 2020-05-08:
Do you remember what the bug was called or got a link?
I am running the "CrealityDwin2.0_Bleeding" branch of the Insanity Automation (Tiny Machines 3D) version on github.
https://github.com/InsanityAutomation/Marlin/tree/CrealityDwin2.0_Bleeding

sjasonsmith, 2020-05-08:
Here is one of the pull requests I made for it:
https://github.com/teemuatlut/TMCStepper/pull/98

There were several related issues for different platforms. I _believe_ that one applied to AVR controllers.

I don't really know the state of the firmware @InsanityAutomation maintains. It seems like it probably has a recent TMCStepper so this might be different than the issues you are seeing.

Do the signals look good if you capture the initial serial traffic after power-on? The problems I was seeing usually resulted in more problems with early communication, and it sounds like maybe yours is happening later.

chunter1, 2020-05-08:
**Problem solved!**
After doing some layout reverse engineering i found out that on the Creality V2.4 motherboard, the TMC2208 pins PDN_UART, INDEX and DIAG are all connected together and simply pulled up to +5V by a 100k resistor.
After doing some mod, i got a perfectly clean signal now and can go on trying to get linear advance to work.
[image]

sjasonsmith, 2020-05-08:
Thanks for the update @chunter1! That is a pretty annoying design decision on their part. Could you go ahead and close this issue, since it isn't anything Marlin related?

closed 2020-05-08 by chunter1