MarlinFirmware/Marlin #17921: [BUG] Hang after filament change

GMagician, 2020-05-08:
This evening I saw something weird... (I have installed a "few" day bugfix version)

After a 4hr of  printing, I selected from LCD the "filament change".
First time no reaction...I thought maybe I waited too much time and lcd went back to main page...then let's select again "filament change", now printer stopped and gone to park position. removed old filament and stopped again. I inserted new filament, pressed lcd button and new filament has been loaded, purged and......lcd menu on main page with "printer paused" message and no way to restart printer...nor with M108...

Now with a little bit of happyness I removed all print, restarted a new print with no filament, just to check what is happening and "beeeep" it works as expected. After purge it ask me to purge again or restart print....

I will retry these actions to see if it will do that again and I'll try to debug it....

GMagician, 2020-05-08:
ok...now two problems with stopping.....While I was trying to repeat above issue I selected "stop print"...printer stopped moved X & Y to 0 (homing?!?!?) and then print restarted automatically with no more possibility to stop it (no lcd item)

thinkyhead, 2020-05-10:
Hmm… Without a host to handle the host action command, yes indeed it is not clear what to do, except beep at you and keep requesting an `M108` in the serial messages. At present it does host keepalive with "waiting for user" messages, but it might not mention `M108` — and that requires Emergency Parser….

Oh well, hopefully most host software will catch on.

GMagician, 2020-05-10:
> Oh well, hopefully most host software will catch on.

But repetier

About filament change...now it works as expected... really don't able to replicate.
Really weird thing I remember, lcd was in main page with a message like printer paused or something like that....can't find in code an exit point that may lead to this...I have some check points out in serial to see what happens..Sure I can't risk to lose printed part so test may be a long trip

thinkyhead, 2020-05-10:
Hmm… Maybe it can be simulated. I have been meaning to add "fake heating" and things like that for the `MARLIN_DEV_MODE`, which would come in handy to pretend to be sitting in filament runout for hours but totally safe with no actual heating.

GMagician, 2020-05-10:
> I have been meaning to add "fake heating"

I made this by lowering min temp and removing cold extrusion. Now bed is off and hotend is just 50° (> min temp * 2 otherwise printer stays in "stopped" mode).
But it works and continue to works as expected. I hope is not some sort of memory corruption (I'm using ramps and mega) that occurs after a lot of "usb" frame have been sent

Rubenazzzo, 2020-05-10:
Hi.
I have the same problem, now I am using a skr v1.3 and when I give it a change of filament in the menu, it freezes me completely, and on top of the piece, I have configured the NOZZLE_PARK_FEATURE and the change in ADVANCED_PAUSE_FEATURE.

[configuracion.zip](https://github.com/MarlinFirmware/Marlin/files/4605783/configuracion.zip)

GMagician, 2020-05-10:
@Rubenazzzo that's seems not the same. In my case I did a correct unload, load & purge and was stopped in main page (I was able to navigate into it) with messages on the message bar (paused or some other message)

GMagician, 2020-05-23:
I'm not able to replicate it anymore...I' close it

closed 2020-05-23 by GMagician