MarlinFirmware/Marlin #17922: [BUG] The M115 command does not work

radek8, 2020-05-08:
The M115 command does not work in the current version of Marlin bugfix 2.0

thisiskeithb, 2020-05-09:
Please follow the [Issue Template](https://raw.githubusercontent.com/MarlinFirmware/Marlin/bugfix-2.0.x/.github/issue_template.md) when filing a bug report so we don't have to hunt for information.

```md
### Description

<!-- Description of the bug or requested feature -->

### Steps to Reproduce

<!-- If this is a Bug Report, please describe the steps needed to reproduce the issue -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
```

radek8, 2020-05-09:
**Expected behavior:**

[image]

**Actual behavior:**

[image]

sjasonsmith, 2020-05-09:
Post your configs.

thisiskeithb, 2020-05-09:
*Please* follow the Issue Template.

radek8, 2020-05-09:
> Post your configs.

no need to send configuration
by commenting #define M115_GEOMETRY_REPORT
output M115 is OK
@guruathwal should check if the settings work as intended.

sjasonsmith, 2020-05-09:
Every issue should have configs posted. Not posting configs requires someone to waste their time creating their own configs which may or may not exhibit the same issue.

If you won’t spend the time to post your configs, why should someone else spend their time trying to help?

radek8, 2020-05-09:
Configuration

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4604566/config.zip)

sjasonsmith, 2020-05-09:
OK, I've reproduced it. That is some pretty hilarious output. Far more entertaining than I expected.

sjasonsmith, 2020-05-10:
There was an infinite loop in some recently added print macros. I've posted a PR to fix it.
@radek8 if you need to use this feature you can look at the PR linked above and easily implement the change yourself.

sjasonsmith, 2020-05-10:
@radek8 the fix for this has been merged into bugfix-2.0.x. Please try it out and close the issue once you verify it is fixed.

closed 2020-05-10 by thinkyhead