MarlinFirmware/Marlin #17923: skr 1.3 chamber temp, heating.

Bulynia, 2020-05-08:
hello
couple week stock on it please help.
**Configuration.h**
#define TEMP_SENSOR_CHAMBER 1 (try 999 set to 60c)

**Configuration_adv.h**
#if TEMP_SENSOR_CHAMBER
  #define CHAMBER_MINTEMP             0
  #define CHAMBER_MAXTEMP            60
  #define TEMP_CHAMBER_HYSTERESIS     1   // (°C) Temperature proximity considered "close enough" to the target
  //#define CHAMBER_LIMIT_SWITCHING
  #define HEATER_CHAMBER_PIN       P1_26   // Chamber heater on/off pin
  #define HEATER_CHAMBER_INVERTING FALSE
#endif

#if ENABLED(THERMAL_PROTECTION_CHAMBER)
  #define THERMAL_PROTECTION_CHAMBER_PERIOD    20 // Seconds
  #define THERMAL_PROTECTION_CHAMBER_HYSTERESIS 2 // Degrees Celsius

  /**
   * Heated chamber watch settings (M141/M191).
   */
  #define WATCH_CHAMBER_TEMP_PERIOD            60 // Seconds
  #define WATCH_CHAMBER_TEMP_INCREASE           2 // Degrees Celsius
#endif

**pins_BTT_SKR_common.h**
after:
#ifndef TEMP_BED_PIN
  #define TEMP_BED_PIN                  P0_23_A0  // A0 (T0) - (67) - TEMP_BED_PIN
#endif
add:
#ifndef TEMP_CHAMBER_PIN                          
  #define TEMP_CHAMBER_PIN              P1_24 
#endif

after update always got: 
Err: MAXTEMP: Enclosun
PRINTER HALTED
please reset.

using skr 1.3v
used pin
2 extruders P2_04, P2_07.
1 head bed P2_05
1 chamber heater P1_26
1 chamber temp P1_24 (try Thermistor 100k Ohm B 3950, NTC thermistor 10k, DHT21)
neopixel P2_00
inside led P1_28

please help to set it

ellensp, 2020-05-09:
Please read https://github.com/MarlinFirmware/Marlin/issues/14694 on asking setup questions.

mlapaglia, 2020-05-10:
Isn't pin `1.24` used for the neopixels? Did you mean to use `0.24` for the thermistor?
[https://reprap.org/forum/file.php?1,file=111776](url)

Bulynia, 2020-05-10:
Yep sorry it pin:
P1_24, P1_26..... axis max pin's.

Bulynia, 2020-05-11:
P2_00 I using for neopixel

closed 2020-06-20 by boelle