MarlinFirmware/Marlin #17925: Unable to compile marlin 2 for SKR v1.3 for Ender 3 pro

Ashmore61, 2020-05-09:
<I have tried compiling Marlin 2 with Platformini in Atom, and get the following output.I have no idea how to fix it. HELP!!

> Executing task: C:\Users\Ashley\.platformio\penv\Scripts\platformio.exe run <

Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.2.zip; framework: arduino; board: nxp_lpc1768)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x 0.1.1 > NXP LPC1768
HARDWARE: LPC1768 100MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES:
 - framework-arduino-lpc176x 0.2.3
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
LibraryManager: Installing TMCStepper
Using cache: C:\Users\Ashley\.platformio\.cache\96\a1c6025a12653c867c10f8556d8ee496
Unpacking...
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\Ashley\\Documents\\Arduino\\BIGTREETECH-SKR-V1.3-master\\BIGTREETECH-SKR-V1.3-master\\BTT SKR V1.3\\firmware\\Marlin-bugfix-2.0.x\\.pio\\libdeps\\LPC1768\\_tmp_installing-xyam5ifi-package\\TMCStepper-pr_lpctimingfix\\examples\\Calibrate_spreadCycle\\Calibrate_spreadCycle.ino':
  File "C:\Users\Ashley\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 156:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Ashley\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Ashley\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Ashley\.platformio\platforms\nxplpc-arduino-lpc176x@src-de6b279104eee6e886c6740cfbf2debe\builder\main.py", line 148:
    target_elf = env.BuildProgram()
  File "C:\Users\Ashley\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Ashley\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\Ashley\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Ashley\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 131:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\Ashley\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Ashley\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1036:
    project.install_dependencies()
  File "C:\Users\Ashley\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 868:
    lm.install(uri)
  File "c:\users\ashley\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 281:
    force=force,
  File "c:\users\ashley\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 698:
    pkg_dir = self._install_from_url(name, url, requirements, track=True)
  File "c:\users\ashley\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 482:
    self.unpack(dlpath, tmp_dir)
  File "c:\users\ashley\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 222:
    return fu.unpack(dest_dir, with_progress=False)
  File "c:\users\ashley\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 149:
    self._unpacker.extract_item(item, dest_dir)
  File "c:\users\ashley\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 39:
    self._afo.extract(item, dest_dir)
  File "C:\Users\Ashley\.platformio\python37\lib\zipfile.py", line 1602:
    return self._extract_member(member, path, pwd)
  File "C:\Users\Ashley\.platformio\python37\lib\zipfile.py", line 1673:
    open(targetpath, "wb") as target:
================================================================================================== [FAILED] Took 1.30 seconds ================================================================================================== 

Environment                     Status    Duration
------------------------------  --------  ------------
megaatmega2560                  IGNORED
megaatmega1280                  IGNORED
at90usb1286_cdc                 IGNORED
at90usb1286_dfu                 IGNORED
DUE                             IGNORED
DUE_USB                         IGNORED
DUE_debug                       IGNORED
LPC1768                         FAILED    00:00:01.299
LPC1769                         IGNORED
sanguino_atmega644p             IGNORED
sanguino_atmega1284p            IGNORED
melzi                           IGNORED
melzi_optiboot                  IGNORED
rambo                           IGNORED
STM32F103RE                     IGNORED
STM32F103RC_fysetc              IGNORED
STM32F103RC_bigtree             IGNORED
STM32F103RC_bigtree_NOUSB       IGNORED
STM32F103RC_bigtree_512K        IGNORED
STM32F103RC_bigtree_512K_NOUSB  IGNORED
STM32F103RE_bigtree             IGNORED
STM32F103RE_bigtree_NOUSB       IGNORED
STM32F4                         IGNORED
STM32F7                         IGNORED
ARMED                           IGNORED
STM32F103VE_longer              IGNORED
mks_robin                       IGNORED
mks_robin_lite                  IGNORED
mks_robin_mini                  IGNORED
mks_robin_nano                  IGNORED
jgaurora_a5s_a1                 IGNORED
STM32F407VE_black               IGNORED
BIGTREE_SKR_PRO                 IGNORED
BIGTREE_BTT002                  IGNORED
teensy31                        IGNORED
STM32F103CB_malyan              IGNORED
chitu_f103                      IGNORED
teensy35                        IGNORED
esp32                           IGNORED
fysetc_f6_13                    IGNORED
linux_native                    IGNORED
SAMD51_grandcentral_m4          IGNORED
include_tree                    IGNORED
============================================================================================= 1 failed, 0 succeeded in 00:00:01.299 =============================================================================================
The terminal process terminated with exit code: 1

randellhodges, 2020-05-09:
You have a few issues:

1) Support questions should be done thru Discord, not thru github issues. See https://github.com/MarlinFirmware/Marlin/issues/14694
2) It looks like you are compiling the BTT fork. You'd need to ask them. If you compile *this* source, the Discord server would be a good place to ask.
3) You are using Atom and support for that is pretty light. You'd have better luck/support with Visual Studio Code and Platformio
4) I don't know about Atom, but some editors had problems with super long paths. Try moving your source closer to the root of your drive. I don't know if Atom has this problem or not.

sjasonsmith, 2020-05-09:
> 4\. I don't know about Atom, but some editors had problems with super long paths. Try moving your source closer to the root of your drive. I don't know if Atom has this problem or not.

The file the error references has a path length of 282 characters, so that is probably the issue.

Ashmore61, 2020-05-09:
Thanks for the help, moved files and now it works.

closed 2020-05-09 by Ashmore61