MarlinFirmware/Marlin #17926: marlin 1.1.9 heats but no print

djbilbao, 2020-05-09:
I recently installed marlin 1.1.9 on my anet a2 plus I launch a printing by usb I connect it heater bed and extruder but it does not launch any printing, I gave the firmware skynet3d 2.3.2 and it works, I even tried marlin 1.1.x bugfix nothing

GMagician, 2020-05-09:
Better to use 2.0 version. 1.1.9 will, maybe, receive some fix but no more development

djbilbao, 2020-05-09:
okay but when we download marlin 2.0 or + we have no marlin file for arduino no matter the link on the internet? thank you

djbilbao, 2020-05-09:
https://github.com/MarlinFirmware/Configurations/tree/release-2.0.5

GMagician, 2020-05-09:
Not sure I understand you but last release, 2.0.5.3 at https://github.com/MarlinFirmware/Marlin/releases, if you download source and check Marlin subfolder there is a Marlin.ino file

closed 2020-05-10 by thinkyhead