MarlinFirmware/Marlin #17931: Probing Z while Y at max value makes both XY motors move the carriage slightly to the right on a CoreXY machine

delgueda, 2020-05-09:
I have UBL activated and tuned in a custom CoreXY 3d printer (hypercube). My mainboard is a RE-ARM/RAMPS 1.4 and I'm using marlin version 2.0.5.3. My Z probe is a precision piezo disk attached to my X carriage.
Whenever there's need of probing the bed at the Ymax bed edge (for example during G29 J 3-point tilt correction) I noticed the X carriage moves slightly to the right. In order to check if this was a problem with my printer build or kinematics, I loosened both XY belts from the carriage and noticed both motors (XY) move counter clockwise when probing the bed.
I can reproduce the issue by doing G28, G1 X100 Y196 (196 is my bed Y max) and then G30 or M48. I haven't noticed any influence of the X position to reproduce the bug.
[conf files.zip](https://github.com/MarlinFirmware/Marlin/files/4604447/conf.files.zip)

boelle, 2020-06-20:
@delgueda is this still an issue?

delgueda, 2020-06-21:
Yes, I just checked again and it's still an issue. This time the X carriage moved left and right in small distances with each probe. I'm still using version 2.0.5.3

boelle, 2020-06-23:
Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

delgueda, 2020-06-25:
I just installed the latest bugfix-2.0.x and tested again. The problem still exists, no change at all.

sjasonsmith, 2020-06-25:
@delgueda, would you mind updating your description to follow the formatting in the issue template? Consistent description formatting makes it much easier for people to quickly understand the issue and how to replicate it.

If you need to see the issue template, just click the buttons to report a new issue, but please don't actually submit it!

delgueda, 2020-06-26:
I edited the report to follow the formatting. I also made a video showing the problem and attached my conf files for the bugfix version. Sorry for not doing it before, it's the first time I ever file a bug report.

boelle, 2020-07-02:
@sjasonsmith did the reformat of the first post cast any light on this one?

sjasonsmith, 2020-07-02:
@delagela I just realized that you have skew correction enabled. This is probably the cause of the unexpected motor movement.

To test that, disable `SKEW_CORRECTION` (or just `SKEW_CORRECTION_FOR_Z`) and see if the behavior goes away.

delgueda, 2020-07-03:
Yes! Disabling `SKEW_CORRECTION_FOR_Z` made the printer behave as expected.
So it looks like this is a bug in the implementation of skew correction for Z.

sjasonsmith, 2020-07-03:
Why would it be a bug? It looks like your configs have values in them that would cause skew correction to occur on Z moves. It is probably doing what it is supposed to according to the numbers entered.

delgueda, 2020-07-03:
> Why would it be a bug? It looks like your configs have values in them that would cause skew correction to occur on Z moves. It is probably doing what it is supposed to according to the numbers entered.

It only happens at the maximum value of Y (196), if I move the toolhead to 195.9 it doesn't do it. Because of that it doesn't seem to be correcting skew. Also the movements in X add up with each probe instead of returning to the starting point. If it was correcting skew, I would imagine that after probing and raising Z the toolhead would return to where it was before probing.

delgueda, 2020-07-04:
Now I noticed that if I leave XZ as YZ skew compensation but set XY compensation to 0 (M852 I0) th printer doesn' t move the toolhead when doing z probing with M48. I hope this is useful.

boelle, 2020-07-04:
i think this is a case where it works as it should but the user expect something entirely different

and its also a case where functions are used to make up for a bad assembled machine, why do you need skew correction if your machine is assembled correctly?

delgueda, 2020-07-04:
OK I give up. I just wanted to help. Good luck.

sjasonsmith, 2020-07-04:
@delgueda I think you actually do have an issue worth looking into, especially the bit where moving the carriage 0.1mm prevents it from happening. I had added labels after your earlier comment, but didn’t reply because I don’t know when anybody will have a chance to look closer at it.

I’m reopening this, as somebody should try to reproduce this to understand why it is happening. I won’t be surprised if this isn’t even CoreXY related.

Perhaps this is a config issue, but if that is the case I would like somebody more experienced with skew correction to explain _why_ the behavior you described two comments ago makes sense.

boelle, 2020-07-04:
i have the same board and printer style, but a bl touch probe, i cant reproduce it

only difference other than the probe is that i dont use skew correction (i have no need for it)

my config is here: https://github.com/boelle/Marlin/tree/Core-XY

i will try again tomorrow just to be sure

sjasonsmith, 2020-07-04:
@boelle if you want to reproduce this you will need to enable the skew correction features like the posted configs. There no point in testing if you don’t.

boelle, 2020-07-04:
of course, but its also nice to know what makes it do what is descriped

so i will test with and without skew enabled

boelle, 2020-07-04:
just for the fun of it

```
//#define SKEW_CORRECTION

#if ENABLED(SKEW_CORRECTION)
  // Input all length measurements here:
  #define XY_DIAG_AC 282.8427124746
  #define XY_DIAG_BD 282.8427124746
  #define XY_SIDE_AD 200

  // Or, set the default skew factors directly here
  // to override the above measurements:
  #define XY_SKEW_FACTOR 0.0

  //#define SKEW_CORRECTION_FOR_Z
  #if ENABLED(SKEW_CORRECTION_FOR_Z)
    #define XZ_DIAG_AC 282.8427124746
    #define XZ_DIAG_BD 282.8427124746
    #define YZ_DIAG_AC 282.8427124746
    #define YZ_DIAG_BD 282.8427124746
    #define YZ_SIDE_AD 200
    #define XZ_SKEW_FACTOR 0.0
    #define YZ_SKEW_FACTOR 0.0
  #endif

  // Enable this option for M852 to set skew at runtime
  //#define SKEW_CORRECTION_GCODE
```

those are the default values, will enabling it  actually do any correction?

why i ask is that i still need a silicone heater for my bed and those take a long time to show up from china
and then a flex plate system, both are more dificult as my bed is 500x500

boelle, 2020-07-07:
i wanted to test this one but sadly i cant

i use a bl touch as probe and not the nozzle and i need to use #define PROBING_MARGIN 30 to stay clear of the ends on my axis
i started out with 50 but could not get it lower than 30

in effect i cant do a G30 at the max of Y as it will not allow a probe if closer than 30 to end of axis

boelle, 2020-07-07:
not sure but a few differences i noted in the config compared to mine

back... just a list of the difference between my config and his:

me classic jerk
you #define NOZZLE_AS_PROBE
me #define BLTOUCH
you #define AUTO_BED_LEVELING_UBL
me #define AUTO_BED_LEVELING_BILINEAR

another odd one is that you do not have ro in the LCD language strings so the config is not the newest

in the adv config file you use babystepping and i dont

not sure if that matters but it was the things that sticks out to me

boelle, 2020-07-08:
> 1 - G28
> 2 - G1 X100 Y196 (X value irrelevant, Y max in my printer is 196)
> 3 - M48 or G30

works fine for me, no XY movement 
as i use BL touch and i have a probe XY offset i have to back of the XY max or else it will not allow me to probe

> Problem also arises when populating the mesh using MESH_INSET = 0
> 1 - G28
> 2 - Set bed temp
> 3 - G29 P1

doing G29 P1 starts a normal probe just like G29 would

@delgueda how can you use nozzle as probe? please explain how your probe is setup

boelle, 2020-07-08:
my config: https://github.com/boelle/Marlin/tree/Core-XY

but i made these changes

[image]

delgueda, 2020-07-08:
> @delgueda how can you use nozzle as probe? please explain how your probe is setup

That information is my first message. I use a piezo disk as Z probe sensor.
That allows me to use the tip of the nozzle to probe the bed. I don't think you'll be able to reproduce the condition (prove at Y MAX) without using the nozzle as probe and margin = 0.

boelle, 2020-07-08:
i do have margin set to 0
but its the probe i have that prevents me from probing at the edge

i do have a 500x500 bed but to try and reproduce i faked a smaller bed so at least i could get to the edges with the nozzle

since i have plenty of space i could "force" it to probe anyway
of course that is dangerous if you dont know what you are doing, i just need to know how to trick it to probe regardless

delgueda, 2020-07-08:
I see you enabled skew correction, but did you enable `SKEW_CORRECTION_FOR_Z`?

If I enable `SKEW_CORRECTION`, but leave `SKEW_CORRECTION_FOR_Z` the XY motors don't move abnormally.

boelle, 2020-07-08:
my plan was to test in steps

sometimes if you enable all at the same time you dont know what settings caused it, so i will enable for Z and retest later

does the values matter? or is it enough just to enable with default values?

delgueda, 2020-07-08:
> does the values matter? or is it enough just to enable with default values?

I have done small changes in the values and the problem persist. I haven't tried the default values. I'll try again later today, with other values and the defaults.

boelle, 2020-07-08:
no problemo, why i ask is that the printer is not completety build yet
silicone heater is not here before tomorrow and the flexplate is sold out so i can print but for testing it should be good enough

boelle, 2020-07-08:
just tested and still no XY move

config: https://github.com/boelle/Marlin/tree/Core-XY

with these changes as to fake your config
[image]

boelle, 2020-07-08:
no probe offset allows me to probe at the edges, since i have a 500x500 bed its safe to do so, but to others watching: Dont do this if you dont know what you are doing, if you break something from reading this you did not know what you did

the half bed size change is just something i picked, this allows me to probe to edge as you do

only left is to figure if the default number for skew correction matter or not

boelle, 2020-07-08:
i added your numbers for skew correction, still no XY move during G30

[image]
[image]

boelle, 2020-07-08:
just curious, how old is your copy of marlin and where did you get it from?
for where please give direct link

thinkyhead, 2020-07-18:
@delgueda — How's it going? I see this issue has been buried in the pile for a while. It's been a challenging couple of months…

My strategem would be to look at the points in Marlin where `skew` is applied, namely the `Planner::apply_modifiers` method, which is called from `Planner::buffer_line` and analyze what goes on and whether the weird behavior is peculiar to Core kinematics, or whether it could affect regular Cartesians. I'd add a bit of logging at the interesting points inside of `buffer_line` to see whether unusual limits are being applied, or if other factors may be at play.

Of course, to ensure that we are all on the same page, I would ask that you do all your testing with `bugfix-2.0.x`.

I'm about to crash for the night, but let's pick this up tomorrow (Saturday, CDT) and compare notes.

closed 2020-08-24 by github-actions[bot]