MarlinFirmware/Marlin #17940: [BUG] Display glitch in LCD PROGRESS BAR

Lord-Quake, 2020-05-10:
[image]

### Bug Description

Progress bar should be (normally was) completely filled.

Compiled with commit 2020.05.10 (https://github.com/MarlinFirmware/Marlin/commit/86c112538084f7718252966f5f9d9278c4837fb2)

ellensp, 2020-05-10:
Config files?, so we can try and replicate...

kmobs, 2020-05-11:
Same issue here. I've attached my config. Ender 3 pro as well. 

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/4607233/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/4607234/Configuration_adv.txt)

ellensp, 2020-05-11:
In Marlin/src/lcd/dogm/status_screen_DOGM.cpp is the following line
#define PROGRESS_BAR_Y (EXTRAS_BASELINE + 2)
change this to 
#define PROGRESS_BAR_Y (EXTRAS_BASELINE + 1)

Not sure if this is correct fix yet, but works..

Lord-Quake, 2020-05-11:
@ellensp I can confirm the correction fixed the issue.

closed 2020-05-11 by Lord-Quake

thinkyhead, 2020-05-12:
This should do the trick…
```cpp
    //
    // Progress bar frame
    //

    if (PAGE_CONTAINS(PROGRESS_BAR_Y, PROGRESS_BAR_Y + 3))
      u8g.drawFrame(PROGRESS_BAR_X, PROGRESS_BAR_Y, PROGRESS_BAR_WIDTH, 4);

    //
    // Progress bar solid part
    //

    if (PAGE_CONTAINS(PROGRESS_BAR_Y + 1, PROGRESS_BAR_Y + 2))
      u8g.drawBox(PROGRESS_BAR_X + 1, PROGRESS_BAR_Y + 1, progress_bar_solid_width, 2);

```

commit 7664e552 (2020-05-12): Finish DOGM progress bar

Fixes #17940