MarlinFirmware/Marlin #17941: [BUG] Dual X Carriage modes

DrumClock, 2020-05-10:
In AUTO_PARK mode, the carriages are exactly replaced according to the X2_MAX_POS setting (hotend offset X) and everything works in the entire range X = 0 to 240 mm.
 
[image]

In DUPLICATION mode, the X2 trolley hits the X + endstop (Of course, I try the movement in the range X= 0 to 115 mm)
In MIRRORED mode, the carriages collide in the middle (Of course I try the movement in the range  X= 0 to 115 mm)

[image]

In DUPLICATION and MIRRORED mode, therebetween carriages X1 and X2  is a gap of 198 mm .

I tried to change the DEFAULT_DUPLICATION_X_OFFSET setting.
But whatever I write there the gap is still 198 mm.

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/4605671/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/4605672/Configuration_adv.txt)

InsanityAutomation, 2020-05-15:
There are no provisions right now for capping the softendstops in duplication mode based on the offset applied, though its something thats been tossed around before. Itll probably get done at some point.

Roxy-3D, 2020-05-16:
@InsanityAutomation    What about the 198mm gap between the two extruders in duplication mode?   It would seem changing DEFAULT_DUPLICATION_X_OFFSET  should change that.   But he says it doesn't? 

I wonder if something got broke in G28.    Or...   Maybe something changed in M605?   Both of those used to unpark the 2nd extruder and put it in the right place, and then unpark the 1st extruder.

DrumClock, 2020-05-16:
The axial distance **between X1 and X2** carrier depends
on the setting of the value **HOTEND_OFFSET_X for T1**

space X1 and X2  =  HOTEND_OFFSET_X   T1
198 mm = 292 mm
121 mm = 370 mm
  50 mm = 440 mm

AUTO_PARK mode must be HOTEND_OFFSET_X for T1 = 292 mm (or X2_MAX_POS)
movement is OK in the whole range of HB (0-240 mm), replacement T0 / T1 accurate.

DUPLICATIONK mode must be HOTEND_OFFSET_X for T1 = 370 mm
movement is limited to 0-170 mm and is OK

MIRRORED mode must be HOTEND_OFFSET_X for T1 = 440 mm
movement is limited in the range 0-170 mm - **carrier X1 and X2 collide at a position of 105 mm**

I hope these measurements help solve the error with the trolley positions.

DrumClock, 2020-05-16:
Hi,
changing IDEX modes only works properly from the LCD.
When using the G-code command M605 S ... carriers paint into each other and do not park as they should.

wartness, 2020-06-12:
My problem is the same as above! At M605 S1 the two carriage took each other positions . When i send the code T0 to the printer, than the T1 goes to parking position and T0 took the actual position. This is working after start up. But when the Z axes goes to Home, the position exchange is not working anymore. During printing the T0 is working , when the T1 command is sent the T0 is stops and T1 is collide with it. ( T0 is not going to parking position. )

boelle, 2020-06-21:
@DrumClock still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

DrumClock, 2020-06-24:
I'll try it in the next few days.
I have to mechanically rebuild the test printer  :)

DrumClock, 2020-06-27:
FW Marlin:       Bugfix 2.0.x 
Downloaded:	27.6.2020 at 15:17 (Prague)
### **No changes the same behavior as described previously.**

Dual X Carriage modes:
1. AUTO_PARK    - OK
2. DUPLICATION  - BUG
3. MIRRORED       - BUG

DrumClock, 2020-07-02:
Due to minimal responses to solving this problem, I closed this issues. The bug is still at 1.7.2020 in Marlin FW.

DrumClock, 2020-07-03:
### I tried this setting:

[image]

```
 #define X1_MIN_POS  0   
 #define X1_MAX_POS  240          
 #define X2_MIN_POS  0             
 #define X2_MAX_POS  240 
 #define X2_HOME_DIR   1           
 #define X2_HOME_POS 292

```
**AUTO_PARK** mode on LCD
- the movement of the carriage X1 is limited between 0 and 240
- the movement of the carriage X2 is limited between 0 and **292**
   setting `#define X2_MAX_POS  `has **no effect** on carriage X2 movement - **always goes to X-max endstop**
- tool change T0 and T1 is exact according.  Offset T0 toT1 = 292  (` #define X2_HOME_POS ` )

**DUPLICATION** and **MIRRORED** modes show the same errors as described previously.

### Please write exactly what I should change and test.

DrumClock, 2020-07-10:
Hi, @thinkyhead  
please write what I should change and test.
Thanks you very much.

thinkyhead, 2020-07-14:
Please don't @ me. I eventually find issues, but there is always a large pile.

thinkyhead, 2020-07-14:
The motion bugs resulting from calls to `M605` need to be looked into.

For testing turn on `DEBUG_DXC_MODE` and if include a `W` flag in your `M605` commands. With the `W` flag, Marlin will print verbose output to the host console. Try various values with `M605 S` and note the differences.

In the meantime I will look at what `M605` is trying to do, and perhaps there is a simple inverted logic bug that can be fixed quickly.

DrumClock, 2020-07-15:
> For testing turn on `DEBUG_DXC_MODE` and include a `W` flag in your `M605` commands.

How do I activate `DEBUG_DXC_MODE` ?

thinkyhead, 2020-07-15:
> How do I activate DEBUG_DXC_MODE ?

Add this to your config:
```cpp
#define DEBUG_DXC_MODE
```

Looks to me as though the bugs are located in `tool_change` and/or `dualx_tool_change`. So, get the latest code and also enable `DEBUG_TOOL_CHANGE` before testing.

DrumClock, 2020-07-15:
Ok, I set in **configuration.h**
```
#define DEBUG_DXC_MODE
#define DEBUG_TOOL_CHANGE
```
In **pronterface**, all you have to do is send it via the `M605 W S2 Xnn `
terminal and I will copy the statement to you.

Is that right ?

DrumClock, 2020-07-16:
**M605** via "pronterface" terminal does not work ....

```
External Reset
Marlin bugfix-2.0.x
echo: Last Updated: 2020-07-16 | Author: (Petr Forejt)
echo: Compiled: Jul 16 2020
echo: Free Memory: 2439 PlannerBufferBytes: 1280
Error: EEPROM CRC mismatch - (stored) 5944! = 10202 (calculated)!
echo: Hardcoded Default Settings Loaded
echo: SD card ok
Printer is now online.
>>> g28
SENDING: G28
echo: Active Extruder: 0
echo: busy: processing
echo: Active Extruder: 0
>>> t1
SENDING: T1
echo: busy: processing
echo: Active Extruder: 1
>>> t0
SENDING: T0
echo: busy: processing
echo: Active Extruder: 0
>>> M605 S2
SENDING: M605 S2
>>> t1
SENDING: T1
echo: T1 Invalid extruder
>>> t0
SENDING: T0
echo: Active Extruder: 0
>>> M605
SENDING:M605
>>> M605 S3
SENDING:M605 S3
Printer must be in DXC_DUPLICATION_MODE prior to
specifying DXC_MIRRORED_MODE.
>>> M605 S2
SENDING:M605 S2
>>> M605 S1
SENDING:M605 S1
>>> t0
SENDING:T0
echo:Active Extruder: 0
>>> t1
SENDING:T1
echo:busy: processing
echo:Active Extruder: 1
>>> t0
SENDING:T0
echo:busy: processing
echo:Active Extruder: 0

```

DrumClock, 2020-07-16:
**Still the same** .... the M605 command via the pronterface terminal does not work.
DUPLICATION and MIRRORED modes via LCD ... gap between carriages still 198 mm.

Extract from pronterface terminal:
[DXC_pronterface.zip](https://github.com/MarlinFirmware/Marlin/files/4930945/DXC_pronterface.zip)

DrumClock, 2020-08-03:
Hi @thinkyhead 
Does anyone solve this problem? 
How do I know, please?
I would very much like to provide more information.

Thanks

boelle, 2020-08-03:
> How do I know, please?

Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

DrumClock, 2020-08-04:
Hi @boelle 
### Nothing has changed since I made a mistake on May 10.
**In the DUPLICATION and MIRRORED modes, there is still a gap of 198 mm between the cars.**

DrumClock, 2020-08-19:
### Bug still not fixed.

boelle, 2020-08-19:
> Bug still not fixed.

Marlin is free, you dont pay for it

Is it fair to demand anything?

Things take time arround here, if you cant live with that there are other firmwares

DrumClock, 2020-08-25:
Hi @boelle 
YES Marlin is free but is starting to have more and more bugs that were not there before.
I think I tested and provided the information you requested, but I have no feedback, is it being worked on?
I'm sorry the most, I'm trying to help but I have the answer .....

boelle, 2020-08-25:
the maintainer of the project @thinkyhead is the man

DrumClock, 2020-08-28:
@thinkyhead
Bugfix 2.0.x - DISTRIBUTION DATE "2020-08-27"

Bug still not fixed.

DrumClock, 2020-09-28:
Hi,
Is anyone solving this question or should I close it?
Looks like there's no interest in it and nothing's happening.

thisiskeithb, 2020-09-28:
Keep in mind that all the code here is contributed by unpaid volunteers who work on it in their free time.

I've added a Help Wanted label to hopefully attract attention of someone who has the hardware, interest, and ability to fix/implement the change.

BAUBLITZ, 2020-10-11:
I have a Tenlog D3S and it works fine with 2.0.6.1, only the Y-Layershift issue i have to.
You must set to 
 #define X1_MIN_POS  -47   
 #define X1_MAX_POS  240          
 #define X2_MIN_POS  0             
 #define X2_MAX_POS  292 
 #define X2_HOME_DIR   1           
 #define X2_HOME_POS 292

 #define DEFAULT_DUAL_X_CARRIAGE_MODE DXC_AUTO_PARK_MODE
 #define DEFAULT_DUPLICATION_X_OFFSET 120

ogland, 2020-10-13:
I can chime in on that this doesn't seem to work. Nothing really happens when I change DEFAULT_DUPLICATION_X_OFFSET

DrumClock, 2020-11-13:
I'm offering a $ 20 reward for quickly fixing this bug
within the next 7 days.

BUG:  
In the DUPLICATION and MIRRORED modes, there is still a gap of 198 mm between the carriages.

ogland, 2020-11-13:
I found that setting the mode in starting gcode in your slicer lets you have manual control over spacing using a simple printer profile.

DrumClock, 2020-11-13:
HI, @ogland 
I don't want to solve BUG at the slicer level but integrate it into FW Marlin for everyone.
So fix a bug in Marlin ......

InsanityAutomation, 2020-11-13:
Just to relay where I got to when I looked earlier for anyone with time to dig in first, the menu items set a fixed position as an argument to M605S# for mirror and duplication modes. This should be dropped from the lcd command, then testing performed to make sure the default applies correctly. Originally this was added to the menu before code in M605 that set based on the default however it may not be required anymore.

Need to verify M605W output to ensure that duplicate_extruder_x_offset is correct with the config_adv default, then check new_pos.x assignment in motion.cpp. If so, menu entry can be edited in menu_configuration.cpp to remove X200 from the hardcoded commands.

DrumClock, 2020-11-13:
Hi @InsanityAutomation 
This is an extract from pronterface M605W:

```
>>> g28
SENDING: G28
echo: Active Extruder: 0
echo: busy: processing
echo: busy: processing
echo: Active Extruder: 0
>>> t1
SENDING: T1
echo: busy: processing
echo: Active Extruder: 1
>>> t0
SENDING: T0
echo: busy: processing
echo: Active Extruder: 0
>>> M605 S2
SENDING: M605 S2
>>> t0
SENDING: T0
echo: Active Extruder: 0
>>> t1
SENDING: T1
echo: T1 Invalid extruder
>>> M605 W
SENDING: M605 W
>>> M605 S2 X120
SENDING: M605 S2 X120
>>> t1
SENDING: T1
echo: T1 Invalid extruder
>>> t0
SENDING: T0
echo: Active Extruder: 0
>>> t1
SENDING: T1
echo: T1 Invalid extruder
>>> M605 S1
SENDING: M605 S1
>>> T1
SENDING: T1
echo: busy: processing
echo: Active Extruder: 1
>>> T0
SENDING: T0
echo: busy: processing
echo: Active Extruder: 0

```

Duplication mode does not work with M605 S2 X120
When used with an LCD, the gap is still 198 mm

DrumClock, 2020-11-13:
AUTO PARK mode 

```
>>> g28
SENDING:G28
echo:Active Extruder: 0
echo:busy: processing
echo:Active Extruder: 0
>>> M605W
SENDING:M605W
echo:Dual X Carriage Mode AUTO_PARK
Active Ext: 0 NOT  parked.
active_extruder_x_pos: 120.00
inactive_extruder_x: 300.00
extruder_duplication_enabled: 0
duplicate_extruder_x_offset: 120.00
duplicate_extruder_temp_offset: 0
delayed_move_time: 0
X1 Home X: -50.00
X1_MIN_POS=0
X1_MAX_POS=240
X2 Home X: 300.00
X2_MIN_POS=0
X2_MAX_POS=240
X2_HOME_DIR=1
X2_HOME_POS=300
DEFAULT_DUAL_X_CARRIAGE_MODE=DXC_AUTO_PARK_MODE	oolchange_settings.z_raise=2.00
DEFAULT_DUPLICATION_X_OFFSET=120
 T0  hotend_offset[0].120=0.00  hotend_offset[0].121=0.00  hotend_offset[0].122=0.00
 T1  hotend_offset[1].120=300.00  hotend_offset[1].121=0.00  hotend_offset[1].122=0.00
>>> t1
SENDING:T1
echo:busy: processing
echo:Active Extruder: 1
>>> M605W
SENDING:M605W
echo:Dual X Carriage Mode AUTO_PARK
Active Ext: 1 NOT  parked.
active_extruder_x_pos: 120.00
inactive_extruder_x: 300.00
extruder_duplication_enabled: 0
duplicate_extruder_x_offset: 120.00
duplicate_extruder_temp_offset: 0
delayed_move_time: 0
X1 Home X: -50.00
X1_MIN_POS=0
X1_MAX_POS=240
X2 Home X: 300.00
X2_MIN_POS=0
X2_MAX_POS=240
X2_HOME_DIR=1
X2_HOME_POS=300
DEFAULT_DUAL_X_CARRIAGE_MODE=DXC_AUTO_PARK_MODE	oolchange_settings.z_raise=2.00
DEFAULT_DUPLICATION_X_OFFSET=120
 T0  hotend_offset[0].120=0.00  hotend_offset[0].121=0.00  hotend_offset[0].122=0.00
 T1  hotend_offset[1].120=300.00  hotend_offset[1].121=0.00  hotend_offset[1].122=0.00

```

DrumClock, 2020-11-13:
DUPLICATION mode   (X2 sill parked on 300)

```
>>> g28
SENDING:G28
echo:Active Extruder: 0
echo:Active Extruder: 0
>>> M605 S2
SENDING:M605 S2
>>> M605W
SENDING:M605W
echo:Dual X Carriage Mode DUPLICATION
Active Ext: 0 NOT  parked.
active_extruder_x_pos: 120.00
inactive_extruder_x: 300.00
extruder_duplication_enabled: 0
duplicate_extruder_x_offset: 120.00
duplicate_extruder_temp_offset: 0
delayed_move_time: 0
X1 Home X: -50.00
X1_MIN_POS=0
X1_MAX_POS=240
X2 Home X: 300.00
X2_MIN_POS=0
X2_MAX_POS=240
X2_HOME_DIR=1
X2_HOME_POS=300
DEFAULT_DUAL_X_CARRIAGE_MODE=DXC_AUTO_PARK_MODE	oolchange_settings.z_raise=2.00
DEFAULT_DUPLICATION_X_OFFSET=120
 T0  hotend_offset[0].120=0.00  hotend_offset[0].121=0.00  hotend_offset[0].122=0.00
 T1  hotend_offset[1].120=300.00  hotend_offset[1].121=0.00  hotend_offset[1].122=0.00
```

InsanityAutomation, 2020-11-13:
I didnt see a G28X after youre M605S2 command, that command is what applies the position.

DrumClock, 2020-11-13:
```
SENDING:M605S2
>>> G28X
SENDING:G28X
echo:Active Extruder: 0
echo:busy: processing
echo:busy: processing
echo:Active Extruder: 0
>>> M605W
SENDING:M605W
echo:Dual X Carriage Mode DUPLICATION
Active Ext: 0 parked.
active_extruder_x_pos: -50.00
inactive_extruder_x: 300.00
extruder_duplication_enabled: 0
duplicate_extruder_x_offset: 200.00
duplicate_extruder_temp_offset: 0
delayed_move_time: 0
X1 Home X: -50.00
X1_MIN_POS=0
X1_MAX_POS=240
X2 Home X: 300.00
X2_MIN_POS=0
X2_MAX_POS=240
X2_HOME_DIR=1
X2_HOME_POS=300
DEFAULT_DUAL_X_CARRIAGE_MODE=DXC_AUTO_PARK_MODE	oolchange_settings.z_raise=2.00
DEFAULT_DUPLICATION_X_OFFSET=120
 T0  hotend_offset[0].120=0.00  hotend_offset[0].121=0.00  hotend_offset[0].122=0.00
 T1  hotend_offset[1].120=300.00  hotend_offset[1].121=0.00  hotend_offset[1].122=0.00
```

DrumClock, 2020-11-13:
`duplicate_extruder_x_offset: 200.00` is always set

InsanityAutomation, 2020-11-13:
Can you power cycle the machine and run G28; M605S2; G28X; M605W? I believe the 200 is there from the menu still and would prefer to see the full scenario without forcing it with M605S2X120 to be sure its not hiding an issue. Then any G1 command, even G1X10Y10 or something just to make it unpark and move. I do see the value of 120 in some of the logs above.

Perhaps an option to automatically apply G28X following M605S2 would be beneficial as it reduces a step that may be missed often.

DrumClock, 2020-11-13:
even if I set `duplicate_extruder_x_offset` to 120 with M605 S2 X120.
When Dulicate mode is selected from the LCD, it is overwritten to 200 anyway

InsanityAutomation, 2020-11-13:
Right, the point is to ensure everything works when relying on the default value, never supplying a static value, to be sure its safe to remove the legacy hard coded number.

DrumClock, 2020-11-13:
```
SENDING:G28
echo:Active Extruder: 0
echo:busy: processing
echo:Active Extruder: 0
>>> M605S2
SENDING:M605S2
>>> G28X
SENDING:G28X
echo:Active Extruder: 0
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:Active Extruder: 0
>>> M605W
SENDING:M605W
echo:Dual X Carriage Mode DUPLICATION
Active Ext: 0 parked.
active_extruder_x_pos: -50.00
inactive_extruder_x: 300.00
extruder_duplication_enabled: 0
duplicate_extruder_x_offset: 120.00
duplicate_extruder_temp_offset: 0
delayed_move_time: 0
X1 Home X: -50.00
X1_MIN_POS=0
X1_MAX_POS=240
X2 Home X: 300.00
X2_MIN_POS=0
X2_MAX_POS=240
X2_HOME_DIR=1
X2_HOME_POS=300
DEFAULT_DUAL_X_CARRIAGE_MODE=DXC_AUTO_PARK_MODE	oolchange_settings.z_raise=2.00
DEFAULT_DUPLICATION_X_OFFSET=120
 T0  hotend_offset[0].120=0.00  hotend_offset[0].121=0.00  hotend_offset[0].122=0.00
 T1  hotend_offset[1].120=300.00  hotend_offset[1].121=0.00  hotend_offset[1].122=0.00
>>> G1X10Y10
SENDING:G1X10Y10
>>> M605W
SENDING:M605W
echo:Dual X Carriage Mode DUPLICATION
Active Ext: 0 NOT  parked.
active_extruder_x_pos: 10.00
inactive_extruder_x: 300.00
extruder_duplication_enabled: 1
duplicate_extruder_x_offset: 120.00
duplicate_extruder_temp_offset: 0
delayed_move_time: 0
X1 Home X: -50.00
X1_MIN_POS=0
X1_MAX_POS=240
X2 Home X: 300.00
X2_MIN_POS=0
X2_MAX_POS=240
X2_HOME_DIR=1
X2_HOME_POS=300
DEFAULT_DUAL_X_CARRIAGE_MODE=DXC_AUTO_PARK_MODE	oolchange_settings.z_raise=2.00
DEFAULT_DUPLICATION_X_OFFSET=120
 T0  hotend_offset[0].120=0.00  hotend_offset[0].121=0.00  hotend_offset[0].122=0.00
 T1  hotend_offset[1].120=300.00  hotend_offset[1].121=0.00  hotend_offset[1].122=0.00

```

InsanityAutomation, 2020-11-13:
It appears that functioned correctly, physically is it correct?

DrumClock, 2020-11-13:
after entering G1 X10 Y10 is X1 in position 10 and X2 in position 130

DrumClock, 2020-11-13:
LCD selection still does not work (gap 200)

InsanityAutomation, 2020-11-13:
Ok, so to sum up the needfull, legacy hardcoded value is safe to remove from the menu code and add config option to auto apply G28X at mode change so a commonly missed step is avoided.

InsanityAutomation, 2020-11-13:
Right, that value was hard coded to the menu when an X argument was required, no default at the time. So the steps above just proved beyond doubt it was safe to remove.

DrumClock, 2020-11-13:
If you use the `M605 S2` command sequence; `G28X` sets the duplication mode with a **gap of 120 mm**.
**The selection of the LCD assistance module is still from a gap of 200 mm.**

**The movement of the trolleys in duplicate mode is not limited to half Hotbed.**
Movement of X is possible over half and then X2 hits the endstop X-max

InsanityAutomation, 2020-11-13:
Movement not being clamped is still a separate item from the first we were just discussing. Thats going to need some setup with the carriage size to avoid crashes in mirror mode and will take more than a few minutes between other things at my day job.

ogland, 2020-11-13:
@DrumClock I think you also need to slice accordingly. Set bed size in the printer profile to half bed size minus half required offset and set your M605 offset to 0 and it should print fine. In mirror mode you can set a profile for exactly half bed width.

DrumClock, 2020-11-13:
I'm fully available, I have a test frame and I can send videos of how it behaves.

DrumClock, 2020-11-13:
@ogland  
Yes, I understand that the slicer must be only half the width. The point is that if a g-code that is larger than half of the HB is sent by mistake, a collision will occur ...... it will probably need to be treated in FW in both modes.

InsanityAutomation, 2020-11-13:
Yup, in my opinion this closes the bug report half of this moving the remaining to a feature request since that safety never existed in the first place however still just fine to all live here. I agree its good to have it just havnt had time to do it.

DrumClock, 2020-11-13:
@InsanityAutomation 
I tried changes 20133 to my configuration.
errors occurred during compilation.
```

Compiling .pio\build\BIGTREE_GTR_V1_0\src\src\lcd\menu\menu_media.cpp.o
In file included from Marlin\src\lcd\menu\menu_configuration.cpp:31:
Marlin\src\lcd\menu\menu_configuration.cpp: In function 'void menu_idex()':
Marlin\src\lcd\menu\menu_configuration.cpp:207:27: error: 'MSG_IDEX_DUPE_GAP' is not a member of 'Language_cz'
  207 |     EDIT_ITEM(float42_52, MSG_IDEX_DUPE_GAP, &duplicate_extruder_x_offset);
      |                           ^~~~~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu_item.h:256:24: note: in definition of macro '_MENU_INNER_P'
Compiling .pio\build\BIGTREE_GTR_V1_0\src\src\lcd\menu\menu_mixer.cpp.o
  256 |   PGM_P const plabel = PLABEL;                                 \
      |                        ^~~~~~
Marlin\src\lcd\menu\menu_item.h:354:57: note: in expansion of macro '_MENU_ITEM_P'
  354 | #define MENU_ITEM_P(TYPE, PLABEL, V...)                 _MENU_ITEM_P(TYPE, false, PLABEL, ##V)
      |                                                         ^~~~~~~~~~~~
Marlin\src\lcd\menu\menu_item.h:392:58: note: in expansion of macro 'MENU_ITEM_P'
  392 | #define EDIT_ITEM_P(TYPE, PLABEL, V...)                  MENU_ITEM_P(TYPE, PLABEL, ##V)
      |                                                          ^~~~~~~~~~~
Marlin\src\lcd\menu\menu_item.h:393:58: note: in expansion of macro 'EDIT_ITEM_P'
  393 | #define EDIT_ITEM(TYPE, LABEL, V...)                     EDIT_ITEM_P(TYPE, GET_TEXT(LABEL), ##V)
      |                                                          ^~~~~~~~~~~
Marlin\src\lcd\menu\menu_item.h:393:76: note: in expansion of macro 'GET_TEXT'
  393 | #define EDIT_ITEM(TYPE, LABEL, V...)                     EDIT_ITEM_P(TYPE, GET_TEXT(LABEL), ##V)
      |                                                                            ^~~~~~~~
Marlin\src\lcd\menu\menu_configuration.cpp:207:5: note: in expansion of macro 'EDIT_ITEM'
  207 |     EDIT_ITEM(float42_52, MSG_IDEX_DUPE_GAP, &duplicate_extruder_x_offset);
      |     ^~~~~~~~~
Marlin\src\lcd\menu\menu_item.h:259:40: error: no matching function for call to 'TMenuEditItem<MenuEditItemInfo_float42_52>::action(const char* const&, float*)'
  259 |     MenuItem_##TYPE::action(plabel, ##V);                      \
      |                                        ^
Marlin\src\lcd\menu\menu_item.h:270:5: note: in expansion of macro '_MENU_INNER_P'
  270 |     _MENU_INNER_P(TYPE, ##V);         \
      |     ^~~~~~~~~~~~~
Marlin\src\lcd\menu\menu_item.h:354:57: note: in expansion of macro '_MENU_ITEM_P'
  354 | #define MENU_ITEM_P(TYPE, PLABEL, V...)                 _MENU_ITEM_P(TYPE, false, PLABEL, ##V)
      |                                                         ^~~~~~~~~~~~
Marlin\src\lcd\menu\menu_item.h:392:58: note: in expansion of macro 'MENU_ITEM_P'
  392 | #define EDIT_ITEM_P(TYPE, PLABEL, V...)                  MENU_ITEM_P(TYPE, PLABEL, ##V)
      |                                                          ^~~~~~~~~~~
Marlin\src\lcd\menu\menu_item.h:393:58: note: in expansion of macro 'EDIT_ITEM_P'
  393 | #define EDIT_ITEM(TYPE, LABEL, V...)                     EDIT_ITEM_P(TYPE, GET_TEXT(LABEL), ##V)
      |                                                          ^~~~~~~~~~~
Marlin\src\lcd\menu\menu_configuration.cpp:207:5: note: in expansion of macro 'EDIT_ITEM'
  207 |     EDIT_ITEM(float42_52, MSG_IDEX_DUPE_GAP, &duplicate_extruder_x_offset);
      |     ^~~~~~~~~
Marlin\src\lcd\menu\menu_item.h:93:17: note: candidate: 'static void TMenuEditItem<NAME>::action(const char*, TMenuEditItem<NAME>::type_t*, TMenuEditItem<NAME>::type_t, TMenuEditItem<NAME>::type_t, screenFunc_t, bool) [with NAME = MenuEditItemInfo_float42_52; TMenuEditItem<NAME>::type_t = float; screenFunc_t = void (*)()]'        
   93 |     static void action(
      |                 ^~~~~~
Marlin\src\lcd\menu\menu_item.h:93:17: note:   candidate expects 6 arguments, 2 provided
*** [.pio\build\BIGTREE_GTR_V1_0\src\src\lcd\menu\menu_configuration.cpp.o] Error 1
==================================================================== [FAILED] Took 57.70 seconds ====================================================================

Environment       Status    Duration
----------------  --------  ------------
BIGTREE_GTR_V1_0  FAILED    00:00:57.704
=============================================================== 1 failed, 0 succeeded in 00:00:57.704 =============================================================== 
```

InsanityAutomation, 2020-11-13:
My bad, one file had the white dot of not saved... Pushed now

DrumClock, 2020-11-13:
still compilation errors. tomorrow I will try a clean installation of bugfix 2.0 and pr / 20133

Roxy-3D, 2020-11-13:
> Right, that value was hard coded to the menu when an X argument was required, no default at the time. So the steps above just proved beyond doubt it was safe to remove.

My apologies for the hard coded number!   I'm glad that is pulled out of the code now!    Thank You!

[27 more replies]

closed 2021-03-29 by DrumClock