MarlinFirmware/Marlin #17944: [BUG] Extruder stops during print (Bugfix-2.0.x)

Exalik, 2020-05-10:
### Bug Description

The extruder randomly stops during prints.

### My Configurations

[Marlin Bugfix-2.0.x Config.zip](https://github.com/MarlinFirmware/Marlin/files/4606041/Marlin.Bugfix-2.0.x.Config.zip)

### Steps to Reproduce

1. Launch a print from sd card
2. The extruder randomly stops during the print.

**Expected behavior:** The extruder should continue to work to the end of the print.

**Actual behavior:** The extruder randomly stops to work during the print.

#### Additional Information
Geeetech A10M (GT2560 v3.1) with TMC2208 & BLTouch

randellhodges, 2020-05-10:
Try disabling stealthchop for your extruder. You probably want to go to Discord for support.

AmiSMB, 2020-05-10:
I have had a similar problem on my Ender 3 but I tried different firmware versions with no change. I  tried changing the stepper driver, then the stepper itself and the mainboard. Sometimes the issue would go away but then comeback. Then I changed the extruder stepper motor wires and the problem is now fixed.

Snagi0, 2020-05-10:
Your problem is TMC stepper as extruder driver in stealthchop mode and linear advance activated. Usually the suggestions are to disable linear advance, change to spreadcycle or switch to an inferior driver. These all are temporary fixes (they work), trying not to address the real problems. 

However, I think I might have a hypothesis what causes part of the TMC stealthchop issue. At the moment linear advance implementation (and possibly some other functions) is not very good, as it causes many very short abrupt control pulses (direction and step) for the driver. When TMC steppers operate in stealthchop mode they use automatic driver calibration as default. As I understand, part of the calibration is an ongoing process during the whole driver use session (print). Those short pulses ether trip the short circuit protection or somehow mess up the calibration causing the driver to halt.

I would like to see  an option to control/read all TMC stepper settings through gcode and also spreadcycle/stealthchop driver parameter calibration functionality through marlin. So we could calibrate suitable values, set them through uart or spi and then disable the automatic calibration functionality (stealthchop) inside the drivers.  

There has been multiple similar bug reports relating to this problem. Unfortunately usually they all get closed by the moderators and there is no real further investigation or bug fixes.

Exalik, 2020-05-11:
Hello to everyone,

The linear advance was already deactivated. I have switched the extruders TMC drivers to spreadCycle mode and relaunched the print. Let's see...

Snagi0, 2020-05-11:
Yeah I see now. The spreadcycle might still help.

Also I noticed you are using default stepper delay values. I believe they default to 
#define MINIMUM_STEPPER_POST_DIR_DELAY 20
#define MINIMUM_STEPPER_PRE_DIR_DELAY 20
#define MINIMUM_STEPPER_PULSE 0
with TMC drivers, but the relevant file is not included in the zip.

I think delay issues are more common with faster 32bit boards, but you might want to try these settings also:
#define MINIMUM_STEPPER_POST_DIR_DELAY 60
#define MINIMUM_STEPPER_PRE_DIR_DELAY 60
#define MINIMUM_STEPPER_PULSE 1 or 2

Also there are some issues with junction deviation that are worked on at the moment. I don't know if it could be related.

Exalik, 2020-05-12:
Hello, the workaround of this issue is the spreadCycle mode.

SirYasko, 2020-06-12:
I can confirm the problem is with TMC2208 (Creality Ender 5 PRO board 1.1.5 & BLTouch 3.1)
Adding:
M569 S0 E ; Disable stealthChop (use spreadCycle) for Extruder 
to gcode is workaround.

boelle, 2020-06-21:
@Exalik still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle