MarlinFirmware/Marlin #17947: [BUG] LASER_POWER_INLINE corrupts EEPROM, M504 crashes Marlin

kaixxx, 2020-05-10:
### Bug Description
I wanted to try out the recent laser improvements in Marlin 2.0 bugfix and compiled it with LASER_FEATURE and LASER_POWER_INLINE enabled. It worked at first, but after I saved some changes to the EEEPROM and restarted, I ran into a boot loop. The only way out was to flash a version without the laser features and rewrite the EEPROM. 
I could narrow it down to LASER_POWER_INLINE: if this is enabled, a combination of M500 and M504 crashes Marlin. 
I think this might have to do with the new laser settings that are introduced into planner_settings_t (planner.h: 235). Maybe a problem with padding? I have no idea.

### My Configurations
BTT SKR 1.4 Board, LPC1768, TMC2209
I am using FLASH_EEPROM_EMULATION, as recommended. I also tried SDCARD_EEPROM_EMULATION, this gave me some kind of size error when loading the stored values.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4606700/Configuration.zip)

### Steps to Reproduce
1. Compile latest Marlin 2.0 bugfix with LASER_FEATURE and LASER_POWER_INLINE enabled. 
2. Save EEPROM with M500 and verify it with M504

**Expected behavior:** Message: "EEPROM is ok"
**Actual behavior:** Crash and boot loop

#### Additional Information
none

naka123, 2020-05-11:
I had same error on SKR1.3 board 

There is my temporary hack to fix for it - https://gist.github.com/naka123/8585872be2d4646fbdd36349b239e8f4

kaixxx, 2020-05-11:
Thank you, naka123! I have tested your hack, it works!
Why are you not turning this into a pull request? The main problem seems to be that planner.settings.laser is not read properly. This is a major issue that makes the new laser functions pretty much unusable.

closed 2020-05-12 by kaixxx

thinkyhead, 2020-05-13:
Now patched in bugfix by moving the laser data out of `planner.settings`.

kaixxx, 2020-05-14:
nice, will test asap.