MarlinFirmware/Marlin #17949: [BUG] With a tilted bed the mbl causes sudden z-jumps of the  nozzel

Vladimir-VS, 2020-05-10:
### Bug Description

I am using the  manual mesh leveling  option and my printer is located on a tilted  surface. As a result the horizontally leveled bed is showing a respectively tilted mesh: 
 | 0 | 1 | 2
0 | -0.1 | 0.475 | 0.712
1 | -0.538 | -0.025 | 0.437
2 | -1.013 | -0.475 | 0.1

With this  mesh loaded the printer shows generally even surface of the 1st layer but in some  points the nozzle suddenly raises up and creates "hills" on the printed surface. The are replicated every time when  the parameters are close and are stable to certain variations of the mesh points. At every  hill point the z-lead rotation is visibly changing, thus  indicating that the hills are caused by the leveling.

If I decrease the mesh segments length to 1mm the hills disapear but a line (a bar) of a higher print appears parallel to axes x.
Increase of the segment length to 10 minimizes the  hills (but does not eliminate them).
Further increase of the segments length does  not change the results.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4606736/Configuration.zip)

### Steps to Reproduce

1. Tilt your bed as per the mesh above
2. Load the mesh to the printer
3. Print a relatively large plane (15x15 cm)

**Expected behavior:** [What you expect to happen]
Plane is even on all the surface

**Actual behavior:** [What actually happens]
Plane is even except the hills (ca. 2x2 to 10x10 mm in size) in several points.

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

sjasonsmith, 2020-05-29:
@Vladimir-VS, could you try this with the latest state of `bugfix-2.0.x`, to determine whether there is still an issue? If there is still a problem pictures of the problem would be helpful.

Vladimir-VS, 2020-05-29:
Jason,I have attached the pictures to the report.Are they enough?

boelle, 2020-06-21:
@Vladimir-VS still an issue?

Vladimir-VS, 2020-06-21:
Yes, no changes.

boelle, 2020-06-29:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

Vladimir-VS, 2021-07-17:
I am awaiting for the issue to be confirmed and resolved.

Vladimir-VS, 2021-09-20:
The issue is still valid. I am still awaiting for a solution.

thisiskeithb, 2021-09-20:
Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

Vladimir-VS, 2021-09-20:
Thisiskeithb, I have done this test several times (last on 22 Apr, [BUG] #17949) but this was never used for any attempts to solution. Now I am pretty sure there were no changes in the code that cpild fix the issue and would like to avoid breaking down current settings again to re-tune the printer again with no expectation to fix.

May I ask, please, to try instead finally replicating this issue at least once on your printer to get this issue moving forward (this has never been done so far)?
Alternatively, can you share the formulas used by the firmware to calculate the level at every point so that I can simulate the tilted bed virtually?

Thank you, 
Vladimir

closed 2022-05-03 by github-actions[bot]